britkat1980/giv_tcp #308: Grid frequency wrong by a factor of 10

DJBenson, 2024-11-26:
I'm running the latest production release and a parallel AIO setup and I've noticed the frequency entity for the gateway is reading incorrectly by a factor of 10. Is this a raw value from the inverter (as such needs reporting to GE) or something within GivTCP?

[image]

plandregan, 2024-12-08:
I have raised this too, I will delete my submission, didnt see this one
[image]

plandregan, 2025-02-19:
This will be fixed in Gateway Firmware. Tested already

DJBenson, 2025-02-19:
Can probably close this then.

closed 2025-02-19 by DJBenson

plandregan, 2025-02-19:
[image]

Will be in next GW release