britkat1980/giv_tcp #309: Suggestion: GivBack Sensor

jswilkinson851, 2024-12-02:
Would it be possible for a GivBack sensor to be implemented? Previously I would have relied on the Predbat-Octopus saving sessions combination, but I am now with E.On so I've moved over to the GivBack sessions instead.

A much as I would love to contribute to the creation of such a sensor, personally I wouldn't even know where to begin as I genuinely don't understand how HA even works.

Thank you for any feedback from the powers that be :)

gcoan, 2024-12-02:
Unless Axel energy (GivEnergy's partner for GivBack) provide an API, its going to be difficult to create a GivBack sensor with the only practical way of doing so is to write something to parse your email or a website if Axel publish the event data on a website?

In terms of Predbat activity, the safest thing to do is to set it to read only mode when a givback event is scheduled so that Predbat stops controlling the inverter and you let Axel take control.

Likely that the benefits will be much more limited this year

jswilkinson851, 2024-12-02:
I hadn't realised that the issue was that the information would have to come from Axel's side, so that makes perfect sense.

Thank you for the speedy response :)