britkat1980/giv_tcp #311: [3.0.4] GivEVC does not work

DJBenson, 2024-12-08:
I am running GivTCP 3.0.4 and have been upgraded to v1.1.13 of the firmware on the unit and it does not work with GivTCP. My EVC is connected by ethernet so I'm aware of the need to wait on reboot of the device for the WLAN to go down which then opens up the modbus port (an nmap trace shows the port is open).

The entities in GivTCP are all available - I get realtime power usage, grid voltage etc. but any commands fail to actually control the device so modbus IS working, but only for reading...

If I monitor the logs, I can see that the command is being picked up but it does not appear to get to the EVC.

Whilst tinkering last week I spotted that if I do the following, sometimes it does work, which looks like the commands are getting stuck somewhere;

- Send a command (like starting the charge)
- Watch the logs to see the command being picked up
- Restart GivTCP
- The logs again show the command being processed and sometimes it actually kicks the EVC into life 

This method does cause the modbus port to get locked up pretty easily though causing the 5 minute wait period for it to become available again. During this time NMAP shows the port is still open but assume from a GivTCP perspective it's not responding.

These are the entities created by GivTCP showing modbus is working to a certain extent;
[image]

matttheotter, 2024-12-21:
I'm noticing some behaviour issues with the EVC too.

GivTCP 3.0.4 - hasn't had a firmware upgrade yet.

Mine is on Wifi, I can see the automation running to initiate a charge however if I check the logs the EVC appears to be ignoring it.

I initiate the command from the cloud and it works.

plandregan, 2025-02-10:
Update EVC to 1.1.3.
Raise a ticket as its a manual update

CannyGent, 2025-03-04:
> Update EVC to 1.1.3. Raise a ticket as its a manual update
Hi - what should I say in the ticket to GivEnergy - just mention GivTCP or are they happy to do this just becasue I ask?

ColinShorts, 2025-07-14:
I raised a ticket and got updated to 1.1.14, and discovered that I was still running the addon at V2.x when it still wasn't working. Double check you've installed the V3 addon before you do anything else :)

[image]

DJBenson, 2025-07-15:
It's definitely v3.

ColinShorts, 2025-07-16:
Should hopefully just be the firmware update then 🤞 (and making sure it's not created another device)

DJBenson, 2025-10-23:
Completed?