britkat1980/giv_tcp #315: Wifi connection to Giv EVC problem

SteveClubley, 2024-12-22:
GivTCP version 3.0.4
GivTCP works fine with inverter and battry data however does not work with GivEnergy EV charger. EVC is connected via WIFI. Have used the correct IP address in the Config setup however continuously receive the following error message found in the log -
2024-12-22 14:54:22,105 - EVC - evc         -  [ERROR   ] - 10 consecutive errors, pausing and waiting 5 mins for EVC modbus port to come back
Please advise on potential causes of the problem.

SteveClubley, 2024-12-24:
Resolved by turning on the local control toggle found in the Settings menu on the EVC section of the Giv portal.

closed 2024-12-24 by SteveClubley

gcoan, 2024-12-24:
Created a PR to update the documentation https://github.com/britkat1980/ha-addons/pull/40