britkat1980/giv_tcp #318: GivTCP failing, "inverter Update failed so using last known good data from cache: ValueError("time data '2024-12-31T02:21:42.233111+00:00' does not match format '%Y-%m-%dT%H:%M:%S%z'"), <traceback object at 0x7f22dbdcbf80>), 1687) 2024-12-31 02:21:42,242 - G - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1825) 2024-12-31 02:22:12,948 - H - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: .. ValueError("time data does not match format")

gcoan, 2024-12-31:
GivTCP 3.0.4
2x Gen 1 Hybrid inverters with 191/193 fast response firmware, each inverter with their own battery

Been working absolutely fine since I upgraded to GivTCP 3.0.4 but noticed tonight that from about 00:10,  no inverter date/time updates coming through and Predbat was reporting repeated errors:

working fine:
[image]

started failing:
[image]
[image]

carried on repeating this failing

In the GivTCP log I see repeated two errors:

```
2024-12-31 02:21:42,241 - G - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'ValueError'>, ValueError("time data '2024-12-31T02:21:42.233111+00:00' does not match format '%Y-%m-%dT%H:%M:%S%z'"), <traceback object at 0x7f22dbdcbf80>), 1687)
2024-12-31 02:21:42,242 - G - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1825)
```

I have:
- checked inverters are online and are reporting data OK to the GivEnergy portal
- restarted the mesh network node in case it was a network issue
- reset the inverters to defaults from the GivEnergy portal
- restarted the inverters from the GivEnergy portal
- shutdown GivTCP, deleted the PKL cache files, restarted GivTCP
- installed and tried the beta and dev versions of GivTCP
- shutdown MQTT, GivTCP, restarted both
- shutdown HA and all the add-ons, rebooted the host PC, restarted the HAOS VM
- turned debug mode on

And all the time I get the same error messages repeated for both inverters:

ValueError("time data '2024-12-31T01:57:03.746086+00:00' does not match format '%Y-%m-%dT%H:%M:%S%z'"), <traceback object at 0x7fd723a0fac0>)

But to my eyes the date/time looks correct and looks correctly formatted.

One thing I had noticed as weird was that using the BBC basic app the inverter time was correctly reported but the date was appearing as "Date: 01--2025" 

In the debug logfile I saw the additional messages:

[DEBUG] - Inverter Time is default... fixing it

Which made me wonder if there was a problem with the inverter time....

On one inverter (G) I did a 'set date and time' through the portal which worked OK and in the GivTCP log I got " G - read        -  [INFO    ] - Publishing Home Assistant Discovery messages" and the normal heart-beat of messages from inverter G started appearing in the logbook:

[image]

Inverter H continued with the same error messages.

Screenshot of BBC basic app showing correct time now on one inverter and 'corrupt' time on the second inverter:

Setting the date and time for the second inverter and inverter H has burst into life:

[image]

I have no idea what caused this, why the date on both inverters got corrupted at the same time, it sort of suggests it was caused by GivTCP, but impossible to prove.

Moreover it looks like GivTCP detected that the datetime was in error 'Inverter Time is default... fixing it' but that auto-correct doesn't work properly.

GivTCP logfiles from today, debug turned on around 01:55

[startup.log](https://github.com/user-attachments/files/18278946/startup.log)
[log_inv_1.log](https://github.com/user-attachments/files/18278943/log_inv_1.log)
[log_inv_2.log](https://github.com/user-attachments/files/18278945/log_inv_2.log)
[write_log_inv_1.log](https://github.com/user-attachments/files/18278947/write_log_inv_1.log)
[write_log_inv_2.log](https://github.com/user-attachments/files/18278950/write_log_inv_2.log)

JRascagneres, 2024-12-31:
Yeah I investigated the same thing. Found in the end that I THINK Giv was returning month as 13, causing parsing to break. GivTCP then tries to fallback but the now() used to fallback provides milliseconds which it doesn't expect :)

gcoan, 2024-12-31:
As mentioned on #145 at least 3 other people have had this problem so looks likely to be an inverter firmware bug as we all had it at the same time ~ 00:10 on 31/12/24, and it appears to affect several different inverter types.

JRascagneres, 2024-12-31:
@gcoan Bunch of people on the GivTCP facebook group too, I've posted update there and tagged a Giv engineer in it. 

Interested to know cause if we'll ever know, whether it was some weird firmware edge-case, an update that got released that we didn't know about or something else...

gcoan, 2024-12-31:
Thanks @JRascagneres I'm not on Facebook so couldn't share it, but I have posted details of it on the GivEnergy community forum.

Tell the folks on FB to send me beer tokens !  Or if Giv want to give me a battery as a thank-you ...

apone2000, 2024-12-31:
Had the same at 00.10 this morning. Reset time via portal all all good. I’m on giv ac3 inverter. Won’t post logs as seems issue is now known, but happy to upload if they are of help. On Givtcp 2.4.9

robsessex, 2024-12-31:
@gcoan Thank you , setting time/date worked Happy New Year

bikerpatch, 2024-12-31:
@gcoan thank you very much, worked for me as well

henry3rd1961, 2024-12-31:
Exactly the same issue for me, forced reset of invertor time fixed the issue.

psweens, 2024-12-31:
Same problem here. Resetting inverter time fixed the issue. Thanks!

egooner, 2024-12-31:
Thanks for this. Been struggling with the same issue since last night! Upgraded to V3 and also upgraded the battery neither worked. Sending new date and time had fixed it!! 😃

Geekerbyname, 2024-12-31:
Been down a rabbit hole looking for this. Got up, saw battery hadn't done my HA automated charge and assumed something with HA! Thanks for the time fix.

gw1urf, 2024-12-31:
Ditto here - I tried accessing using some older code I'd written which used givenergy_modbus. It looked like the inverter's date was set to 2025-13-31 (!). The GivEnergy Android app was also failing for me. Eventually I was able to reset the date and time via the GivEnergy cloud service, and everything started working again.

I've opened issue 319 because it looks like the MQTT setDateTime function is broken - had intended to create a home assistant automation to set the time daily, as an interim workaround for this (likely) firmware bug.

zaheerm, 2024-12-31:
I believe I have a fix in #320 and #321

gcoan, 2024-12-31:
> I believe I have a fix in #320 and #321 that makes giv_tcp more robust to this issue. 

Brilliant thanks for these PR's.

We might have to wait 4 more years to test them !

HipHopAst, 2025-01-01:
Hi.
So how to solve this problem?
> [ERROR   ] - Error processing Inverter data:

I have Send command "Set Date and Time" on Givenergy cloud portal multiple times as well as reset inverter on this same portal. Still getting error on GivTCP Despite Givenergy Local working as it should. So what i need to do to solve this problem?

gcoan, 2025-01-01:
> I have Send command "Set Date and Time" on Givenergy cloud portal multiple times as well as reset inverter on this same portal. Still getting error on GivTCP Despite Givenergy Local working as it should. So what i need to do to solve this problem?

Resetting the inverter from the portal won't make any difference, and iof you done set date and time (its actually 'sync date and time' on the remote control page), that should have fixed it.

Strange that GivLocal is working but not GivTCP

First thing is to see what the inverter date and time is set to.  Either use the RubikCube Android app or the BBC app (available for Android and iOS - search for givenergy in the play/app store) and it should show you what the inverter time is set to.

I believe the issue only affected 31/12/24 being the last day of the leap year, so it should have been corrected today, so also strange that it's still affecting GivTCP.

If the date is not OK, you can set it from the BBC app.

If the date is OK, must be a givtcp issue.  Shut it down, delete all the /config/GivTCP/*.pkl files which are cache files and restart givtcp