britkat1980/giv_tcp #322: 2.4.9 - inverter no longer found overnight

JulesAchiel, 2024-12-31:
Overnight I observed my automation predbat didn't charge my battery which appears to be caused by a failure in givtcp.

Restarted givtcp but it's failing to find the inverter. Double checked the ip address which is correct. I'm able to control the battery through the givenergy app as well as the cloud. 

Restarted the inverter without any difference in givtcp. 

GivTCP yaml:
NUMINVERTORS: 1
INVERTOR_IP_1: 192.168.68.125
INVERTOR_NAME_1: Inv1
INVERTOR_AIO_1: false
INVERTOR_AC_1: false
NUMBATTERIES_1: 1
HADEVICEPREFIX: GivTCP
INVERTOR_IP_2: 192.168.68.125
INVERTOR_NAME_2: Inv2
NUMBATTERIES_2: 0
INVERTOR_AIO_2: false
INVERTOR_AC_2: true
HADEVICEPREFIX_2: GivTCP2
INVERTOR_IP_3: 192.168.68.117
INVERTOR_NAME_3: Inv3
NUMBATTERIES_3: 0
INVERTOR_AIO_3: false
INVERTOR_AC_3: true
HADEVICEPREFIX_3: GivTCP3
MQTT_OUTPUT: true
MQTT_ADDRESS: core-mosquitto
MQTT_USERNAME: ""
MQTT_PASSWORD: ""
MQTT_TOPIC: ""
MQTT_TOPIC_2: ""
MQTT_TOPIC_3: ""
MQTT_PORT: 1883
MQTT_RETAIN: false
LOG_LEVEL: Info
PRINT_RAW: true
SELF_RUN: true
SELF_RUN_LOOP_TIMER: 15
QUEUE_RETRIES: 2
HA_AUTO_D: true
DYNAMICTARIFF: true
DAYRATE: "0.155"
NIGHTRATE: "0.055"
EXPORTRATE: "0.04"
DAYRATESTART: "04:30"
NIGHTRATESTART: "00:30"
INFLUX_OUTPUT: false
INFLUX_URL: ""
INFLUX_TOKEN: ""
INFLUX_BUCKET: ""
INFLUX_ORG: ""
PYTHONPATH: /app
TZ: Europe/London
WEB_DASH: true
WEB_DASH_PORT: 3041
CACHELOCATION: /config/GivTCP
DATASMOOTHER: medium
SMARTTARGET: false
PALM_WINTER: 01,02,11,12
PALM_SHOULDER: 03,04,09,10
PALM_MIN_SOC_TARGET: 25
PALM_MAX_SOC_TARGET: 45
PALM_BATT_RESERVE: 4
PALM_BATT_UTILISATION: 0.85
PALM_WEIGHT: 35
GEAPI: ""
SOLCASTAPI: ""
SOLCASTSITEID: ""
SOLCASTSITEID2: ""
LOAD_HIST_WEIGHT: "1"
EVC_ENABLE: false
EVC_IP_ADDRESS: ""
EVC_SELF_RUN_TIMER: 5
HOSTIP: 192.168.68.110

Logfile:

2024-12-31 05:17:58,143 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2024-12-31 05:17:58,151 - startup - [INFO] - Supervisor Timezone: Europe/London
2024-12-31 05:18:00,164 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-12-31 05:18:04,041 - startup - [ERROR] - Gathering inverter details for 192.168.68.125 failed.
2024-12-31 05:18:04,041 - startup - [ERROR] - Unable to interrogate inverter to get base details
2024-12-31 05:18:04,219 - startup - [ERROR] - Gathering inverter details for 192.168.68.125 failed.
2024-12-31 05:18:04,220 - startup - [ERROR] - Unable to interrogate inverter to get base details
2024-12-31 05:18:04,221 - startup - [INFO] - Searching for Inverters again
2024-12-31 05:18:04,222 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-12-31 05:18:08,133 - startup - [ERROR] - Gathering inverter details for 192.168.68.125 failed.
2024-12-31 05:18:08,133 - startup - [ERROR] - Unable to interrogate inverter to get base details
2024-12-31 05:18:08,305 - startup - [ERROR] - Gathering inverter details for 192.168.68.125 failed.
2024-12-31 05:18:08,306 - startup - [ERROR] - Unable to interrogate inverter to get base details
2024-12-31 05:18:08,307 - startup - [INFO] - Searching for Inverters again
2024-12-31 05:18:08,307 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2024-12-31 05:18:12,257 - startup - [ERROR] - Gathering inverter details for 192.168.68.125 failed.
2024-12-31 05:18:12,257 - startup - [ERROR] - Unable to interrogate inverter to get base details
2024-12-31 05:18:12,462 - startup - [ERROR] - Gathering inverter details for 192.168.68.125 failed.
2024-12-31 05:18:12,463 - startup - [ERROR] - Unable to interrogate inverter to get base details
2024-12-31 05:18:12,474 - startup - [CRITICAL] - Running Redis
2024-12-31 05:18:12,483 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2024/12/31 05:18:12 [notice] 190#190: using the "epoll" event method
2024/12/31 05:18:12 [notice] 190#190: nginx/1.20.2
2024/12/31 05:18:12 [notice] 190#190: OS: Linux 6.6.62-haos-raspi
2024/12/31 05:18:12 [notice] 190#190: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2024/12/31 05:18:12 [notice] 190#190: start worker processes
2024/12/31 05:18:12 [notice] 190#190: start worker process 195
2024/12/31 05:18:12 [notice] 190#190: start worker process 196
2024/12/31 05:18:12 [notice] 190#190: start worker process 197
2024/12/31 05:18:12 [notice] 190#190: start worker process 198
2024-12-31 05:18:12,520 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2024-12-31 05:18:12,521 - startup - [CRITICAL] - Running Invertor (192.168.68.125) read loop every 15s
2024-12-31 05:18:12,546 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2024-12-31 05:18:12,568 - startup - [CRITICAL] - Starting Gunicorn on port 6345
2024-12-31 05:18:12,587 - startup - [CRITICAL] - Serving Web Dashboard from port 3041
[2024-12-31 05:18:13 +0000] [202] [INFO] Starting gunicorn 23.0.0
[2024-12-31 05:18:13 +0000] [202] [INFO] Listening at: http://0.0.0.0:6345 (202)
[2024-12-31 05:18:13 +0000] [202] [INFO] Using worker: sync
[2024-12-31 05:18:13 +0000] [214] [INFO] Booting worker with pid: 214
[2024-12-31 05:18:13 +0000] [215] [INFO] Booting worker with pid: 215
[2024-12-31 05:18:13 +0000] [216] [INFO] Booting worker with pid: 216
2024-12-31 05:18:16,318 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
 UPDATE  The latest version of `serve` is 14.2.4
 INFO  Accepting connections at http://localhost:3041
2024-12-31 05:19:56,445 - Inv1 - mqtt_client -  [ERROR   ] - No serial_number found in MQTT queue. MQTT Control not available.
Traceback (most recent call last):
  File "/app/GivTCP_1/mqtt_client.py", line 402, in <module>
    client.loop_forever()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2297, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 1686, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 2100, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3142, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3814, in _packet_handle
    return self._handle_connack()
  File "/usr/local/lib/python3.10/site-packages/paho/mqtt/client.py", line 3934, in _handle_connack
    on_connect(
  File "/app/GivTCP_1/mqtt_client.py", line 386, in on_connect
    client.subscribe(MQTT_Topic+"/control/"+GiV_Settings.serial_number+"/#")
AttributeError: type object 'GiV_Settings' has no attribute 'serial_number'
2024-12-31 05:20:13,107 - Inv1 - read        -  [ERROR   ] - 10 failed inverter reads in a row so removing regCache to force update...

Appreciate any help.

papagb, 2024-12-31:
I have the same symptoms

papagb, 2024-12-31:
ah, read 

https://github.com/britkat1980/giv_tcp/issues/145

You need to reset your inverters time

JulesAchiel, 2024-12-31:
@papagb how do I do that?

JulesAchiel, 2024-12-31:
Got it - from the GE portal

JulesAchiel, 2024-12-31:
After sending the inverter's time from the GE web portal and restarting givtcp and predbat all is working again.
Thanks @papagb for the tip

closed 2024-12-31 by JulesAchiel