britkat1980/giv_tcp #324: Large Startup Delay, Unable to Configure Until Network Discovery Completed

KingDigweed, 2024-12-31:
I've been trying to move from V2 to V3 of the add-on but am running into an issue with the setup process of V3. I'm unable to reach the webgui config stage because the add-on spends a couple of hours looking for devices on my network. I'm wondering if this is because of the size of my subnet?

The scanning process does eventually succeed, although in this case I deliberately left V2 running (so as not to have a large downtime) so got a connection in use error afterwards. I can however confirm this occurs even with the V2 add-on fully stopped before V3 is started.

I appreciate I could accept the outage time and just wait for this process to complete, this time with the V2 add-on stopped, but surely this is not intended behaviour? Cheers.

```
2024-12-31 01:51:50,449 - startup     -  [INFO    ] - Network Found: 192.168.0.1/16
2024-12-31 01:51:52,450 - startup     -  [INFO    ] - Searching for Inverters
2024-12-31 01:51:52,450 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2024-12-31 02:02:50,184 - startup     -  [INFO    ] - INV- Scanning network (1):192.168.0.1/16
2024-12-31 02:57:30,278 - startup     -  [INFO    ] - INV- Scanning network (2):192.168.0.1/16
2024-12-31 03:52:10,442 - client      -  [INFO    ] - Connection established to 192.168.0.66:8899
2024-12-31 03:52:10,442 - client      -  [INFO    ] - Detecting plant
2024-12-31 03:52:11,593 - client      -  [INFO    ] - Plant Detected

gcoan, 2024-12-31:
I don't have the same multi-hour startup as @KingDigweed has but I have noticed that V3 starts up considerably slower than V2, to the point that after a clean HA start Predbat will often complain that GivTCP isn't working and force a restart of GivTCP.

Could some form of cache of saved network/inverter configuration be provided to prevent this extended startup?

KingDigweed, 2024-12-31:
To update, I decided to put up with the delay and attempted to properly set-up the V3 add-on. After "only" a 1 hour wait this time, I managed to get to the web-ui and went through the set-up procedure. Got to the end, saved and the add-on rebooted... Again, seems like I'll be waiting a long time even after confirming the IP address etc. :(

britkat1980, 2025-01-06:
Current DEV version has an "auto_scan" hidden config item, which you can set to false to bypass ip scanning. If you directly modify the allsettings.json file and add or amend "auto_scan":true to "auto_scan":false then it should boot quicker. You will need to make sure you update the inverter config items manually if not previously picked up.