britkat1980/giv_tcp #327: 3.0.4 KeyError Traceback on startup: setts["evc_ip_address"]=evcList[1][0]

sOckhamSter, 2025-01-07:
This issue has been present for a while but I had multiple issues and wasn't certain which ones were related or not. I'm now certain that this issue is unique and related to detecting the EVC. I am running v3.0.4 in a test virtual machine (VirtualBox) for the moment, I haven't moved from 2.4.9 on my 'production' installation yet due to this issue.

- I install giv_tcp as normal and start it
- It detects everything fine and starts the configuration interface
- I run through the configuration interface adding my batteries, MQTT, and it makes no difference to the upcoming error whether or not I then configure the EVC page or not.
- Save and restart and the Traceback appears in the logs on the restart:

```
2025-01-07 09:28:35,032 - GivTCP - write       -  [CRITICAL] - Restarting the GivTCP Addon in 2s...
2025-01-07 09:28:37,811 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-01-07 09:28:37,845 - startup     -  [INFO    ] - Network Found: 192.168.0.1/22
2025-01-07 09:28:39,847 - startup     -  [INFO    ] - Searching for Inverters
2025-01-07 09:28:39,847 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-01-07 09:28:44,961 - startup     -  [INFO    ] - GivEVC 11288853545979 found at: 192.168.1.187
2025-01-07 09:28:44,961 - startup     -  [INFO    ] - INV- Scanning network (1):192.168.0.1/22
2025-01-07 09:29:34,081 - client      -  [INFO    ] - Connection established to 192.168.1.62:8899
2025-01-07 09:29:34,082 - client      -  [INFO    ] - Detecting plant
2025-01-07 09:29:35,331 - client      -  [INFO    ] - Plant Detected
2025-01-07 09:29:41,352 - startup     -  [INFO    ] - Inverter CE2122G257 which is a Gen1 - Ac with 1 batteries has been found at: 192.168.1.62
2025-01-07 09:29:41,356 - client      -  [INFO    ] - Connection established to 192.168.1.81:8899
2025-01-07 09:29:41,356 - client      -  [INFO    ] - Detecting plant
2025-01-07 09:29:42,518 - client      -  [INFO    ] - Plant Detected
2025-01-07 09:29:47,546 - startup     -  [INFO    ] - Inverter CH2315G086 which is a Gen1 - All_in_one with 4 batteries has been found at: 192.168.1.81
2025-01-07 09:29:47,551 - client      -  [INFO    ] - Connection established to 192.168.1.241:8899
2025-01-07 09:29:47,551 - client      -  [INFO    ] - Detecting plant
2025-01-07 09:29:48,718 - client      -  [INFO    ] - Plant Detected
2025-01-07 09:29:48,720 - startup     -  [INFO    ] - Inverter GW2314G032 which is a Gen1 - Gateway with 0 batteries has been found at: 192.168.1.241
2025-01-07 09:29:48,729 - startup     -  [INFO    ] - evcList: {2: ['192.168.1.187', '11288853545979']}
Traceback (most recent call last):
  File "/app/startup.py", line 500, in <module>
    setts["evc_ip_address"]=evcList[1][0]
                            ~~~~~~~^^^
KeyError: 1
KeyError: 1
```

britkat1980, 2025-01-13:
Ok, something is amiss here. I’ll need to make you a special dev version with some extra debugging added. I’ll dm you

sOckhamSter, 2025-01-15:
So really strangely GivTCP started working ok on my test VM so I restarted HA a few times and it carried on working. So I then decided to try and reproduce the whole thing by wiping the GivTCP folder. Second time around it now just gives me the traceback error on every start of GivTCP for the evcList. Really annoying because I can't find a consistency in its behaviour, but 99% of the time it failed with the same error on startup so at least I'm guaranteed to reproduce the error once you have a debug version available.

britkat1980, 2025-01-15:
My feeling is that this is driven by networking/subnets etc... Will try and get you a dev test version in the next day or so.

sOckhamSter, 2025-01-15:
Sounds promising thank you :)  Unfortunately I didn't snapshot the VM before wiping the GivTCP config so now I can't get it to work. I'll try several more times this evening and if it eventually decides to work I'll grab the log.

sOckhamSter, 2025-01-16:
Ok, so I've managed to get it running again after many dozens of startup attempts. So firstly, this is a failed startup log:

```
2025-01-16 08:34:48,195 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-01-16 08:34:48,207 - startup     -  [INFO    ] - Network Found: 192.168.0.1/22
2025-01-16 08:34:50,210 - startup     -  [INFO    ] - Searching for Inverters
2025-01-16 08:34:50,211 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-01-16 08:34:55,359 - startup     -  [INFO    ] - GivEVC 11288853545979 found at: 192.168.1.187
2025-01-16 08:34:55,359 - startup     -  [INFO    ] - INV- Scanning network (1):192.168.0.1/22
2025-01-16 08:35:44,484 - client      -  [INFO    ] - Connection established to 192.168.1.62:8899
2025-01-16 08:35:44,486 - client      -  [INFO    ] - Detecting plant
2025-01-16 08:35:45,735 - client      -  [INFO    ] - Plant Detected
2025-01-16 08:35:48,498 - startup     -  [INFO    ] - Inverter CE2122G257 which is a Gen1 - Ac with 1 batteries has been found at: 192.168.1.62
2025-01-16 08:35:48,503 - client      -  [INFO    ] - Connection established to 192.168.1.81:8899
2025-01-16 08:35:48,504 - client      -  [INFO    ] - Detecting plant
2025-01-16 08:35:53,175 - client      -  [INFO    ] - Plant Detected
2025-01-16 08:35:58,236 - startup     -  [INFO    ] - Inverter CH2315G086 which is a Gen1 - All_in_one with 4 batteries has been found at: 192.168.1.81
2025-01-16 08:35:58,239 - client      -  [INFO    ] - Connection established to 192.168.1.241:8899
2025-01-16 08:35:58,239 - client      -  [INFO    ] - Detecting plant
2025-01-16 08:35:59,404 - client      -  [INFO    ] - Plant Detected
2025-01-16 08:35:59,406 - startup     -  [INFO    ] - Inverter GW2314G032 which is a Gen1 - Gateway with 0 batteries has been found at: 192.168.1.241
2025-01-16 08:35:59,413 - startup     -  [INFO    ] - Inverter CE2122G257 not in settings file
2025-01-16 08:35:59,413 - startup     -  [INFO    ] - Adding Inverter CE2122G257 to slot 3
2025-01-16 08:35:59,413 - startup     -  [INFO    ] - evcList: {3: ['192.168.1.187', '11288853545979']}
Traceback (most recent call last):
  File "/app/startup.py", line 500, in <module>
    setts["evc_ip_address"]=evcList[1][0]
                            ~~~~~~~^^^
KeyError: 1
```

It started once and allowed me to run the web config, but then failed again a lot after that. But then after many more attempts it suddenly burst into life! No mention of the EVC in the working startup though (.187) so I'm wondering if it just didn't notice it in the scan on this occasion which is why it worked?

```
2025-01-16 09:06:34,423 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-01-16 09:06:34,439 - startup     -  [INFO    ] - Network Found: 192.168.0.1/22
2025-01-16 09:06:36,443 - startup     -  [INFO    ] - Searching for Inverters
2025-01-16 09:06:36,443 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-01-16 09:06:46,711 - startup     -  [INFO    ] - INV- Scanning network (1):192.168.0.1/22
2025-01-16 09:07:36,876 - client      -  [INFO    ] - Connection established to 192.168.1.62:8899
2025-01-16 09:07:36,876 - client      -  [INFO    ] - Detecting plant
2025-01-16 09:07:38,129 - client      -  [INFO    ] - Plant Detected
2025-01-16 09:07:47,146 - startup     -  [INFO    ] - Inverter CE2122G257 which is a Gen1 - Ac with 1 batteries has been found at: 192.168.1.62
2025-01-16 09:07:47,151 - client      -  [INFO    ] - Connection established to 192.168.1.81:8899
2025-01-16 09:07:47,152 - client      -  [INFO    ] - Detecting plant
2025-01-16 09:07:48,315 - client      -  [INFO    ] - Plant Detected
2025-01-16 09:07:58,432 - client      -  [INFO    ] - Connection established to 192.168.1.81:8899
2025-01-16 09:07:58,432 - client      -  [INFO    ] - Detecting plant
2025-01-16 09:08:02,096 - client      -  [INFO    ] - Plant Detected
2025-01-16 09:08:07,139 - startup     -  [INFO    ] - Inverter CH2315G086 which is a Gen1 - All_in_one with 4 batteries has been found at: 192.168.1.81
2025-01-16 09:08:07,144 - client      -  [INFO    ] - Connection established to 192.168.1.241:8899
2025-01-16 09:08:07,144 - client      -  [INFO    ] - Detecting plant
2025-01-16 09:08:08,309 - client      -  [INFO    ] - Plant Detected
2025-01-16 09:08:08,312 - startup     -  [INFO    ] - Inverter GW2314G032 which is a Gen1 - Gateway with 0 batteries has been found at: 192.168.1.241
2025-01-16 09:08:08,324 - startup     -  [INFO    ] - Inverter CE2122G257 not in settings file
2025-01-16 09:08:08,325 - startup     -  [INFO    ] - Adding Inverter CE2122G257 to slot 3
2025-01-16 09:08:08,343 - startup     -  [INFO    ] - Setting up invertor: 1
2025-01-16 09:08:08,467 - startup     -  [INFO    ] - ==============================================================
2025-01-16 09:08:08,467 - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2025-01-16 09:08:08,467 - startup     -  [INFO    ] - ====     http://192.168.1.14:8099/config.html             ====
2025-01-16 09:08:08,467 - startup     -  [INFO    ] - ==============================================================
2025-01-16 09:08:08,469 - startup     -  [INFO    ] - Running Invertor 1 (CH2315G086) read loop every 30/120s
2025-01-16 09:08:08,475 - startup     -  [INFO    ] - Setting up invertor: 2
[... cut ...]
2025-01-16 09:08:18,951 - GivTCP3 - read        -  [INFO    ] - Dynamic Tariff enabled so defaulting to Day Rate and waiting external control
2025-01-16 09:08:18,956 - GivTCP3 - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
2025-01-16 09:08:23,570 - GivTCP - read        -  [INFO    ] - Dynamic Tariff enabled so defaulting to Day Rate and waiting external control
2025-01-16 09:08:23,574 - GivTCP - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
```

m47een, 2025-01-17:
Having a similar issue here.

2025-01-17 13:40:54,814 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-01-17 13:40:54,824 - startup     -  [INFO    ] - Network Found: 192.168.30.1/24
2025-01-17 13:40:56,825 - startup     -  [INFO    ] - Searching for Inverters
2025-01-17 13:40:56,825 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-01-17 13:40:58,075 - startup     -  [INFO    ] - GivEVC 87023906844969 found at: 192.168.30.144
2025-01-17 13:40:58,076 - startup     -  [INFO    ] - INV- Scanning network (1):192.168.30.1/24
2025-01-17 13:41:07,229 - client      -  [INFO    ] - Connection established to 192.168.30.141:8899
2025-01-17 13:41:07,229 - client      -  [INFO    ] - Detecting plant
2025-01-17 13:41:08,384 - client      -  [INFO    ] - Plant Detected
2025-01-17 13:41:13,405 - startup     -  [INFO    ] - Inverter CH2316G179 which is a Gen1 - All_in_one with 4 batteries has been found at: 192.168.30.141
2025-01-17 13:41:13,406 - client      -  [INFO    ] - Connection established to 192.168.30.142:8899
2025-01-17 13:41:13,406 - client      -  [INFO    ] - Detecting plant
2025-01-17 13:41:14,571 - client      -  [INFO    ] - Plant Detected
2025-01-17 13:41:14,571 - startup     -  [INFO    ] - Inverter GW2316G059 which is a Gen1 - Gateway with 0 batteries has been found at: 192.168.30.142
2025-01-17 13:41:14,573 - startup     -  [INFO    ] - evcList: {2: ['192.168.30.144', '87023906844969']}
Traceback (most recent call last):
  File "/app/startup.py", line 502, in <module>
    setts["serial_number_evc"]=evcList[1][1]
                               ~~~~~~~^^^
KeyError: 1

sOckhamSter, 2025-02-18:
Hi @britkat1980 have you managed to do any investigation on this yet at all? Still happy to try a dev version at your request and provide more logs if needed?

m47een, 2025-02-24:
I'm still having the issue on 3.0.4, please advise if there is a dev version I can run to help fix this bug.

britkat1980, 2025-03-03:
Beta 3.1.2 has fix to (hopefully) properly capture the discovered EVC in a well defined list.

sOckhamSter, 2025-03-09:
@britkat1980 I'm getting this on startup of 3.1.2. Currently running HA in a VM (bridged so it should be fine). I'll try it on a spare Pi too just to be certain:

```
2025-03-09 09:35:05,370 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-03-09 09:35:05,374 - startup     -  [INFO    ] - Network Found: 192.168.1.104/22
2025-03-09 09:35:07,379 - startup     -  [INFO    ] - Searching for Inverters
2025-03-09 09:35:07,379 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-03-09 09:35:07,379 - startup     -  [ERROR   ] - Error scanning for Inverters- ('ValueError', 'startup.py', 225)
2025-03-09 09:35:07,379 - startup     -  [ERROR   ] - =============================================================
2025-03-09 09:35:07,380 - startup     -  [ERROR   ] - ====               NO INVERTERS FOUND                    ====
2025-03-09 09:35:07,380 - startup     -  [ERROR   ] - ====      add manually using a file editor to modify     ====
2025-03-09 09:35:07,380 - startup     -  [ERROR   ] - ====          /config/GivTCP/allsettings.json            ====
2025-03-09 09:35:07,380 - startup     -  [ERROR   ] - =============================================================
```

sOckhamSter, 2025-03-09:
@britkat1980 nope, sorry... same error on a physical and ethernet-connected Raspberry Pi so I think v3.1.2 is having issues scanning the network for anything at all now :(

sOckhamSter, 2025-03-28:
@britkat1980 After many hours of debugging I think I have cracked it. I've submitted a PR to the dev3 branch I'd be grateful for you to cast your eyes over and maybe push into the beta build in HA. It was all to do with the iteration over the network range which I have corrected.

What made this trickier to debug is that for the HA add-on you have code that detects the subnet mask. But if running in Docker it is hard-coded to /24. So despite me running it in Docker on a /22 network, it never had the bug in Docker until I discovered I could manually change the hard-coded value. It was really puzzling me why the same code worked fine in Docker but not HA! Anyway, I'm hoping this is finally sorted with this update. #361

sOckhamSter, 2025-04-09:
@britkat1980 this is all working now in the beta 3.1.3 that you have just pushed, thank you so much. Everything appears to be working and stable in my test VM. Looking forward to see this pushed into the main release when you get chance :) I can then work on my beginners guide video at long last!

closed 2025-04-20 by sOckhamSter