britkat1980/giv_tcp #328: Control Force Charge on 3phase inverter

Valex1p, 2025-01-11:
Maybe not a HA issue but I'll mention it anyway.

The Control Force Charge control has various duration options. If selected it seems to work by sticking the appropriate time period into Charge Slot 1. If you happen to be already using that slot (as I was!) then it is overwritten.

Not sure if this is the way GE have built it or the way giv_tcp has implemented it.

sjs-1956, 2025-01-11:
Hi
I think you'll find that the slot settings are restored after the force expires or is Cancelled.

Regards
Steve Saunders

Valex1p, 2025-01-11:
Hmm not for me. They stay at the those values until you initiate a control again

Valex1p, 2025-01-11:
Inverter Firmware Version DA0.007-DD0.007-A0.007

gcoan, 2025-01-11:
> The Control Force Charge control has various duration options. If selected it seems to work by sticking the appropriate time period into Charge Slot 1. If you happen to be already using that slot (as I was!) then it is overwritten.

TBH I thought that was the way it worked.  The quick options such as force charge 60 minutes writes a charge into slot 1 on the inverter, overwriting any prior slot info.

If you want to keep a pre-programmed slot and use force charge alongside it, use slot 2 or higher

closed 2025-10-23 by britkat1980