britkat1980/giv_tcp #329: BMS Temperature and Battery Temperature sensor values swapped over?

IWR69, 2025-01-12:
Here is a graph that shows the BMS Temperature and Battery Temperature sensor values but I've also added the other four Cell Temperature values too. You would expect the BMS Temperature to spike during the charge (2am) and discharge (4pm) periods but instead the Battery Temperature spikes while the BMS Temperature closely tracks the four cell temperatures. It appears to me that these two sensors have been mistakenly mapped to each others sensor value.

[image]

I have a GivEnergy 5.0kW Hybrid Inverter Gen 3 and 9.5kW Battery Gen 2.
I'm running HA 2025.1.2 and GivTCP 3.0.4

gcoan, 2025-01-12:
See https://github.com/britkat1980/ha-addons/issues/39

That was my conclusion as well that the sensor names are swapped round

springfall2008, 2025-01-12:
I can confirm that also, as the GE cloud integration gives battery temperature as the value reported as bms temperature in GivTCP. It would be useful for GivTCP to expose both of these.

ProphetOfDoom, 2025-01-13:
This is my AIO tonight, where the BMS temp also closely follows a battery stack cell temp rather than that of the inverter, which would seem to support your observations.

[image]

IWR69, 2025-04-18:
This is still an issue in 3.1.6