britkat1980/giv_tcp #330: Recurrent disconnects from MQTT server

jdiviney, 2025-01-12:
GivTCP spams the MQTT broker every second or so with a connect and disconnect. This is continuous and fills the whole log.
Running latest versions on home assistant as add-on

Log:
2025-01-12 10:11:27: New client connected from 172.30.32.1:33321 as GivEnergy_GivTCP_1 (p2, c1, k60, u'addons').
2025-01-12 10:11:27: Client GivEnergy_GivTCP_1 disconnected.
2025-01-12 10:11:48: New connection from 172.30.32.1:57119 on port 1883.
2025-01-12 10:11:49: Client <unknown> closed its connection.
2025-01-12 10:11:49: New connection from 172.30.32.1:51001 on port 1883.
2025-01-12 10:11:49: New client connected from 172.30.32.1:51001 as GivEnergy_GivTCP_1 (p2, c1, k60, u'addons').
2025-01-12 10:11:49: Client GivEnergy_GivTCP_1 disconnected.
2025-01-12 10:12:11: New connection from 172.30.32.1:43523 on port 1883.
2025-01-12 10:12:12: Client <unknown> closed its connection.
2025-01-12 10:12:12: New connection from 172.30.32.1:45227 on port 1883.
2025-01-12 10:12:12: New client connected from 172.30.32.1:45227 as GivEnergy_GivTCP_1 (p2, c1, k60, u'addons').
2025-01-12 10:12:12: Client GivEnergy_GivTCP_1 disconnected.
2025-01-12 10:12:33: New connection from 172.30.32.1:44699 on port 1883.
2025-01-12 10:12:34: Client <unknown> closed its connection.
2025-01-12 10:12:34: New connection from 172.30.32.1:59827 on port 1883.
2025-01-12 10:12:34: New client connected from 172.30.32.1:59827 as GivEnergy_GivTCP_1 (p2, c1, k60, u'addons').

gcoan, 2025-01-13:
I am running givtcp v3 and am still seeing repeated disconnects and connects in mosquitto broker, albeit not as bad as v2:

[image]

hwb45, 2025-01-13:
Exactly the same here