britkat1980/giv_tcp #331: Conversion of V2 to V3 Failing

nbullus, 2025-01-13:
Stopped v2, started V3 and I get error connecting to the MQTT Broker:

[image]

I'm running Mosquitto as my broker from within HA.

Connecting to MQTT explorer connects fine to the broker using my default username/password:

[image]

I spot that the serial number for my AIO inverter is reported at 2 levels and imbedded within itself:

- GivEnergy\CH2316G434\CH2316G434

This duplicates some entries below (with some differences in values, e.g. battery capacity):

- GivEnergy/CH2316G434/Invertor_Details

I also got no notification of V2 settings being found and migrated across. I have therefore decided to stop and revert back to V2 whilst this issue is investigated and further instructions on how to install.

gcoan, 2025-01-13:
I didn't get any MQTT errors from GivTCP v3 when I upgraded, GivTCP just auto discovered my MQTT.  I think you can delete the username and password configured in GivTCP to force it to reconnect.

What I did experience though is a load of 'duplicate sensor' errors in the HA core due to the tree structure for the battery sensors changing in v3.
I had to delete all the GivTCP battery sensor in MQTT and then restart GivTCP v3, and everything then worked fine.

Didn't lose any history because the sensor history is held in HA not MQTT, but the mapping of MQTT paths to HA battery entities got broken due to the changed pathnames.

In hindsight I could have deleted the entire GivTCP tree structures and it would have recreated everything

nbullus, 2025-01-16:
> > Not sure about the MQTT username/password. Tried with the default and also tried changing to my own credentials that V2 used, and that failed as well.
> 
> v3 definitely doesn't use the same saved MQTT credentials as v2 (and I have a recollection that if you enter them, it doesn\t work). I am pretty sure I just clicked enable MQTT in GivTCP config and that was it. When I look now there is a username and password set that I didn't create and isn't the v2 credentials.
> 
> Update: In the Predbat install instructions for GivTCP v3 I wrote: "GivTCP now auto-populates the MQTT page so as long as you're using Mosquitto broker within Home Assistant; you won't need to create a dedicated MQTT user or enter the details on the MQTT page"

Yes you are right - v3 sets the MQTT addon username/password automatically (which is what I have), but it still cannot connect to the MQTT broker as per original log message. Further advice needed.

gcoan, 2025-01-16:
sorry no idea, I haven't seen this issue before.

Can only suggest uninstalling GivTCP and Mosquitto completely, restarting HA and the add-on's and repeating the install process to see if that cures it.

nbullus, 2025-01-16:
> sorry no idea, I haven't seen this issue before.
> 
> Can only suggest uninstalling GivTCP and Mosquitto completely, restarting HA and the add-on's and repeating the install process to see if that cures it.

I have uninstalled v2 and v3 of GivTCP, Uninstalled Mosquitto and restarted HA. Reinstalled Mosquitto and GivTCP v3, but on starting I now get bad connection:

[image]

gcoan, 2025-01-16:
strange

when you re-installed mosquitto add-on did you configure the mosquitto integration?

https://springfall2008.github.io/batpred/inverter-setup/#givenergy-with-givtcp

Anything in the mosquitto log or the HA core log?

nbullus, 2025-01-16:
> strange
> 
> when you re-installed mosquitto add-on did you configure the mosquitto integration?
> 
> https://springfall2008.github.io/batpred/inverter-setup/#givenergy-with-givtcp
> 
> Anything in the mosquitto log or the HA core log?

I did, but to be sure I un-installed the Mosquitto add-on, and then deleted the MQTT integration as there was a load of crap still lying there. When I reinstalled Mosquitto it added the MQTT integration back in and I configured as per guide which is basically no configuration needed, i.e. just click OK.

nbullus, 2025-01-16:
> I did, but to be sure I un-installed the Mosquitto add-on, and then deleted the MQTT integration as there was a load of crap still lying there. When I reinstalled Mosquitto it added the MQTT [...]

Here is the Mosquitto Log:

[image]

nbullus, 2025-01-16:
HA log shows nothing interesting.

I still have my MQTTUser in HA which is obviously left over from the last install, but this is just no longer being used. I can still login to MQTT explorer with it to get a client connection to the broker.

[image]

But obviously its empty as GivTCP cannot connect to it yet.

nbullus, 2025-01-16:
> strange
> 
> when you re-installed mosquitto add-on did you configure the mosquitto integration?
> 
> https://springfall2008.github.io/batpred/inverter-setup/#givenergy-with-givtcp
> 
> Anything in the mosquitto log or the HA core log?

I'v managed to sort it myself I think. Looks like it cannot connect to the broker using the built-in MQTT user so i have replaced the pre-populated user/password (opposite to what the instructions say) with my dedicate mqtt user/password and now it seems to connect. Deleting everything and reinstalling has lost all my sensor history which I wanted to avoid, but better to have it working. Just need to reconfigure Predbat as all the sensors have now been renamed (does not default to GIVTCP prefix) and so just need to point it at the new names.