britkat1980/giv_tcp #332: Can't talk to AIO --- ValueError("time data '2025-01-16T10:07:01.446944+00:00' does not match format '%Y-%m-%dT%H:%M:%S%z'"), <traceback object at 0x7efff5ed8400>

peter1rhodes, 2025-01-16:
When trying to connect to my AIO using GivTCP v3.0.4, I can talk to most of my devices OK, but it won't get data from the AIO, instead it reports the following error:

ValueError("time data '2025-01-16T10:07:01.446944+00:00' does not match format '%Y-%m-%dT%H:%M:%S%z'"), <traceback object at 0x7efff5ed8400>

This suggests to me that the AIO is reporting correctly with the right time, but it's being rejected because there are decimal seconds, perhaps? Does anyone have any insight? It might be that my AIO firmware is old (?) and they have since removed the decimal seconds, perhaps that's why it's not affecting everyone?

gcoan, 2025-01-16:
The actual error that occurred at the start of the year was due to the inverter having an invalid time, then GivTCP defaulting to current time but the current time defaulting had microseconds in it so that was actually what failed.

i.e. the error we saw was a secondary error caused by the original bad inverter time.

There is a PR to fix this but its not yet merged into the code.

Do you have the BBC inverter app installed or the Rubik Cube Android app, can you see what the inverter time actually is (both of these report it).

It might be worth doing a synchronise time from the givenergy portal to your inverter or set time in the BBC app to force the time to be correct in case there is something wrong with the inverter time

peter1rhodes, 2025-01-16:
@gcoan Thank you for the reply. Neither "BBC inverter", nor "Rubik Cube" give any results in the Play Store that are anything to do with inverters, networking, etc. Do you have links?

gcoan, 2025-01-16:
the so called 'rubik cube' app https://play.google.com/store/apps/details?id=uk.rubikcube.invertermonitor&hl=en_GB

the 'BBC' inverter app https://bbcbasic.co.uk/inverter.apk

gcoan, 2025-01-16:
when I had the 31/12/24 bug (which the BBC app enabled me to find), the way I fixed it with my inverter was to do 'synchronise time' from the givenergy portal 

check the time zone is correct in your givenergy account settings first 

it was only afterwards that using these apps was suggested to set the clock so I haven't tried it myself

peter1rhodes, 2025-01-16:
@gcoan Thank you, you've been really helpful. I've had a look and can't find "Synchronise time" in the Givenergy portal, can you remember where that option is?

closed 2025-01-17 by peter1rhodes