britkat1980/giv_tcp #333: Intermittant error message in log

sjs-1956, 2025-01-17:
Using v3.0.4 with 2 AC3.0 inverters. One (GIVTCP2) has just been installed. Other, older one is two years old. Both on same version of firmware (D0.535-A0.535). I see an error communicating with new inverter in the log file maybe 4-5 times a day. No apparent symptoms seen in the data displayed.

`
2025-01-17 07:31:17,707 - GivTCP2 - mqtt        -  [INFO    ] - Requesting Control Action: enableChargeSchedule - {'state': 'disable'}
2025-01-17 07:31:17,750 - GivTCP2 - mqtt        -  [INFO    ] - Requesting Control Action: setChargeRate - {'chargeRate': '0'}
2025-01-17 07:31:18,021 - GivTCP2 - write       -  [INFO    ] - Setting Charge Schedule to disable was a success
2025-01-17 07:31:18,516 - GivTCP2 - write       -  [INFO    ] - Setting battery charge rate 0 was a success
2025-01-17 08:00:05,387 - GivTCP - mqtt        -  [INFO    ] - Requesting Control Action: enableChargeSchedule - {'state': 'disable'}
2025-01-17 08:00:05,427 - GivTCP - mqtt        -  [INFO    ] - Requesting Control Action: setChargeRate - {'chargeRate': '0'}
2025-01-17 08:00:06,592 - GivTCP - write       -  [INFO    ] - Setting Charge Schedule to disable was a success
2025-01-17 08:00:07,084 - GivTCP - write       -  [INFO    ] - Setting battery charge rate 0 was a success
2025-01-17 09:04:42,045 - GivTCP2 - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'TypeError'>, TypeError("unsupported operand type(s) for +: 'float' and 'NoneType'"), <traceback object at 0x7fa1a24200>), 1040)
2025-01-17 09:04:42,046 - GivTCP2 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7fa1ee4100>), 1657)
2025-01-17 09:04:42,047 - GivTCP2 - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1825)
2025-01-17 15:04:45,474 - GivTCP2 - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'TypeError'>, TypeError("unsupported operand type(s) for +: 'float' and 'NoneType'"), <traceback object at 0x7fa1e7c200>), 1040)
2025-01-17 15:04:45,475 - GivTCP2 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7fa1a7b100>), 1657)
2025-01-17 15:04:45,476 - GivTCP2 - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1825)
2025-01-17 15:52:45,994 - GivTCP2 - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'TypeError'>, TypeError("unsupported operand type(s) for +: 'float' and 'NoneType'"), <traceback object at 0x7fa1f02cc0>), 1040)
2025-01-17 15:52:45,995 - GivTCP2 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7fa1edb400>), 1657)
2025-01-17 15:52:45,995 - GivTCP2 - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1825)
2025-01-17 16:00:00,271 - GivTCP - mqtt        -  [INFO    ] - Requesting Control Action: enableDischargeSchedule - {'state': 'enable'}
2025-01-17 16:00:00,311 - GivTCP2 - mqtt        -  [INFO    ] - Requesting Control Action: enableDischargeSchedule - {'state': 'enable'}
2025-01-17 16:00:00,315 - GivTCP - mqtt        -  [INFO    ] - Requesting Control Action: setDischargeRate - {'dischargeRate': '3000'}
`

gcoan, 2025-01-17:
It could be a network issue with the new inverter.  How are they connected to your wifi, do you have a mesh network and its maybe connected to a weaker node in your house?

Another thought: If you login to the wifi hotspot on each of your inverters maybe you need to change the broadcast channels on them to stop conflict with your home wifi?  (Read the givenergy wifi setup guide if you've not done this before)

sjs-1956, 2025-01-17:
Hi
Connected via GivEnergy dongle to WiFi access point in shed (6ft away from dongles) that has no other connections. Strange that is always the new inverter whilst they are supposedly on the same firmware version.

Regards
Steve Saunders