britkat1980/giv_tcp #336: Invalid spikes in sensor values

IWR69, 2025-01-23:
I have an issue that has been reported previously, https://github.com/britkat1980/giv_tcp/issues/268, and although closed as fixed it still exists as an issue in 3.0.4 (which otherwise is running well, and fast :).

I am experience spikes in values of several sensors, some of which I have detailed below, but which do not appear listed as changed under GivEnergys Portal Remote Control History.

Here's an example of the Battery Capacity kWh sensor...

[image]

At the same time these spikes occur Home Assistants Logbook records GivTCP retrieving some invalid, possibly default, values.

For example when Battery Capacity kWh spikes I noticed Control Mode change to Unknown, Battery Type to Lead_acid (I have Lithium) and Meter Type to Ct_or_em418 (should be Em115). I have checked the Logbook for several of these spikes and they all show these invalid changes.

When Power SOC kWh incorrectly changes to 13.16, I have a 9.52 battery, Control Mode changes to Unknown (should be Eco) and Battery Type changes to Lead_acid. And when Power SOC kWh incorrect changed to 0, Control Mode changed to Eco (Paused). 

And for Inverter Max Bat Rate I noticed Meter Type change to Ct_or_em418, Inverter Type to Gen1 Hybrid (I have Gen3 Hybrid) and Inverter Firmware to D0.0-A0.0 (I have D0.311-A0.311).

I hope you are able to track this issue down as it causes my own sensors to react to these invalid changes causing other issues for me.
Thanks for your help.

IWR69, 2025-04-13:
This is still an issue in 3.1.3.

[image]

I've checked Remote Control History in the Portal and none of the spikes up to 3600 are logged and therefore never happened.

gcoan, 2025-04-13:
I don't see battery discharge rate change in mine, but do see battery type and battery firmware change randomly, and then a period of time change back again

[image]
[image]

givtcp beta 3.1.4

when you see these blips occurring do you see anything in the givtcp log at the same time?

IWR69, 2025-04-13:
There's nothing in the logbooks unfortunately but here are the images of the logbook and GivEnergys Remote Control History too.

I initially added some of the GivTCP devices and then detected that the new Remote Control History entity is also affected.
[image]

[image]

[image]

This next image shows you how both Charge and Discharge Rates have been affect this month.
[image]

GivEnergys Remote Control History
Octopus Energy control my battery but you can see there are very few changes to the rates (compared to the graphs above)
[image]

closed 2025-10-23 by britkat1980