britkat1980/giv_tcp #337: Error publishing data from pickle

TheWhiteDragonsRage, 2025-01-23:
new GivEnergy install on current HA Branch.,
tried uninstall/reinstall/reconfig/cleared cache/purged GivTCP Folder, but still getting this error..

any ideas

2025-01-23 16:48:06,985 - GivTCP - write       -  [CRITICAL] - Restarting the GivTCP Addon in 2s...
2025-01-23 16:48:10,034 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-01-23 16:48:10,048 - startup     -  [INFO    ] - Network Found: 192.168.1.xxxx/24
2025-01-23 16:48:12,049 - startup     -  [INFO    ] - Searching for Inverters
2025-01-23 16:48:12,049 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-01-23 16:48:14,484 - startup     -  [INFO    ] - INV- Scanning network (1):192.168.1.xxxx/24
2025-01-23 16:48:21,535 - client      -  [INFO    ] - Connection established to 192.168.1.xxxx:8899
2025-01-23 16:48:21,536 - client      -  [INFO    ] - Detecting plant
2025-01-23 16:48:22,807 - client      -  [INFO    ] - Plant Detected
2025-01-23 16:48:28,809 - startup     -  [INFO    ] - Inverter CExxxxx which is a Gen1 - Ac with 1 batteries has been found at: 192.168.1.xxxxx
2025-01-23 16:48:28,824 - startup     -  [INFO    ] - Setting up invertor: 1
2025-01-23 16:48:28,899 - startup     -  [INFO    ] - ==============================================================
2025-01-23 16:48:28,900 - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2025-01-23 16:48:28,900 - startup     -  [INFO    ] - ====     http://192.168.1.xxxxx:8099/config.html             ====
2025-01-23 16:48:28,900 - startup     -  [INFO    ] - ==============================================================
2025-01-23 16:48:28,901 - startup     -  [INFO    ] - Running Invertor 1 (CEccccc) read loop every 30/120s
2025-01-23 16:48:28,909 - startup     -  [INFO    ] - Serving Web Dashboard from port 3000
2025-01-23 16:48:28,931 - startup     -  [INFO    ] - Setting daily charge target forecast job to run at: 23:20
[2025-01-23 16:48:29 +0000] [70] [INFO] Starting gunicorn 23.0.0
[2025-01-23 16:48:29 +0000] [70] [INFO] Listening at: http://0.0.0.0:6350 (70)
[2025-01-23 16:48:29 +0000] [70] [INFO] Using worker: sync
[2025-01-23 16:48:29 +0000] [86] [INFO] Booting worker with pid: 86
[2025-01-23 16:48:29 +0000] [80] [INFO] Starting gunicorn 23.0.0
[2025-01-23 16:48:29 +0000] [80] [INFO] Listening at: http://0.0.0.0:6345 (80)
[2025-01-23 16:48:29 +0000] [80] [INFO] Using worker: sync
[2025-01-23 16:48:29 +0000] [87] [INFO] Booting worker with pid: 87
[2025-01-23 16:48:29 +0000] [88] [INFO] Booting worker with pid: 88
[2025-01-23 16:48:29 +0000] [89] [INFO] Booting worker with pid: 89
2025-01-23 16:48:29,639 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-01-23 16:48:29,640 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.1.xxxx
2025-01-23 16:48:29,646 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-01-23 16:48:36,686 - GivTCP - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
2025-01-23 16:48:55,518 - GivTCP - read        -  [ERROR   ] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-23 16:49:26,647 - GivTCP - read        -  [ERROR   ] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-23 16:49:57,727 - GivTCP - read        -  [ERROR   ] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-23 16:50:28,826 - GivTCP - read        -  [ERROR   ] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-23 16:51:03,924 - GivTCP - read        -  [ERROR   ] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-23 16:51:35,007 - GivTCP - read        -  [ERROR   ] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-23 16:52:06,112 - GivTCP - read        -  [ERROR   ] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-23 16:52:37,211 - GivTCP - read        -  [ERROR   ] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-23 16:53:09,579 - GivTCP - read        -  [ERROR   ] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-23 16:53:40,697 - GivTCP - read        -  [ERROR   ] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-23 16:54:11,794 - GivTCP - read        -  [ERROR   ] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-23 16:58:01,457 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-01-23 16:58:01,471 - startup     -  [INFO    ] - Network Found: 192.168.1.xxxxx/24
2025-01-23 16:58:03,473 - startup     -  [INFO    ] - Searching for Inverters
2025-01-23 16:58:03,474 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-01-23 16:58:05,922 - startup     -  [INFO    ] - INV- Scanning network (1):192.168.1.xxxxx/24
2025-01-23 16:58:12,210 - client      -  [INFO    ] - Connection established to 192.168.1.xxxx:8899
2025-01-23 16:58:12,211 - client      -  [INFO    ] - Detecting plant
2025-01-23 16:58:16,705 - client      -  [INFO    ] - Plant Detected
2025-01-23 16:58:19,464 - startup     -  [INFO    ] - Inverter CExxx which is a Gen1 - Ac with 1 batteries has been found at: 192.168.1.xxx
2025-01-23 16:58:19,472 - startup     -  [INFO    ] - Setting up invertor: 1
2025-01-23 16:58:19,561 - startup     -  [INFO    ] - ==============================================================
2025-01-23 16:58:19,561 - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2025-01-23 16:58:19,561 - startup     -  [INFO    ] - ====     http://192.168.1.xxx:8099/config.html             ====
2025-01-23 16:58:19,561 - startup     -  [INFO    ] - ==============================================================
2025-01-23 16:58:19,562 - startup     -  [INFO    ] - Running Invertor 1 (CExxxx) read loop every 30/120s
2025-01-23 16:58:19,566 - startup     -  [INFO    ] - Serving Web Dashboard from port 3000
2025-01-23 16:58:19,586 - startup     -  [INFO    ] - Setting daily charge target forecast job to run at: 23:20
[2025-01-23 16:58:19 +0000] [70] [INFO] Starting gunicorn 23.0.0
[2025-01-23 16:58:19 +0000] [70] [INFO] Listening at: http://0.0.0.0:6350 (70)
[2025-01-23 16:58:19 +0000] [70] [INFO] Using worker: sync
[2025-01-23 16:58:19 +0000] [86] [INFO] Booting worker with pid: 86
[2025-01-23 16:58:19 +0000] [80] [INFO] Starting gunicorn 23.0.0
[2025-01-23 16:58:19 +0000] [80] [INFO] Listening at: http://0.0.0.0:6345 (80)
[2025-01-23 16:58:19 +0000] [80] [INFO] Using worker: sync
[2025-01-23 16:58:19 +0000] [87] [INFO] Booting worker with pid: 87
[2025-01-23 16:58:19 +0000] [88] [INFO] Booting worker with pid: 88
[2025-01-23 16:58:19 +0000] [89] [INFO] Booting worker with pid: 89
2025-01-23 16:58:21,445 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-01-23 16:58:21,445 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.1.xxxxx
2025-01-23 16:58:21,452 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-01-23 16:58:31,755 - GivTCP - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
2025-01-23 16:58:50,368 - GivTCP - read        -  [ERROR   ] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-23 16:59:21,472 - GivTCP - read        -  [ERROR   ] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-23 16:59:52,567 - GivTCP - read        -  [ERROR   ] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-23 17:00:23,673 - GivTCP - read        -  [ERROR   ] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-23 17:00:56,035 - GivTCP - read        -  [ERROR   ] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-23 17:01:27,132 - GivTCP - read        -  [ERROR   ] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)
2025-01-23 17:01:58,232 - GivTCP - read        -  [ERROR   ] - Error publishing data from pickle: ('KeyError', 'read.py', 1843)

gcoan, 2025-01-23:
you;ve tried most things, maybe do a 'reset to defaults' on the inverter as it may not be responding to modbus requests and this sometimes can clear it

dodger421, 2025-01-23:
Have you enabled InfluxDB in the web config? I got the same error when I tried to setup GivTCP to write to an InfluxDB instance both on the same host as HA and on an independent host.

Issue #294 also mentions this with a little more in depth debugging on the dictionary error in read.py.

TheWhiteDragonsRage, 2025-01-24:
> Have you enabled InfluxDB in the web config? I got the same error when I tried to setup GivTCP to write to an InfluxDB instance both on the same host as HA and on an independent host.
>
> Issue [...]

I have influx on another container and it's working fine for all my other ha sensors, all the details are on the config page correctly as far as I can see,  could it be problem feeding into the same db as HomeAssistant? Haven't tried pushing it to different db name...

TheWhiteDragonsRage, 2025-01-24:
> you;ve tried most things, maybe do a 'reset to defaults' on the inverter as it may not be responding to modbus requests and this sometimes can clear it

How do I "reset to default" on the inverter ? Not something I have tried ?

gcoan, 2025-01-24:
In the givenergy portal, go to inverters, then the TV symbol,  scroll across to settings, and its in there

dodger421, 2025-01-24:
> I have influx on another container and it's working fine for all my other ha sensors, all the details are on the config page correctly as far as I can see, could it be problem feeding into the same db as HomeAssistant? Haven't tried pushing it to different db name...

It looks like a fault with the specific GivTCP InfluxDB script.

If I disable InfluxDB in the GivTCP web config and include the HA sensors created by GivTCP in the same way I do my other HA sensors they write to InfluxDB fine and there are no errors in the GivTCP log.

kamenmackay, 2025-02-10:
I have the very same issue. I am still running influxdb1.8 so I'm using Telegraf's influxdb_v2 input filter which worked fine for the longest time but not so much recently. I will note that I recently upgraded Influxdb to 1.11 so perhaps there might be an issue there.

chrisckc, 2025-04-16:
I encountered the same issue with v3.1.6:
[ERROR   ] - Error publishing data from pickle: ('KeyError', 'read.py', 2148)

Looking at the source I spotted the issue, looks like a copy-paste error in the  publish(SN,data) function in the file influx.py

```
for key in flow_output:
            if not power_output[key] == None:
```
should be:
```
        for key in flow_output:
            if not flow_output[key] == None:
```

and same issue for the other for loops.

I forked the repo, fixed the issue and specified my fork in the docker compose file like this:

```
  givtcp:
    #image: britkat/giv_tcp-dev:latest
    build: https://github.com/chrisckc/giv_tcp.git#dev3_chrisckc
```

Feel free to use my fork (dev3_chrisckc branch) but it may not be there forever and may not pull in any new commits from here.

TheWhiteDragonsRage, 2025-04-27:
> givtcp:
>     #image: britkat/giv_tcp-dev:latest
>     build: https://github.com/chrisckc/giv_tcp.git#dev3_chrisckc

Thanks Chris, thats a great find, how do i implement this fix ? do you have a ID10-T guide i can use ?

mbridgett, 2025-06-09:
I have just stumbled over this same issue as well.  I can see the fix in github but nothing seems to have happened with it.  Does anyone know how long it normally takes for these things to be accepted/merged and rolled into the HA update?  Sorry for the dumb questions, this this is all new to me.

TheWhiteDragonsRage, 2025-06-15:
> I have just stumbled over this same issue as well. I can see the fix in github but nothing seems to have happened with it. Does anyone know how long it normally takes for these things to be accepted/merged and rolled into the HA update? Sorry for the dumb questions, this this is all new to me.

i asked much the same a while ago - there was a fix released, but no process on how to actually implement it for those of us who need a bit more help.