britkat1980/giv_tcp #338: Spurious data appearing in "GivTCP Energy Battery Charge Energy Total kWh"

BuhJuhWuh, 2025-01-26:
Hi,

I noticed an issue over the last few weeks where my Home Assistant energy dashboard has been showing strange values (more detail [here](https://community.givenergy.cloud/d/5384-spurious-givtcp-readings-in-ha-energy-dashboard), but the crux of the problem seems to be as follows).  I've checked through all the "GivTCP Energy Total kWh"-type sensor which previously worked fine to populate the energy dashboard, and all look fine except for "GivTCP Energy Battery Charge Energy Total kWh".  Now and then, instead of incrementing gradually upwards, it seems to jump to a random constant value for a few days before (sometime) jumping back again:

[image]

The regions roughly 1st-4th, 7th-9th, and 15th-19th Jan look normal, but otherwise it's just sitting on these constant values.

[image]

One previous instance in mid-Dec, but otherwise looks perfect since I first set it up in Feb '24.

I had a bit of a delve to see what other data sources I could check, as follows:

[image]

The values in red correspond to the current random constant value - it seems to be getting this from GivTCP via MQTT, but I couldn't find a value for this using the BBC Basic Inverter App.  Any ideas...?

Thanks,
Jo

BuhJuhWuh, 2025-01-27:
Another random jump up today...

[image]

poolski, 2025-01-28:
You can correct for those spikes by:
- Going to **Developer Tools**
- Clicking over to the **Statistics** tab
- Finding the sensor value (usually it's the `_total` one, not `_today`)
- Clicking this icon 
[image]

That'll bring up an interface for you to be able to manually adjust values for specific time buckets. If you hit the **Outliers** button, HA will try to find values which are _wildly_ out of whack which you can adjust. 

For example, if you find a value of `1248kWh` in the `_battery_charge_energy_total_kwh` sensor at midnight, that's clearly not going to be the case. If you don't have any forced charging enabled, that's doubly true - so you can zero that value. 

>[!NOTE]
>This isn't a **fix** for the underlying issue (which I've yet to work out myself (I'm not a contributor)) but it'll hopefully get your charts back under control

BuhJuhWuh, 2025-01-28:
Yes, thanks, aware of that method.  Unfortunately, as you hint at, that only gets us back to a flat line rather than the real data.
Cheers.

BuhJuhWuh, 2025-01-28:
And, just like that, we're back again... Nope, no idea.

[image]

BuhJuhWuh, 2025-02-03:
And broken again.  Is "GivTCP Energy Battery Charge Energy Total kWh" a property read directly from the inverter? (I'm thinking not, as it doesn't show up in BBC Basic Inverter app.)  If not, how is it calculated in GivTCP?  Something's not quite right there, anyway...

[image]

BuhJuhWuh, 2025-02-07:
I made a separate sensor which simply subtracts "GivTCP Energy Battery Disharge Energy Total kWh" from "GivTCP Energy Battery Throughput Energy Total kWh", and it appears to work perfectly.  So something's obviously going awry with the calculation of "GivTCP Energy Battery Charge Energy Total kWh" - could it be something as simple as an incorrect logic step or variable overflow...?
[image]

BuhJuhWuh, 2025-02-08:
A fellow givenergy forum user spotted some changes in the code, as described here: https://community.givenergy.cloud/d/5384/21
@britkat1980 please could you explain what this piece of code is doing, and where it is getting its numbers from, as it's behaving like a random number generator at present.
Thanks.

poolski, 2025-02-09:
Hey, @BuhJuhWuh, might be me, but your last message came across a bit forceful. @britkat1980 maintains and develops this project for free in their own time. The tone in your message read more like a support ticket for a piece of paid enterprise software.

BuhJuhWuh, 2025-02-09:
Fair point, you're right, it does. My sincere apologies.

A combination of excitement at finding a possible line of investigation,
impatience at the puzzle (not the product, not it's author, which are both
much appreciated!), and my usual habit of forgetting that poor attempts at
levity never translate well on a forum!

Any help or suggestions which might solve this one very much appreciated.

gcoan, 2025-02-11:
Looks like I have suffered the same problem with GivTCP 3.0.4

Noticed some absolutely massive battery charge and discharge values in the energy dashboard for 9th and 10th.  I get these occasionally usually after a HA restart, so zero'd these via developer tools/statistics and quick look, all was good.

But looking at the energy dashboard now I see absolutely no charging for the few days

And the sensor.givtcp_xxx_battery_charge_energy_total_kwh is all over the place:

[image]

on both inverters:

BuhJuhWuh, 2025-02-11:
Thanks for your input @gcoan.  The reversion to the correct value on restart is an interesting finding in itself; replicated on my system just now:

("Charge" in orange, "Throughput - Discharge" in blue.)

[image]

gcoan, 2025-02-11:
This could explain the pattern you have seen @BuhJuhWuh that the code goes off into the Boonies with weird values, and then seemingly at random resets itself to the correct value - these resets could be when you (or Predbat or HA) have restarted GivTCP

It was that there was no battery charging showing up on the energy dashboard that caused me to look for there being an issue.

Gonna have to rewrite the long term stats with the corrected data from the GivEnergy Portal - a how-to guide on this is coming soon in the GivEnergy community forum

gcoan, 2025-02-12:
Since restarting givtcp last night and the battery charge total sensor returning to it's correct value it has continued to accumulate yesterday and today 

So this appears to be a workaround 

I think I'll put in place an automation to alert me and restart givtcp if the sensor value goes above a set limit which should keep this bug in check 
[image]

gcoan, 2025-02-12:
@britkat1980 have you been able to look into this?

Tmp2k, 2025-02-15:
I also have this same issue. Specifically with 'Battery Charge Total Energy' and 'Battery Discharge Total Energy'. Both jumped to a huge value circa 13,000,000.  Restarting the addon fixed the issue.

wrighar-too, 2025-02-20:
I too have been getting this for month on the battery entities of charge or discharge..
A restart of GivTCP is my workaround, althoiugh Dev 3.0.50 and a few earlier seems to have been much better.
It seems to occur when this happens:  

[image]

[image]

[image]

gcoan, 2025-02-23:
I have another clue to the cause/symptoms of this issue...

Today upgraded my HA version, which restarted HA, but didn't restart any of the addons.  Afterwards I noticed in the Energy dashboard a classic entity jump:

[image]

Restarting GivTCP caused a reverse and balancing out jump:

[image]

The Energy dashboard shows 'sum' data from long term statistics, looking at the sensor state I saw the same pattern:
[image]

Its the sensor state jumping that causes the long term statistics sum to jump.

In the GivTCP log, it was all running fine until the point of the HA upgrade when presumably GivTCP lost communication to HA:

```
2025-02-23 13:30:30,062 - H - write       -  [INFO    ] - Setting Charge Slot 1 to: 14:00 - 16:00 was a success
2025-02-23 13:30:30,555 - H - write       -  [INFO    ] - Setting Charge Schedule to enable was a success
2025-02-23 13:30:31,042 - H - write       -  [INFO    ] - Setting Battery Pause Mode to Disabled was a success
2025-02-23 13:40:34,785 - H - mqtt        -  [INFO    ] - Requesting Control Action: setBatteryPauseMode - {'state': 'PauseCharge'}
2025-02-23 13:40:35,876 - H - write       -  [INFO    ] - Setting Battery Pause Mode to PauseCharge was a success
2025-02-23 13:45:08,409 - H - write       -  [INFO    ] - Setting Battery Pause Mode to Disabled was a success
2025-02-23 13:56:13,418 - H - base_events -  [ERROR   ] - Task exception was never retrieved
future: <Task finished name='network_consumer' coro=<Client._task_network_consumer() done, defined at /app/GivTCP_2/givenergy_modbus_async/client/client.py:329> exception=TimeoutError(110, 'Operation timed out')>
Traceback (most recent call last):
  File "/app/GivTCP_2/givenergy_modbus_async/client/client.py", line 386, in _task_network_producer
    await self.close()
  File "/app/GivTCP_2/givenergy_modbus_async/client/client.py", line 98, in close
    await self.writer.wait_closed()
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 358, in wait_closed
    await self._protocol._get_close_waiter(self)
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1005, in _read_ready__data_received
    data = self._sock.recv(self.max_size)
TimeoutError: [Errno 110] Operation timed out
2025-02-23 13:56:13,431 - H - base_events -  [ERROR   ] - Task exception was never retrieved
future: <Task finished name='network_producer' coro=<Client._task_network_producer() done, defined at /app/GivTCP_2/givenergy_modbus_async/client/client.py:373> exception=TimeoutError(110, 'Operation timed out')>
Traceback (most recent call last):
  File "/app/GivTCP_2/givenergy_modbus_async/client/client.py", line 386, in _task_network_producer
    await self.close()
  File "/app/GivTCP_2/givenergy_modbus_async/client/client.py", line 98, in close
    await self.writer.wait_closed()
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 358, in wait_closed
    await self._protocol._get_close_waiter(self)
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1005, in _read_ready__data_received
    data = self._sock.recv(self.max_size)
TimeoutError: [Errno 110] Operation timed out
```

"H base_events" is where the timeout error occurs, and it was an inverter H sensor state that got corrupted (in this case battery_charge_energy_total_kwh.  None of the other 'totoal' sensors got corrupted and neither did any of the 'total' sensors.  All of the inverter G sensors didn't get corrupted.

After this point when GivTCP reports a timeout error, the log then shows:

```
2025-02-23 14:00:07,905 - G - write       -  [INFO    ] - Setting battery charge rate 1900 was a success
2025-02-23 14:00:28,506 - G - write       -  [INFO    ] - Setting Battery Pause Mode to PauseDischarge was a success
2025-02-23 14:01:12,266 - startup     -  [ERROR   ] - Self Run loop process stuck. Killing and restarting...
2025-02-23 14:01:12,267 - startup     -  [INFO    ] - Restarting Invertor read loop every 30s
2025-02-23 14:01:12,839 - H - read        -  [INFO    ] - Starting watch_plant loop...
2025-02-23 14:01:12,839 - H - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.1.159
2025-02-23 14:01:12,847 - H - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-02-23 14:01:24,255 - H - read        -  [ERROR   ] - Battery Object empty so skipping
2025-02-23 14:01:24,716 - H - write       -  [INFO    ] - Setting Battery Pause Mode to PauseDischarge was a success
2025-02-23 14:01:25,198 - H - write       -  [INFO    ] - Setting Battery Pause Mode to PauseDischarge was a success
2025-02-23 14:01:25,847 - H - write       -  [INFO    ] - Setting Battery Pause Mode to PauseDischarge was a success
2025-02-23 14:01:56,944 - H - read        -  [ERROR   ] - Battery Object empty so skipping
2025-02-23 14:02:28,658 - H - read        -  [ERROR   ] - Battery Object empty so skipping
2025-02-23 14:03:00,440 - H - read        -  [ERROR   ] - Battery Object empty so skipping
2025-02-23 14:03:33,449 - H - read        -  [ERROR   ] - Battery Object empty so skipping
[... 6 lines cut ...]
And moreover it doesn't attempt any further restarts of itself

And this pattern of battery object errors continues until I restart GivTCP, the sensor state returns to normal and there's no further read errors from GivTCP:

```
[2025-02-23 20:02:32 +0000] [88] [INFO] Booting worker with pid: 88
[2025-02-23 20:02:32 +0000] [89] [INFO] Booting worker with pid: 89
[2025-02-23 20:02:32 +0000] [91] [INFO] Booting worker with pid: 91
[... cut ...]
2025-02-23 20:08:17,920 - H - mqtt        -  [INFO    ] - Requesting Control Action: setBatteryPauseMode - {'state': 'PauseCharge'}
2025-02-23 20:08:18,450 - H - write       -  [INFO    ] - Setting Battery Pause Mode to PauseCharge was a success
2025-02-23 20:10:06,892 - H - write       -  [INFO    ] - Setting Battery Pause Mode to Disabled was a success
2025-02-23 20:12:33,370 - H - mqtt        -  [INFO    ] - Requesting Control Action: setBatteryPauseMode - {'state': 'PauseCharge'}
2025-02-23 20:12:34,334 - H - write       -  [INFO    ] - Setting Battery Pause Mode to PauseCharge was a success
```

BuhJuhWuh, 2025-02-23:
Thanks. Any idea if/how this might correlate with Rubikcube's findings noted here, and his following comment? 
https://community.givenergy.cloud/d/5384-spurious-givtcp-readings-in-ha-energy-dashboard/21

BuhJuhWuh, 2025-02-24:
@gcoan interesting - I'm pretty sure most (if not all) of my jumps are unrelated to HA restarts, and it usually negotiates those without issue. And yes, like you, I can follow the logic to some extent, but not knowing what the various registers refer to in practice hobbles my understanding.

pbyrne-ms, 2025-02-27:
Hi Guys, I am having similar issues with my battery entities for "Today" charge and discharge Kwh (specifically they don't reset at exactly midnight so they foul up the stats for the next day), so I was looking into the suggestion above to use the "Total" entities instead. But when I check them, they are 'unavailable' and seem to have always been this way, there is no history recorded for them. I have the AIO model (latest firmware and bms as of Feb25) if this is relevant..and GivTCP 3.0.4.

These entities are 'unavailable'
givtcp_chxxx_battery_charge_energy_total_kwh
givtcp_chxxx_battery_discharge_energy_total_kwh

Any clues as to how to diagnose this or what might be wrong? Does anyone else have working versions of these when using an AIO?

Tmp2k, 2025-02-27:
Just an update on our setup...

I've created an automation that restarts the GivTcp addon whenever an abnormal value is detected e.g. > 1,000,000. This has stopped large amount of error accumulation, but there is still the odd blip, which massively throws the energy dashboard data out, making it unusable.

We're only seeing this on  'Battery Charge Total Energy' and 'Battery Discharge Total Energy' on the inverter device.

We've also been unable to use the daily totals as these reset 1 hour before/after midnight. This could be GMT Vs BST ?

I've now changed the energy dashboard to use the Battery Stack 1 entities instead of the inverter entities which are erroring.

So for now we have a work around, but it's not ideal.

pbyrne-ms, 2025-02-27:
> It may be something wrong with the migration fro GivTCP v2 to v3? Is there anything in the Home Assistant core log (when you restart HA) about duplicate entity id's? [this was a problem I had with my v3 upgrade, and is fixable with MQTT Explorer]

I think I've gotten a little further, thanks for the tip on looking via MQTT explorer! My own issue seems to be due to a combination of GivTCP3 and also the latest firmware the AIOs have to enable parallel units. I only have one at the moment, but the tag hierarchy seems to have been modified to accommodate multiple units.... I found these in MQTT and they seem to do what I need for now:

```
aio/CHxyz/Battery_Details/Battery_Stack_1/Stack_Charge_Energy_Today_kWh
aio/CHxyz/Battery_Details/Battery_Stack_1/Stack_Discharge_Energy_Today_kWh
```
&
```
aio/CHxyz/Battery_Details/Battery_Stack_1/Stack_Charge_Energy_Total_kWh
aio/CHxyz/Battery_Details/Battery_Stack_1/Stack_Discharge_Energy_Total_kWh
```

I assume if I get a second AIO unit I'll have ...Stack_2.... variants... I could add them up with utility meter helpers to get a total maybe...
The 'today' tags above still annoyingly don't reset to zero at exactly midnight, but the 'total' variants are working fine in my Battery Card...

gcoan, 2025-02-27:
These 'battery_stack' entries appeared in GivTCP3 for me as well, although with my Gen 1 hybrid they only have voltage and temperature entities.

Glad you are making progress, I sometimes feel I'm not.

Just finished off executing my procedure to restore 'total' charge entity data from 'today' entity data on both of my inverters on a 4 day period in early February, changing the energy dashboard from:

[image]

no battery charging at all (due to the spike) to:

[image]

and then would you believe it, I noticed battery charge total on one inverter spiked up yesterday, by 50kWh, and after restarting givtcp its come back down again.  Means I need to re-run the migration process AGAIN for another day of history.

Am wondering about putting in a daily restart of givtcp as @Tmp2k has, but this feels a real cludge.  Am wondering about whether its worth trying either the beta or dev version of GivTCP3.  Has anyone done this, does it spike less?

johnwb87, 2025-03-28:
I've started experiencing this issue in the last few weeks. SPike on battery charge that is resolved with a restart.
Has the cause been identified?

johnwb87, 2025-04-14:
What's going on with this issue? It is still occuring. Solution to #352 was to use the total sensor and now that is not reliable. I noticed a note in the recent release about changes to address midnight reset, but since updating I've seen no difference in the energy dashboard when using the today sensor and still experiencing the mis-reads on total.

leocrawford, 2025-04-28:
Ditto. I'm running beta, and the _today sensors don't work because of the 20s delay, and the _total discharge sensor (not charge) is giving me a rectangular wave. The lower values (around 600kwh) appear right and do slowly increase. The larger values are wrong.

[image]

gcoan, 2025-04-28:
@leocrawford @johnwb87 what givenergy kit are you running?

I have two gen 1 hybrid inverters and am currently running the latest version of the production code and I haven't seen an example of the total spiking for about 2 months now

Can you identify any correlation between the spikes upwards and the givtcp, HA and mosquitto logs?  I saw evidence of the spikes upwards being when mosquitto reported a packet error https://github.com/britkat1980/giv_tcp/issues/338#issuecomment-2679285837 and the spikes downwards were when givtcp was restarted

leocrawford, 2025-04-28:
I'm running a GivEnergy Hybrid Inverter V3 with a single battery. Details below. No obvious correlation but will look properly through logs tomorrow. 

Inverter Model
GIV-HY5.0
Software Version
D0.315-A0.315
Battery Type
Lithium-ion
Battery Capacity
186 Ah | 9.52 kWh | (1 Pack)
Last Software Update
2025-04-01 22:50:37
Last Update Time
2025-04-01 22:34:02
Battery Firmware Version
4007

I will update the battery firmware now, as I see it needs an update.

leocrawford, 2025-04-30:
Still analysing logs, but I do have hundreds of repeats of this:

`Invalid option for select.givtcp_fd2320g056_battery_calibration: 'disable' (valid options: ['Off', 'Start', '', 'Charge Only'])`

I can also see many of my states turning to 'unknown' for 10s a time, sometimes several times an hour and sometimes not for eight hours in a row.

gcoan, 2025-04-30:
> I can also see many of my states turning to 'unknown' for 10s a time, sometimes several times an hour and sometimes not for eight hours in a row.

That is very likely to be a communications issue between givtcp and your inverter.  I got the network submask wrong on my Home Assistant server and that's what I saw, sensors repeatedly going to unknown.
Quite a few timeout errors in the givtcp log as well.

leocrawford, 2025-04-30:
> That is very likely to be a communications issue between givtcp and your inverter. I got the network submask wrong on my Home Assistant server and that's what I saw, sensors repeatedly going to [...]

Is that the IP setting within the GivTCP config file? I've checked that and it's right? Can't see a particular subnet setting though.

I'll run some network tests to see if there are occasional link problems.

leocrawford, 2025-04-30:
No correlation in logs, but do have hundreds of..

```
2025-04-29 20:31:09,091 - read - [CRITICAL] - Detecting inverter characteristics...
2025-04-29 20:31:09,094 - read - [ERROR] - Error in inital detect/refresh: ('AttributeError', 'read.py', 77)
2025-04-29 20:31:09,095 - read - [ERROR] - Error in self_run. Re-running watch_plant: ('AttributeError', 'read.py', 2191)
2025-04-29 20:31:11,100 - read - [INFO] - Starting watch_plant loop...
```

leocrawford, 2025-04-30:
> I too have been getting this for month on the battery entities of charge or discharge.. A restart of GivTCP is my workaround, althoiugh Dev 3.0.50 and a few earlier seems to have been much better. It seems to occur when this happens:

It's interesting that both our anomalous readings are 2261.6

wrighar-too, 2025-08-04:
My issue went away when I really passed 2261kWh....

[image]

stikonas, 2025-08-21:
> What's going on with this issue? It is still occuring. Solution to [#352](https://github.com/britkat1980/giv_tcp/issues/352) was to use the total sensor and now that is not reliable.

Utility meter can be used to handle this. It can be configured to handle daily resets (disable net meter option), so you can use daily charge/discharge sensors until total sensor passes 2261 kWh.

In fact you should use utility meters even with properly working total consumption to abstract away underlying sensor. E.g. if you don't want to lose your energy history in case of inverter replacement.

gcoan, 2025-08-21:
> In fact you should use utility meters even with properly working total consumption to abstract away underlying sensor. E.g. if you don't want to lose your energy history in case of inverter replacement.

I started off with utility meters to abstract away from the underlying sensors, in part for the reason you suggest (inverter replacement).

But I have now removed all that and now directly use the underlying total sensors.

There's a number of reasons why I've done this:
- its simpler and less storage required in the HA database
- utility meters are great except when they're not. If for any reason your HA isn't working properly or isn't running over midnight, the utility meter doesn't reset as is should do
- if you know what you're doing, moving entity long term statistics history from one entity to another or correcting history isn't too difficult

But the big reason is that the original premise for the utility meters was actually false.  I had one of my inverters replaced under warranty because the battery control board failed, and when the new inverter was installed the givenergy engineer simply assigned my old serial number to it so all the givtcp sensors and the givenergy portal just continued unchanged.

gcoan, 2025-08-21:
> Hmm, interesting, I've had the opposite with battery replacements under warranty - the new unit has a new serial number, so have to recreate/reassign all the sensors.

The battery has a new serial number, yes. My experience was that the inverter that they can copy the serial number over. Meant that the system didn't have to be recommissioned as well

stikonas, 2025-09-19:
> There are still regular problems with the Charge and Discharge total entities. Are others still finding this?

Not 100% if that was caused by it, but I had a wrong timezone (UTC instead of Europe/London) inside givtcp docker container. After fixing it I don't see any more spurious data, though I need to run it for a bit longer. So far it has been 5 days without those 2261 kWh values.

stikonas, 2025-10-18:
So on my system those spurious 2261 kWh values correlate with the following error in the logs:

`2025-10-18 01:46:23,506 - GivTCP - read        -  [ERROR   ] - Battery Object empty so skipping`

poolski, 2025-11-13:
> * if you know what you're doing, moving entity long term statistics history from one entity to another or correcting history isn't too difficult

Could you elaborate on your process?

gcoan, 2025-12-02:
Back to the original issue of givtcp randomly spiking for no apparent reason, which for me still occurs occasionally, the last occurrence I had was on 21st September (which I spotted and fixed on the 22nd), and then again on the 24th which I spotted much quicker,  but since then, no occurrences.

I have written an automation to detect if a total sensor jumps by more than 10, indicating this spike issue.  At the moment it just sends an alert to my mobile and writes to the logfile, but could easily be extended to restart givtcp which is the fix to bring these spikes back down again.

```
alias: GivTCP sensor jump alert
description: ""
triggers:
  - entity_id: sensor.givtcp_xxx_battery_charge_energy_total_kwh
    trigger: state
    to: null
    variables:
      trigger_name: G Battery Charge Total
  - entity_id: sensor.givtcp_xxx_battery_discharge_energy_total_kwh
    trigger: state
    to: null
    variables:
      trigger_name: G Battery Discharge Total
  - entity_id: sensor.givtcp_xxx_import_energy_total_kwh
    trigger: state
    to: null
    variables:
      trigger_name: G Import Energy Total
  - entity_id: sensor.givtcp_xxx_export_energy_total_kwh
    trigger: state
    to: null
    variables:
      trigger_name: G Export Energy Total
  - entity_id: sensor.givtcp_xxx_pv_energy_total_kwh
    trigger: state
    to: null
    variables:
      trigger_name: G PV Energy Total
actions:
  - variables:
      delta: >-
        {{ ((trigger.to_state.state | float(0)) - ( trigger.from_state.state |
        float(0))) | abs }}
  - condition: template
    value_template: "{{ delta > 10 }}"
  - action: system_log.write
    metadata: {}
    data:
      level: warning
      message: "{{ trigger_name }} changed by {{ delta }}"
  - action: script.notify_all_devices
    metadata: {}
    data:
      title: "ISSUE: GivTCP sensor value jump"
      url: " "
      message: >-
        {{ trigger_name }} has changed from {{ trigger.from_state.state }} to {{
        trigger.to_state.state }}
      critical: "Y"
trace:
  stored_traces: 20
mode: single
```

gcoan, 2026-01-20:
I wouldn't worry about the throughput total not matching the charge and discharge totals, this is almost certainly caused by the charge and discharge totals wrapping round to zero when they get to 6550 because the inverter register is a 16 bit unsigned number.  This has happened to me a month ago on my charge total and the discharge total will roll round in another month or so.

The charge total flipping between different values sounds like the same issue.  It occurs very infrequently for me, but @BuhJuhWuh who initially raised this, it seemed to happen more often.

Only cure seems to be an automation to detect it and restart givtcp.  If yours is happening often then maybe setting debug logs on will give a clue as to the cause of it?

steve-daedilus, 2026-01-20:
It also detected itself as a hybrid, probably because of the CT to the solar production? Might have to tweak some configuration.

britkat1980, 2026-10-11:
Closing this as it's had no activity for over 8 weeks.

GivTCP 3.6 has since been released ([latest: v3.6.2](https://github.com/britkat1980/giv_tcp/releases/latest)). It moves to a new Modbus library and includes a large number of fixes, so this may well behave differently now. If you still see this on 3.6.2, please reopen it, or open a new issue, with a debug log.

closed 2026-10-11 by britkat1980

hollytreecook, 2026-10-29:
Since 12th October my HomeAssistant dashboard has shown wrong data for my inverter and battery.
All sensors have flatlined.
The GivTCP dashboard in HomeAssistant shows numbers that look correct (PV power etc), but those same sensors in MQTT and GivTCP are all zero.
I have uninstalled and reinstalled MQTT and GivTCP etc. I have deleted .pkl files in HA/GivTCP. I have cleared out MQTT topics using MQTT explorer. I am on GivTCP 3.6.7
Does this mean there could be something wrong with my inverter registers?(dont really know what I am talking about)
Any ideas please?
Tried 3.6.8 but that made no difference .