britkat1980/giv_tcp #339: [ERROR   ] - Setting Charge Rate failed: ('ZeroDivisionError', 'write.py', 447)

DrJohnM61, 2025-01-30:
I have started to get the following message in the log files. Overnight charging stops and Predbat fails. Requires restart of GIVTCP and Predbat.

```
2025-01-28 08:53:00,137 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-01-28 08:53:00,167 - startup     -  [INFO    ] - Network Found: 192.168.10.1/24
2025-01-28 08:53:02,168 - startup     -  [INFO    ] - Searching for Inverters
2025-01-28 08:53:02,170 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-01-28 08:53:04,394 - startup     -  [INFO    ] - INV- Scanning network (1):192.168.10.1/24
2025-01-28 08:53:09,043 - client      -  [INFO    ] - Connection established to 192.168.10.87:8899
2025-01-28 08:53:09,043 - client      -  [INFO    ] - Detecting plant
2025-01-28 08:53:10,208 - client      -  [INFO    ] - Plant Detected
2025-01-28 08:53:10,209 - startup     -  [INFO    ] - Inverter GW2347G062 which is a Gen1 - Gateway with 0 batteries has been found at: 192.168.10.87
2025-01-28 08:53:10,210 - client      -  [INFO    ] - Connection established to 192.168.10.171:8899
2025-01-28 08:53:10,210 - client      -  [INFO    ] - Detecting plant
2025-01-28 08:53:11,357 - client      -  [INFO    ] - Plant Detected
2025-01-28 08:53:16,389 - startup     -  [INFO    ] - Inverter CH2330G417 which is a Gen1 - All_in_one with 4 batteries has been found at: 192.168.10.171
2025-01-28 08:53:16,390 - client      -  [INFO    ] - Connection established to 192.168.10.189:8899
2025-01-28 08:53:16,390 - client      -  [INFO    ] - Detecting plant
2025-01-28 08:53:17,539 - client      -  [INFO    ] - Plant Detected
2025-01-28 08:53:24,931 - startup     -  [INFO    ] - Inverter CH2423G315 which is a Gen1 - All_in_one with 4 batteries has been found at: 192.168.10.189
2025-01-28 08:53:24,937 - startup     -  [INFO    ] - Setting up invertor: 1
2025-01-28 08:53:24,981 - startup     -  [INFO    ] - ==============================================================
2025-01-28 08:53:24,981 - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2025-01-28 08:53:24,981 - startup     -  [INFO    ] - ====     http://192.168.10.35:8099/config.html             ====
2025-01-28 08:53:24,981 - startup     -  [INFO    ] - ==============================================================
2025-01-28 08:53:24,982 - startup     -  [INFO    ] - Running Invertor 1 (GW2347G062) read loop every 30/120s
2025-01-28 08:53:24,986 - startup     -  [INFO    ] - Setting up invertor: 2
2025-01-28 08:53:25,056 - startup     -  [INFO    ] - ==============================================================
2025-01-28 08:53:25,062 - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2025-01-28 08:53:25,062 - startup     -  [INFO    ] - ====     http://192.168.10.35:8099/config.html             ====
2025-01-28 08:53:25,062 - startup     -  [INFO    ] - ==============================================================
2025-01-28 08:53:25,066 - startup     -  [INFO    ] - Running Invertor 2 (CH2330G417) read loop every 30/120s
2025-01-28 08:53:25,070 - startup     -  [INFO    ] - Setting up invertor: 3
2025-01-28 08:53:25,137 - startup     -  [INFO    ] - ==============================================================
2025-01-28 08:53:25,142 - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2025-01-28 08:53:25,142 - startup     -  [INFO    ] - ====     http://192.168.10.35:8099/config.html             ====
2025-01-28 08:53:25,143 - startup     -  [INFO    ] - ==============================================================
2025-01-28 08:53:25,150 - startup     -  [INFO    ] - Running Invertor 3 (CH2423G315) read loop every 30/120s
[2025-01-28 08:53:25 +0000] [70] [INFO] Starting gunicorn 23.0.0
[2025-01-28 08:53:25 +0000] [70] [INFO] Listening at: http://0.0.0.0:6350 (70)
[2025-01-28 08:53:25 +0000] [70] [INFO] Using worker: sync
[2025-01-28 08:53:25 +0000] [88] [INFO] Booting worker with pid: 88
[2025-01-28 08:53:25 +0000] [81] [INFO] Starting gunicorn 23.0.0
[2025-01-28 08:53:25 +0000] [81] [INFO] Listening at: http://0.0.0.0:6345 (81)
[2025-01-28 08:53:25 +0000] [81] [INFO] Using worker: sync
[2025-01-28 08:53:25 +0000] [89] [INFO] Booting worker with pid: 89
[2025-01-28 08:53:25 +0000] [90] [INFO] Booting worker with pid: 90
[2025-01-28 08:53:25 +0000] [91] [INFO] Booting worker with pid: 91
[2025-01-28 08:53:25 +0000] [84] [INFO] Starting gunicorn 23.0.0
[2025-01-28 08:53:25 +0000] [84] [INFO] Listening at: http://0.0.0.0:6346 (84)
[2025-01-28 08:53:25 +0000] [84] [INFO] Using worker: sync
[2025-01-28 08:53:25 +0000] [92] [INFO] Booting worker with pid: 92
[2025-01-28 08:53:25 +0000] [94] [INFO] Booting worker with pid: 94
[2025-01-28 08:53:26 +0000] [95] [INFO] Booting worker with pid: 95
[2025-01-28 08:53:26 +0000] [87] [INFO] Starting gunicorn 23.0.0
[2025-01-28 08:53:26 +0000] [87] [INFO] Listening at: http://0.0.0.0:6347 (87)
[2025-01-28 08:53:26 +0000] [87] [INFO] Using worker: sync
[2025-01-28 08:53:26 +0000] [97] [INFO] Booting worker with pid: 97
[2025-01-28 08:53:26 +0000] [98] [INFO] Booting worker with pid: 98
[2025-01-28 08:53:26 +0000] [101] [INFO] Booting worker with pid: 101
2025-01-28 08:53:27,294 - GIVTCPgw - read        -  [INFO    ] - Starting watch_plant loop...
2025-01-28 08:53:27,295 - GIVTCPgw - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.10.87
2025-01-28 08:53:27,302 - GIVTCPgw - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-01-28 08:53:27,750 - GIVTCPaio1 - read        -  [INFO    ] - Starting watch_plant loop...
2025-01-28 08:53:27,750 - GIVTCPaio1 - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.10.171
2025-01-28 08:53:27,752 - GIVTCPaio1 - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-01-28 08:53:28,096 - GIVTCPaio2 - read        -  [INFO    ] - Starting watch_plant loop...
2025-01-28 08:53:28,097 - GIVTCPaio2 - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.10.189
2025-01-28 08:53:28,099 - GIVTCPaio2 - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-01-28 08:53:33,997 - GIVTCPgw - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
[... 16 lines cut ...]
2025-01-29 00:51:25,844 - GIVTCPgw - write       -  [ERROR   ] - Setting Charge Rate failed: ('ZeroDivisionError', 'write.py', 447)
2025-01-29 00:51:41,365 - GIVTCPgw - write       -  [ERROR   ] - Setting Charge Rate failed: ('ZeroDivisionError', 'write.py', 447)
2025-01-29 00:51:52,085 - GIVTCPgw - write       -  [INFO    ] - Setting Charge Slot 1 to: 23:30 - 01:00 was a success
2025-01-29 00:51:52,392 - GIVTCPgw - write       -  [ERROR   ] - Setting Discharge Rate failed: ('ZeroDivisionError', 'write.py', 512)
2025-01-29 00:52:07,915 - GIVTCPgw - write       -  [ERROR   ] - Setting Discharge Rate failed: ('ZeroDivisionError', 'write.py', 512)
2025-01-29 00:52:23,240 - GIVTCPgw - write       -  [ERROR   ] - Setting Discharge Rate failed: ('ZeroDivisionError', 'write.py', 512)
2025-01-29 00:55:03,953 - GIVTCPgw - write       -  [ERROR   ] - Setting Charge Rate failed: ('ZeroDivisionError', 'write.py', 447)
2025-01-29 00:55:19,277 - GIVTCPgw - write       -  [ERROR   ] - Setting Charge Rate failed: ('ZeroDivisionError', 'write.py', 447)
2025-01-29 00:55:34,792 - GIVTCPgw - write       -  [ERROR   ] - Setting Charge Rate failed: ('ZeroDivisionError', 'write.py', 447)
2025-01-29 00:55:35,098 - GIVTCPgw - write       -  [ERROR   ] - Setting Discharge Rate failed: ('ZeroDivisionError', 'write.py', 512)
2025-01-29 00:55:50,423 - GIVTCPgw - write       -  [ERROR   ] - Setting Discharge Rate failed: ('ZeroDivisionError', 'write.py', 512)
2025-01-29 00:56:05,951 - GIVTCPgw - write       -  [ERROR   ] - Setting Discharge Rate failed: ('ZeroDivisionError', 'write.py', 512)
2025-01-29 01:00:04,274 - GIVTCPgw - write       -  [INFO    ] - Setting Charge Schedule to disable was a success
2025-01-29 01:00:04,714 - GIVTCPgw - write       -  [INFO    ] - Setting Battery Pause Mode to Disabled was a success
2025-01-29 01:00:05,020 - GIVTCPgw - write       -  [ERROR   ] - Setting Discharge Rate failed: ('ZeroDivisionError', 'write.py', 512)
2025-01-29 01:00:20,346 - GIVTCPgw - write       -  [ERROR   ] - Setting Discharge Rate failed: ('ZeroDivisionError', 'write.py', 512)
2025-01-29 01:00:35,872 - GIVTCPgw - write       -  [ERROR   ] - Setting Discharge Rate failed: ('ZeroDivisionError', 'write.py', 512)
```
While there are a couple successful writes, the rest of the log until restarting is only ZeroDivisionError messages.

Running GivTCP Current version: 3.0.4 on a HAOS Pi5.

All three devices are connected using ethernet.

Only configuration change of note is that the parallel AIO was installed a few weeks ago and influxDB was installed on the HAOS aboiut two weeks ago.

DrJohnM61, 2025-03-02:
I am continuing to receive this error and can only be fixed with a manual restart:
`2025-03-02 00:00:43,015 - GIVTCPaio2 - read        -  [INFO    ] - Midnight, so resetting Day/Night stats...
2025-03-02 00:00:51,743 - GIVTCPaio1 - read        -  [INFO    ] - Midnight, so resetting Day/Night stats...
2025-03-02 04:05:30,211 - GIVTCPgw - write       -  [INFO    ] - Setting Charge Slot 1 to: 23:30 - 07:30 was a success
2025-03-02 04:20:03,592 - GIVTCPgw - write       -  [ERROR   ] - Setting Charge Rate failed: ('ZeroDivisionError', 'write.py', 447)
2025-03-02 04:20:20,105 - GIVTCPgw - write       -  [ERROR   ] - Setting Charge Rate failed: ('ZeroDivisionError', 'write.py', 447)
2025-03-02 04:20:35,431 - GIVTCPgw - write       -  [ERROR   ] - Setting Charge Rate failed: ('ZeroDivisionError', 'write.py', 447)
2025-03-02 04:20:35,738 - GIVTCPgw - write       -  [ERROR   ] - Setting Discharge Rate failed: ('ZeroDivisionError', 'write.py', 512) `

gcoan, 2025-03-02:
> influxDB was installed on the HAOS aboiut two weeks ago.

I think I have seen others reporting issues with influx db and GivTCP v3.  Have a look, but might be worth you disabling influx and see if this at least stops the error

DrJohnM61, 2025-03-02:
> > influxDB was installed on the HAOS aboiut two weeks ago.
> 
> I think I have seen others reporting issues with influx db and GivTCP v3.  Have a look, but might be worth you disabling influx and see if this at least stops the error

I actually don't have the influx DB setting turned on for GivTCP, so it should not be that.

gcoan, 2025-03-02:
sorry I have no real idea what's causing this.  I know of other predbat/givtcp v3 users with multiple AIO's that haven't reported this issue at all

only thing I can suggest is to do a reset to defaults of the gateway and see if that cures anything?

Otherwise turn debug mode on in givtcp and attach the log for @britkat1980 to look at

sorry