britkat1980/giv_tcp #341: No support for single phase HV inverter

Texelo, 2025-02-11:
Recently had the 10kw HV 1ph inverter installed, but GivTCP doesn't seem to detect it properly. 
It is auto detected as a Gen 1 - All in One, and then times out trying to connect on the same IP.

```
2025-02-11 22:09:02,373 - startup - [CRITICAL] - HA MQTT Service has been found at core-mosquitto
2025-02-11 22:09:02,383 - startup - [INFO] - Supervisor Timezone: Europe/London
2025-02-11 22:09:04,397 - startup - [CRITICAL] - Scanning network for GivEnergy Devices...
2025-02-11 22:09:08,223 - startup - [CRITICAL] - Inverter FH2442G134 which is a Gen 1 - All in One has been found at: 192.168.0.187
2025-02-11 22:09:08,224 - startup - [INFO] - Searching for Inverters again
2025-02-11 22:09:08,240 - startup - [CRITICAL] - Running Redis
2025-02-11 22:09:08,255 - startup - [CRITICAL] - Setting up invertor: 1 of 1
2025/02/11 22:09:08 [notice] 69#69: using the "epoll" event method
2025/02/11 22:09:08 [notice] 69#69: nginx/1.20.2
2025/02/11 22:09:08 [notice] 69#69: OS: Linux 6.6.62-haos-raspi
2025/02/11 22:09:08 [notice] 69#69: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2025/02/11 22:09:08 [notice] 69#69: start worker processes
2025/02/11 22:09:08 [notice] 69#69: start worker process 74
2025/02/11 22:09:08 [notice] 69#69: start worker process 75
2025/02/11 22:09:08 [notice] 69#69: start worker process 76
2025/02/11 22:09:08 [notice] 69#69: start worker process 77
2025-02-11 22:09:08,306 - startup - [CRITICAL] - Running RQ worker to queue and process givernergy-modbus calls
2025-02-11 22:09:08,306 - startup - [CRITICAL] - Running Invertor (192.168.0.187) read loop every 15s
2025-02-11 22:09:08,312 - startup - [CRITICAL] - Subscribing MQTT Broker for control
2025-02-11 22:09:08,320 - startup - [CRITICAL] - Starting Gunicorn on port 6345
[2025-02-11 22:09:09 +0000] [81] [INFO] Starting gunicorn 23.0.0
[2025-02-11 22:09:09 +0000] [81] [INFO] Listening at: http://0.0.0.0:6345 (81)
[2025-02-11 22:09:09 +0000] [81] [INFO] Using worker: sync
[2025-02-11 22:09:09 +0000] [82] [INFO] Booting worker with pid: 82
[2025-02-11 22:09:09 +0000] [83] [INFO] Booting worker with pid: 83
[2025-02-11 22:09:09 +0000] [84] [INFO] Booting worker with pid: 84
2025-02-11 22:09:11,427 - Inv1 - mqtt_client -  [CRITICAL] - Connecting to MQTT broker for control- core-mosquitto
2025-02-11 22:09:16,291 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.0.187, 8899) failed: timed out
2025-02-11 22:09:18,795 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.0.187, 8899) failed: timed out
2025-02-11 22:09:21,299 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.0.187, 8899) failed: timed out
2025-02-11 22:09:23,803 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.0.187, 8899) failed: timed out
2025-02-11 22:09:26,307 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.0.187, 8899) failed: timed out
2025-02-11 22:09:29,490 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.0.187, 8899) failed: timed out
2025-02-11 22:09:31,995 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.0.187, 8899) failed: timed out
```

gcoan, 2025-02-12:
Have you tried the dev version of GivTCP, that contains the latest codebase so might be compatible?

Texelo, 2025-02-12:
Looking at the code, it may have been updated yesterday to support this, how do I go about switching to the dev version?

Texelo, 2025-02-12:
I've just tried the dev version, which looks like it's working on this, gets stuck in detecting plant loop. Happy to help test this with my hardware.
```
2025-02-12 01:03:03,062 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-02-12 01:03:03,099 - startup     -  [INFO    ] - Network Found: 192.168.0.249/24
2025-02-12 01:03:05,102 - startup     -  [INFO    ] - Searching for Inverters
2025-02-12 01:03:05,103 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-02-12 01:03:09,917 - client      -  [INFO    ] - Connection established to 192.168.0.187:8899
2025-02-12 01:03:09,918 - client      -  [INFO    ] - Detecting plant
2025-02-12 01:03:49,492 - client      -  [INFO    ] - Connection established to 192.168.0.187:8899
2025-02-12 01:03:49,493 - client      -  [INFO    ] - Detecting plant
```

gcoan, 2025-02-12:
Great to hear it's working, trail blazing with the new kit !