britkat1980/giv_tcp #343: [ERROR] - Error getting data from cache: ('AttributeError', 'read.py', 2137)

mpartington, 2025-02-17:
Error started out of the blue

Running AC3 on 209 firmware with 2x5.2 batteries

Get this error when ever I make a http://192.168.68.128:6345/runAll call in the browser.  Tried completely reinstalling, but same on Prod, beta and dev versions of V3

{"Result":"Error, no data available"}

log with debug attached

[log_inv_1.log](https://github.com/user-attachments/files/18832079/log_inv_1.log)

mpartington, 2025-02-17:
Fixed by turning the new REAL TIME CONTROL off on the portal.  The runall then worked, so I turned it back on. All still seems OK.

Will leave issue open in case GivTCP needs some refinement for this new firmware feature

73ttv627re77, 2025-03-19:
get the same but nothing fixes it, clearing everything from mqtt explorer after removal of the add-on and with RTC disabled in the portal still shows the same error message, any solutions? (gen2 hybrid)

cdrajb, 2025-08-21:
I just had the same problem - fixed as above by toggling RTC in the portal

britkat1980, 2025-08-26:
Yes, debug logs might help, that call is working for me

britkat1980, 2025-08-26:
So does this happen only on REST calls, or is happening all the time and only when RTC is enabled?

The error appears to suggest its with a data point related to a time.

gcoan, 2025-08-26:
> So does this happen only on REST calls, or is happening all the time and only when RTC is enabled?

It happens only when REST calls are made (from Predbat).  If I disable REST in predbat and instead use the GivTCP HA entities then I don't get this error at all.

RTC is disabled for me.  I have tried enabling it and disabling it, both in the GivEnergy portal and via the givtcp control, because others have suggested that this stopped the error occurring for them, and it has no impact/improvement for me, every REST call gives an error (note its a different error now with v3.2, was getting the read.py error with GivTCP beta 3.1.9).

It happens on both of my Gen 1 hybrid inverters. They are running firmware A0.187-D0.187 (which is the version with RTC capability).  It never used to error like this on REST so whether something has corrupted in the inverter or its as a result of the firmware version, I don't know.

apone2000, 2025-09-20:
I have RTC enabled (ac3) and get the same 

2025-09-20 02:21:06,639 - GivTCP - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7fdf611cb0c0>)
2025-09-20 02:21:06,650 - GivTCP - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7fdf6120f940>)
2025-09-20 02:21:06,699 - GivTCP - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7fdf613b0380>)
2025-09-20 02:21:06,723 - GivTCP - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7fdf61299a00>)
2025-09-20 03:45:05,254 - GivTCP - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7fdf6124f800>)
2025-09-20 03:45:05,265 - GivTCP - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7fdf611fbf40>)
2025-09-20 03:45:05,319 - GivTCP - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7fdf611ffe80>)
2025-09-20 03:45:05,330 - GivTCP - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7fdf6134b2c0>)

It seems for this one example to be via REST initially

(predbat)
2025-09-20 02:21:06.730518: Warn: Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled
2025-09-20 02:21:06.758663: Warn: record_status Inverter 0 read bad REST data from http://homeassistant.local:6345/runAll - REST will be disabled

im assuming this is a problem in the read.py call to read the data, would it help to check for None  in validateTImeslot  from 

`def validateTimeslot(slot,key,multi_output_old):
    if slot:
        output = slot.isoformat()
    elif multi_output_old:
        logger.debug("Suprious Timeslot data: using last good data")
        output=multi_output_old['Timeslots'][key]
    else:
        logger.debug("Suprious Timeslot data: setting to Midnight")
        output="00:00:00"
    return output`

to this ?

`def validateTimeslot(slot, key, multi_output_old):
    if slot is not None:
        return slot.isoformat()
    elif multi_output_old:
        logger.debug("Spurious Timeslot data: using last good data")
        return multi_output_old['Timeslots'].get(key, "00:00:00")
    else:
        logger.debug("Spurious Timeslot data: setting to Midnight")
        return "00:00:00"`

that might then stop predbat getting a failed rest call and restarting everything?

i might try turning RTC off for a bit as i get a lot of cache read errors and time format errors still.

gcoan, 2025-09-20:
> that might then stop predbat getting a failed rest call and restarting everything?

I have ended up turning off REST calls from predbat and rely on the HA entities from givtcp instead.  This works quite well although is slower than rest.

BTW I have commented out the predbat restart of givtcp, I find it is too aggressive in restarting givtcp and with givtcp taking a few minutes to get going you can end up in a race condition that predbat keeps on restarting givtcp because its not responding fast enough.
Instead I rely on the givtcp internal error restarting and an external automation (in the predbat docs) to detect if givtcp has stalled.  This works well.

Are you able to try changing the givtcp code as you suggested above and see if it fixes the REST issue we are getting?

gcoan, 2025-10-19:
I can report that with v3.3 installed just now I have been able to re-enable REST calls from Predbat and they are working perfectly !

britkat1980, 2026-10-11:
Closing this as it's had no activity for over 8 weeks.

GivTCP 3.6 has since been released ([latest: v3.6.2](https://github.com/britkat1980/giv_tcp/releases/latest)). It moves to a new Modbus library and includes a large number of fixes, so this may well behave differently now. If you still see this on 3.6.2, please reopen it, or open a new issue, with a debug log.

closed 2026-10-11 by britkat1980