britkat1980/giv_tcp #344: 2xAIO - wrong battery data reported while charging or discharging from grid

PianSom, 2025-02-18:
I had a second AIO installed a few days ago, and have been hesitant about posting this (as it seems just to be me?) but have run out of ideas.

I am using GivTCP 3.04 in a Docker container. It feeds into another container running a MQTT broker, which in turn is subscribed to by a HA container. I have cleaned out all retained GivTCP retained messages from the broker.

Everything seems to be working just fine - except when the system is either charging from or discharging to the grid. It's ok when charging from PV. When grid charging/discharging I am finding thet GivTCP is posting incorrect (static, or seemingly random) entries to Battery_power. See this screenshot of a recent Saving Session when an export of 10kW was going to the grid, but not - apparently - coming from the battery.

[image]

At the same time the MQTT entries for one of the AIO units show a flow to the grid, though shows the same erroneous battery power:

[image]

Here (from a different day) is a typical power chart. It is clear that during a night-time charge-discharge cycle on the left the battery flows are wrong - they bear no relation to the SoC reported. But on the right, during the day, all looks normal.

[image]

I am at a loss. Is this a GivTCP bug? What other information can I provide?

PianSom, 2025-02-19:
So, to add to Paul's comment, after a suggestion by hoggy I amended [this](https://github.com/britkat1980/giv_tcp/blob/2ef5d6b27c3ecde930ab9ffe803b7718d73b13e7/GivTCP/GivLUT.py#L76) to 13000 (and the subsequent negative to -13000). On further testing I found that this was not, in fact, the issue, as the problem persists.

However, I did find that when I forced a charge (so grid flow of 10kW+++) the battery sensor stops updating in MQTT (and hence HA). When the power flow stopped the MQTT started updating again.  This, at least, explains why the values I am seeing on high power flow appear to be random.

britkat1980, 2025-03-03:
I've modified limits to 13kw. However doesn't explain the issue seen here. Is it only the battery charge power sensor that stops reporting, or all of it?

PianSom, 2025-03-03:
The Gateway power sensor does not appear to report any flow (+ or -) over 6k

As a work around I have created templates of AIO1 + AIO2 charge power and AIO1 + AIO2 discharge power. The charge power works just fine - both batteries will report charge of 6kW, and so the sum works great. The discharge power does not work - neither battery reports high power output (though are both fine a lower power output)

ma9mwah, 2025-03-15:
Hi just to say I get the same issue with a single AIO. Power Battery Power reads the last know entry before a force export. Power Inverter Power is fine 

[image]

ma9mwah, 2025-04-16:
I've just installed 3.1.6 and for my single system it is all working again. @PianSom you might want to see if this version has fixed your issue as well.

PianSom, 2025-04-16:
> I've just installed 3.1.6 and for my single system it is all working again. [@PianSom](https://github.com/PianSom) you might want to see if this version has fixed your issue as well.

Thanks for the report. It was on my to-do list but has now moved up to the top!

PianSom, 2025-04-18:
Confirmed issue solved by 3.1.6. 

Phew.

closed 2025-04-18 by PianSom