britkat1980/giv_tcp #345: Control commands not taking effect via MQTT

wilb, 2025-02-18:
Long time GivTCP user but always in "monitor" mode - I've recently upgraded to v3 and attempted to integrate with Intelligent Octopus via Home Assistant to make use of the extra cheap rate slots to charge my battery.

After some head scratching, I've traced the issue to what appears to be the "control" config not being picked up and acted upon. This screenshot of MQTT explorer illustrates the issue - you'll see a number of config items that have been published by me fiddling with the config in Home Assistant, but them not being mapped across to the charging timeslots. I don't think HA is the problem as the same happens if I publish to the topic manually.

[image]

I'm running GivTCP v3, Mosquito and Home Assistant via standalone Docker containers rather than integrated - but that shouldn't make a huge amount of difference. I've also completely nuked all traces of GivTCP from MQTT and allowed it to regenerate to rule out stupidity on my part when upgrading from v2 to v3. I have a single 8.2 battery attached to a v1 inverter.  

Despite this I have a suspicion there's probably a basic error on my part but the control topic looks to be in line with how the documentation suggests, and I've used HA_Auto_D to configure HomeAssistant automatically... 

Nothing obvious stands out in the Debug logs for GivTCP - it's just looping through its usual activities without any glaring errors.

britkat1980, 2025-03-03:
Are you issuing commands via HA entities or trying to control direct through mqtt? Can you share your allsettings.json file?

wilb, 2025-03-03:
Both - initially trying to control via HA entities, but then realising they weren't working... I've tried manually up

notnownikki, 2025-03-12:
Having the same issue here. Upgraded from 2 to 3 and my automations stopped working, so I tried to verify I'd got the right control entities manually, and trying to turn off eco mode using the HA controls had no effect.

notnownikki, 2025-03-12:
Ran with debug log level and the write debug logs never show up.

I'm not specifying a topic name, everything is defaults except the inverter details.

notnownikki, 2025-03-13:
I don't have access to the facebook discussion, but yeah the control commands aren't making it to `requestcommand` , there's no pickle file with the commands.

If I get time in the next few weeks I'll see if I can work on a fix.

notnownikki, 2025-03-13:
pinging @britkat1980 so you see what might be the fix :)

notnownikki, 2025-03-13:
Oh, there's also a slight issue that we need to trigger a partial update as soon as any control command are processed, so the UI in HA gets the new state as quickly as it can and doesn't flip things back to how they were for 30 seconds before showing the new state.