britkat1980/giv_tcp #346: Security Vunerability - If port 8099 is accidently exposed to the internet

plandregan, 2025-02-19:
This open a way in, allows much fun to be had with inverters and can even expose API keys for the affected user. If you know what your looking for
A recent security scan revealed a number of exposed REST ports for Giv Users, maybe they didnt know what it was for.

Please lock this access to local IPs only, so even if exposed accidently, it cant be exploited

REST API port 8099 only accept from local addresses as dictated by IANA below
10.x.x.x
172.16.0.0 - 172.31.255.255
192.168.x.x

This will reduce the chances of this being explioted

JshGrn, 2025-02-27:
It is not GivTCP nor the inverters job to lock down your internal network, ports are not open by default on your router.

britkat1980, 2025-03-03:
I have added a local network access only config to nginx, will need to test that it doesn't have any untoward consequences in beta testing

poolski, 2025-03-05:
Under what circumstances would you expect an internal service port to be exposed to the Internet, without user input?

I don't think it should be the responsibility of any service to ensure that you *can't* expose its ports to the Internet - it shouldn't matter if it binds to a RFC1918 IP or a publicly-routable one. Even if it bound to an address in a private range, you could still dNAT that port and make it publicly available.

Adding a guard against binding to non-RFC1918 addresses doesn't really protect you from much IMO

closed 2025-10-23 by britkat1980