britkat1980/giv_tcp #348: EMS-r Charge/Discharge Slot SC %  not written

sjs-1956, 2025-02-20:
2 * AC3 Gen 1(D0.291-A0.282) + EMS-r (A0.1036). GivTCP v3.0.4

Unable to write to the EMS Control SOC % charge and discharge targets for all 3 slots. However, these values can be written via remote console to EMS unit, then GivTCP reads them correctly.

sjs-1956, 2025-02-24:
Forgot to add one example entry from log;

2025-02-24 22:11:22,500 - GivTCP3 - mqtt        -  [INFO    ] - Requesting Control Action: setExportTarget - {'exportToPercent': '47', 'slot': 2}
2025-02-24 22:11:22,913 - GivTCP3 - write       -  [ERROR   ] - Setting Export Target 2 failed: ('TypeError', 'write.py', 317)

britkat1980, 2025-03-03:
Can you turn on debug logging and share the output, type errors should be easy enough to fix, but need a bit more detail

sjs-1956, 2025-03-04:
Using BETA v3.1.2
Changing slot start and end times works
Changing slot target SOC does not work

Below excerpt from debug log. It shows a successful attempt to change the finish time of charge slot 1 to 16:01, followed by two failed attempts to change charge and discharge targets - just one line in log each. If it helps, I can do the same exercise with the DEV version.

2025-03-04 08:22:31,268 - GivTCP3 - mqtt        -  [DEBUG   ] - MQTT Message Recieved: GivEnergy/control/EMS2308146/setEMSChargeEnd1= 16:01:00
2025-03-04 08:22:31,269 - GivTCP3 - mqtt        -  [DEBUG   ] - Requesting Control Action: setChargeSlotEnd - {'finish': '16:01', 'slot': 1, 'EMS': True}
2025-03-04 08:22:31,402 - GivTCP3 - read        -  [DEBUG   ] - Write Request recieved
2025-03-04 08:22:31,404 - GivTCP3 - read        -  [DEBUG   ] - Command: setChargeSlotEnd was recieved: {'finish': '16:01', 'slot': 1, 'EMS': True}
2025-03-04 08:22:31,404 - GivTCP3 - write       -  [DEBUG   ] - Setting Charge Slot End 1 to: 16:01
2025-03-04 08:22:31,727 - GivTCP3 - write       -  [DEBUG   ] - Pushing control update to mqtt: GivEnergy/EMS2308146/Timeslots/EMS_Charge_end_time_slot_1 - 16:01:00
2025-03-04 08:22:31,732 - GivTCP3 - write       -  [DEBUG   ] - Pushing control update to pkl cache: EMS_Charge_end_time_slot_1 - 16:01:00
2025-03-04 08:22:31,733 - GivTCP3 - write       -  [INFO    ] - Setting Charge Slot End 1 to: 16:01 was a success
2025-03-04 08:22:32,034 - GivTCP3 - read        -  [DEBUG   ] - Write Commands lengths: 1 -> 1
2025-03-04 08:22:32,035 - GivTCP3 - read        -  [DEBUG   ] - No new writes, removing writerequest file

2025-03-04 08:22:46,819 - GivTCP3 - mqtt        -  [DEBUG   ] - MQTT Message Recieved: GivEnergy/control/EMS2308146/setEMSChargeTarget2= 79

2025-03-04 08:23:03,189 - GivTCP3 - mqtt        -  [DEBUG   ] - MQTT Message Recieved: GivEnergy/control/EMS2308146/setEMSDischargeTarget= 47

rdenham99, 2025-03-28:
> I've found the error in the code, Dev version 3.0.52 (and up) will have the fix if you want to try, or wait for next bets

I have installed Dev 3.0.52 and the latest Dev version 

EMS Export Target SOC will change, and I get a message in the log "Setting Export Target 1 was a success"

But the EMS Charge Target SOC's and EMS Discharge Target SOC's won't change, and no message in the log.

jakekeeys, 2025-07-13:
did you push/build the fix, i just ran the latest dev build and it still ignores my `setEMSChargeTarget1` messages
```
2025-07-13 18:45:11,854 - GivTCP - mqtt        -  [DEBUG   ] - Publishing: GivEnergy/EMS*******/Timeslots
2025-07-13 18:45:25,826 - GivTCP - mqtt        -  [DEBUG   ] - MQTT Message Recieved: GivEnergy/control/EMS*******/setEMSChargeTarget1= 95
2025-07-13 18:45:41,910 - GivTCP - read        -  [DEBUG   ] - Running partial refresh
```
at a glance i see no condition for the `setEMSChargeTarget1` "command" type
https://github.com/britkat1980/giv_tcp/blob/main/GivTCP/mqtt.py#L133
and i also noticed a discrepancy here
https://github.com/britkat1980/giv_tcp/blob/bbb9d9e36a9cb0584edef5ce99faa11db433dd40/GivTCP/entity_lut.py#L310
where it seems to be missing the slot suffix

closed 2025-08-26 by britkat1980