britkat1980/giv_tcp #349: Unprompted Change To Setting

StratoCumulus450, 2025-02-20:
I am running a Home Assistant automation that sets the battery discharge rate to 300W prior to starting a battery force charge for 3 hours at a specific time. However, as shown in the log extract below, after setting the battery discharge rate to 300W (00:15:33) the battery discharge rate is reset to 3600W (00:15:47), this latter command is not part of my HA automation. I am running Predbat, but that is placed into 'read-only-mode' 30 seconds before the battery discharge rate is set to 300W to avoid Predbat overwriting my discharge setting.

2025-02-20 00:00:58,544 - Inv1 - write       -  [INFO    ] - Setting Charge Target to: 100
2025-02-20 00:01:00,270 - Inv1 - write       -  [INFO    ] - Setting Charge Target 100 was a success
2025-02-20 00:15:20,519 - Inv1 - write       -  [INFO    ] - Disabling Charge Schedule
2025-02-20 00:15:22,119 - Inv1 - write       -  [INFO    ] - Disabling Charge was a success
2025-02-20 00:15:33,718 - Inv1 - write       -  [INFO    ] - Setting battery discharge rate to: 300 (3)
2025-02-20 00:15:38,854 - Inv1 - write       -  [INFO    ] - Enabling Charge Schedule
2025-02-20 00:15:39,154 - Inv1 - write       -  [INFO    ] - Setting battery discharge limit 3 was a success
2025-02-20 00:15:40,749 - Inv1 - write       -  [INFO    ] - Setting Charge Enable was a success
2025-02-20 00:15:47,463 - Inv1 - write       -  [INFO    ] - Setting battery discharge rate to: 3600 (50)
2025-02-20 00:15:47,855 - Inv1 - write       -  [INFO    ] - Setting battery discharge limit 50 was a success
 
I have looked at the various logs, but can find no command being issued to change the battery discharge rate to 3600W. I would be grateful for any guidance that can be offered as clearly I am missing something.

I am running;
HAOS 14.2
HA Supervisor 2025.02.1
HA Core 2025.2.4
Predbat 1.2.4
Predbat Core 8.15.1
GivTCP 2.4.9
MQTT 6.5.0

gcoan, 2025-02-20:
Might be worth adding the inverter type in case that is relevant,  and I suggest double checking the predbat logs at about the same time to check that Predbat didn't inadvertently send the command.  Not saying it is Predbat but there have been bugs like this in Predbat that it writes when it shouldn't in read only mode

StratoCumulus450, 2025-02-20:
Inverter details added to opening post and I will check Predbat logs.

closed 2025-02-25 by StratoCumulus450