britkat1980/giv_tcp #350: EMS comms errors in log

sjs-1956, 2025-02-26:
2 * AC3 Gen 1(D0.291-A0.282) + EMS-r (A0.1036). GivTCP v3.0.4

Seeing this error in app log every 2-3 hours (GivTCP3 is the EMS unit)

2025-02-26 08:22:27,529 - GivTCP3 - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-02-26 10:30:05,459 - GivTCP3 - mqtt        -  [INFO    ] - Requesting Control Action: setChargeSlotStart - {'start': '00:00', 'slot': 1, 'EMS': True}
2025-02-26 10:30:05,474 - GivTCP3 - mqtt        -  [INFO    ] - Requesting Control Action: setChargeSlotEnd - {'finish': '16:00', 'slot': 1, 'EMS': True}
2025-02-26 10:30:09,981 - GivTCP3 - base_events -  [ERROR   ] - Task exception was never retrieved
future: <Task finished name='network_consumer' coro=<Client._task_network_consumer() done, defined at /app/GivTCP_3/givenergy_modbus_async/client/client.py:329> exception=ConnectionResetError(104, 'Connection reset by peer')>
Traceback (most recent call last):
  File "/app/GivTCP_3/givenergy_modbus_async/client/client.py", line 386, in _task_network_producer
    await self.close()
  File "/app/GivTCP_3/givenergy_modbus_async/client/client.py", line 98, in close
    await self.writer.wait_closed()
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 358, in wait_closed
    await self._protocol._get_close_waiter(self)
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1005, in _read_ready__data_received
    data = self._sock.recv(self.max_size)
ConnectionResetError: [Errno 104] Connection reset by peer
2025-02-26 10:30:09,994 - GivTCP3 - base_events -  [ERROR   ] - Task exception was never retrieved
future: <Task finished name='network_producer' coro=<Client._task_network_producer() done, defined at /app/GivTCP_3/givenergy_modbus_async/client/client.py:373> exception=ConnectionResetError(104, 'Connection reset by peer')>
Traceback (most recent call last):
  File "/app/GivTCP_3/givenergy_modbus_async/client/client.py", line 386, in _task_network_producer
    await self.close()
  File "/app/GivTCP_3/givenergy_modbus_async/client/client.py", line 98, in close
    await self.writer.wait_closed()
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 358, in wait_closed
    await self._protocol._get_close_waiter(self)
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1005, in _read_ready__data_received
    data = self._sock.recv(self.max_size)
ConnectionResetError: [Errno 104] Connection reset by peer
2025-02-26 10:30:12,009 - GivTCP3 - write       -  [ERROR   ] - Setting Charge Slot 1 failed: ('Exception', 'write.py', 627)
2025-02-26 10:30:15,395 - GivTCP3 - write       -  [ERROR   ] - Setting Charge Slot 1 failed: ('Exception', 'write.py', 649)
2025-02-26 10:39:26,623 - startup     -  [ERROR   ] - Self Run loop process stuck. Killing and restarting...
2025-02-26 10:39:26,624 - startup     -  [INFO    ] - Restarting Invertor read loop every 60s
2025-02-26 10:39:27,747 - GivTCP3 - read        -  [INFO    ] - Starting watch_plant loop...
2025-02-26 10:39:27,747 - GivTCP3 - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.1.182
2025-02-26 10:39:27,750 - GivTCP3 - read        -  [CRITICAL] - Detecting inverter characteristics...

sjs-1956, 2025-02-26:
GivTCP  3.0.4. I'll try the beta first then.

sjs-1956, 2025-02-26:
Similar results with Beta version. Not sure I'm ready for the DEV version. What do you think?

2025-02-26 19:38:20,931 - GivTCP3 - base_events -  [ERROR   ] - Task exception was never retrieved
future: <Task finished name='network_consumer' coro=<Client._task_network_consumer() done, defined at /app/GivTCP_3/givenergy_modbus_async/client/client.py:329> exception=ConnectionResetError(104, 'Connection reset by peer')>
Traceback (most recent call last):
  File "/app/GivTCP_3/givenergy_modbus_async/client/client.py", line 386, in _task_network_producer
    await self.close()
  File "/app/GivTCP_3/givenergy_modbus_async/client/client.py", line 98, in close
    await self.writer.wait_closed()
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 358, in wait_closed
    await self._protocol._get_close_waiter(self)
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1005, in _read_ready__data_received
    data = self._sock.recv(self.max_size)
ConnectionResetError: [Errno 104] Connection reset by peer

gcoan, 2025-02-27:
I'd suggest trying the dev, you have nothing to loose, and if it doesn't work then its a question for Mark as GivTCP may not get support the EMS.  It kind of looks like the problem is that GivTCP is trying to work out what kind of device it is, and failing

sjs-1956, 2025-02-27:
Ok. I'll try after my GivEnergy visit today. I have a developer visiting me to inspect my installation. Perhaps he may be able to help.

sjs-1956, 2025-02-27:
A thought! I have 2 inverters and 1 EMS unit. Should I be changing the switches in the Inverters tab of the config page> I'm referring to the 'Inverter X Only  Report Battery Data' switches.

sjs-1956, 2025-02-27:
Simillar error with the DEV version. Also small error at start up (09:54:00.786) during auto-discovery;

2025-02-27 09:53:55,205 - GivTCP - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
2025-02-27 09:54:00,786 - GivTCP3 - HA_Discovery -  [ERROR   ] - Error connecting to MQTT Broker: ('RuntimeError', 'HA_Discovery.py', 399)
2025-02-27 10:30:06,848 - GivTCP3 - write       -  [INFO    ] - Setting Charge Slot 1 Start to: 00:00 was a success
2025-02-27 10:30:07,468 - GivTCP3 - write       -  [INFO    ] - Setting Charge Slot End 1 to: 16:00 was a success
2025-02-27 11:30:03,106 - GivTCP3 - base_events -  [ERROR   ] - Task exception was never retrieved
future: <Task finished name='network_consumer' coro=<Client._task_network_consumer() done, defined at /app/GivTCP_3/givenergy_modbus_async/client/client.py:329> exception=ConnectionResetError(104, 'Connection reset by peer')>
Traceback (most recent call last):
  File "/app/GivTCP_3/givenergy_modbus_async/client/client.py", line 386, in _task_network_producer
    await self.close()
  File "/app/GivTCP_3/givenergy_modbus_async/client/client.py", line 98, in close
    await self.writer.wait_closed()
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 358, in wait_closed
    await self._protocol._get_close_waiter(self)
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1005, in _read_ready__data_received
    data = self._sock.recv(self.max_size)
ConnectionResetError: [Errno 104] Connection reset by peer
2025-02-27 11:30:03,114 - GivTCP3 - base_events -  [ERROR   ] - Task exception was never retrieved
future: <Task finished name='network_producer' coro=<Client._task_network_producer() done, defined at /app/GivTCP_3/givenergy_modbus_async/client/client.py:373> exception=ConnectionResetError(104, 'Connection reset by peer')>
Traceback (most recent call last):
  File "/app/GivTCP_3/givenergy_modbus_async/client/client.py", line 386, in _task_network_producer
    await self.close()
  File "/app/GivTCP_3/givenergy_modbus_async/client/client.py", line 98, in close
    await self.writer.wait_closed()
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 358, in wait_closed
    await self._protocol._get_close_waiter(self)
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1005, in _read_ready__data_received
    data = self._sock.recv(self.max_size)
ConnectionResetError: [Errno 104] Connection reset by peer
2025-02-27 11:31:08,140 - GivTCP3 - read        -  [ERROR   ] - 5 consecutive timeout errors in watch loop. Restarting modbus connection:
2025-02-27 11:37:31,724 - GivTCP2 - read        -  [ERROR   ] - Battery Object empty so skipping

gcoan, 2025-02-27:
> Should I be changing the switches in the Inverters tab of the config page. I'm referring to the 'Inverter X Only Report Battery Data' switches.

you can try it but I think this is designed for multi-AIO users so that givtcp doesn't present all the control switches for the AIOs since the control now goes via the gateway

since your control now goes via the EMS which is analogous to the AIO gateway, you could try turning this on on your AC3 inverters.   I suggest this needs to be off on the EMS so that the EMS presents all the controls to HA.

Only other thing I can suggest is that you turn debug mode on in givtcp config and then attach the debug log here so @britkat1980 can take a look, it is looking like its a coding/EMS compatibility issue

britkat1980, 2025-03-03:
If you have an EMS you should only turn on GivTCP for the EMS. Do not connect directly to the inverters themselves. connection reset error implies that something else is trying to connect. Can you confirm you only have one instance of GivTCP running and only connecting to the EMS?