britkat1980/giv_tcp #351: missing individual battery cell temperatures

rsarwar87, 2025-02-27:
hi, i noticed that the individual battery cell temps are missing, as is the mosfet temp. they are available in mqtt explorer under raw. would it be possible to expose them to home assistant as well.

[image]

gcoan, 2025-02-28:
have you looked under battery stack 1, the battery cell temperatures are normally exposed as is the BMS temperature

LeeFarm11, 2025-02-28:
is this related - https://github.com/britkat1980/giv_tcp/issues/306

rsarwar87, 2025-02-28:
> @rsarwar87 would be useful to explain what inverter you have, software version, etc

Ho sorry. I do have a gen 3 inverter 3.6kw. Battery is 5.2 kw, gen 1. Running 3.0.4 as a docker container. HA is the current latest release.

rsarwar87, 2025-02-28:
> have you looked under battery stack 1, the battery cell temperatures are normally exposed as is the BMS temperature
> 

Hi. Yes they donot contain the info[image]

[image]

[image]

rsarwar87, 2025-02-28:
> is this related - https://github.com/britkat1980/giv_tcp/issues/306
> 

Potentially, but I am not getting any of the cell temps.

britkat1980, 2025-03-03:
This is odd, the data should be there as you expect and the code shows no reason it wouldn't appear. Do you see the values under the GivEnergy Topic under the battery serial number?
Mosfet temp (raw) is used for the Battery_Temperature value

rsarwar87, 2025-03-11:
Hi, they entries are there on explorer, but not on home assistant 

[image]