britkat1980/giv_tcp #353: Increase delay times

longlivedigital, 2025-03-01:
Is there a reason why the delay times only go up to 180 mins? Is it possible to set this in the ui or maybe add a few extra values like 4/5 or 6 hours delay? Reason being i am octopus intelligent go which give me a 6 hour charging window and i want to charge at a lower rate for those 6 hours.

https://github.com/britkat1980/giv_tcp/blob/2ef5d6b27c3ecde930ab9ffe803b7718d73b13e7/GivTCP/GivLUT.py#L717

britkat1980, 2025-03-03:
I can add extra values, but if its for IOG then I use an automation to set the charge time slots, rather than use Force Charge control.

longlivedigital, 2025-03-04:
I've had issues with Givenergy's time slots (timeslots not saving and phantom timeslots getting added) but will try it from home assistant. thanks.

closed 2025-03-04 by longlivedigital