britkat1980/giv_tcp #356: Entities "Unavailable" / duplicate entities (appended with "_2") after restart

rholligan, 2025-03-12:
Hi,

This morning I updated to my Home Assistant Green to HA Core 2025.3.2, triggering a restart.

I then found GivTCP entities reported "Unavailable" values, and discovered that the original entities had all been duplicated with "_2" appended, e.g. sensor.givtcp_fd2249g332_pv_power_2.

This is undesirable, as all of the history is associated with the original entities (without "_2")

GivTCP version: 3.0.4.

I have attempted the following remediation, without success:

- Stop & uninstall GivTCP add-on
- Delete /homeassistant/GivTCP directory
- Clear "GivTCP" topic & 479 child topics using MQTT Explorer
- Stop & uninstall Mosquitto Broker
- Delete all duplicate entities
- Reinstall MQTT
- Reinstall GivTCP

The duplicate entities unfortunately return 🫤

Can anyone help resolve this? I'd like to continue using the original entities, and retain the associated history within HA.

rholligan, 2025-03-12:
I've checked similar items #202, #100, #29 but couldn't find any solution.

In #31 I noticed this slightly different approach to mine - specifically Step 5 which asks to delete all entities, whereas so far I have only been deleting the duplicates. Perhaps this might work?

> The new entity_id naming convention is: . So if that’s what your current entity ids are then you may get the _2 entities. To return to the original and keep the link to historical data, then you need to:
> 
> 1. In GivTCP settings, change the HADevicePrefix to 'GivTCP' for ALL inverters - IMPORTANT STEP IF YOU HAVE MULTIPLE INVERTERS!
> 2. Install MQTT Explorer
> 3. STOP GivTCP
> 4. Log in to MQTT Explorer, using same login details as GivTCP uses
**> 5. Delete ALL MQTT entries that refer to GivTCP (some are hidden in different hierarchies)**
> 6. Go to settings/Devices & Services - select the entites under Mosquitto Broker and DELETE all entities that are associated with GivTCP
> 7. Restart GivTCP

gcoan, 2025-03-12:
I'd suggest shutting down givtcp, deleting all Mqtt entries that refer to givtcp, restarting mosquito, so when you restart givtcp it recreates all the givtcp entities with the correct pathname.

rholligan, 2025-03-13:
All sorted by following these steps:

1. Stop & uninstall GivTCP add-on
2. Stop & uninstall Mosquitto Broker
3. Delete all GivTCP entities
4. Reinstall MQTT
5. Reinstall GivTCP