britkat1980/giv_tcp #357: One of two inverters not working on givtcp install

Mattntate, 2025-03-16:
I hope this is the right place to ask this... I'm pretty new to HA and can't code! 
I am trying to install GivTCP v3, my first inverter is being found and all the entities are showing in MQTT, but there is an error and the second inverter / entities are not showing. I get the below errors in the log:

2025-03-16 14:08:33,632 - GivTCP2 - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'IndexError'>, IndexError('list index out of range'), <traceback object at 0x7fa0945d40>), 918)
2025-03-16 14:08:33,635 - GivTCP2 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7fa0946400>), 1657)
2025-03-16 14:08:33,636 - GivTCP2 - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1825)
2025-03-16 14:09:04,689 - GivTCP2 - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'IndexError'>, IndexError('list index out of range'), <traceback object at 0x7fa092f6c0>), 918)
2025-03-16 14:09:04,691 - GivTCP2 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7fa091c740>), 1657)
2025-03-16 14:09:04,692 - GivTCP2 - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1825)
2025-03-16 14:09:18,575 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-03-16 14:09:18,611 - startup     -  [INFO    ] - Network Found: 192.168.68.1/24
2025-03-16 14:09:20,614 - startup     -  [INFO    ] - Searching for Inverters
2025-03-16 14:09:20,614 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-03-16 14:09:23,246 - startup     -  [INFO    ] - INV- Scanning network (1):192.168.68.1/24
2025-03-16 14:09:31,237 - client      -  [INFO    ] - Connection established to 192.168.68.110:8899
2025-03-16 14:09:31,238 - client      -  [INFO    ] - Detecting plant
2025-03-16 14:09:32,489 - client      -  [INFO    ] - Plant Detected
2025-03-16 14:09:35,261 - startup     -  [INFO    ] - Inverter CE2224G083 which is a Gen1 - Ac with 1 batteries has been found at: 192.168.68.110
2025-03-16 14:09:35,279 - startup     -  [INFO    ] - Setting up invertor: 1
2025-03-16 14:09:35,437 - startup     -  [INFO    ] - ==============================================================
2025-03-16 14:09:35,438 - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2025-03-16 14:09:35,439 - startup     -  [INFO    ] - ====     http://192.168.68.102:8099/config.html             ====
2025-03-16 14:09:35,440 - startup     -  [INFO    ] - ==============================================================
2025-03-16 14:09:35,442 - startup     -  [INFO    ] - Running Invertor 1 (CE2224G083) read loop every 30/120s
2025-03-16 14:09:35,448 - startup     -  [INFO    ] - Setting up invertor: 2
2025-03-16 14:09:35,784 - startup     -  [INFO    ] - ==============================================================
2025-03-16 14:09:35,785 - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2025-03-16 14:09:35,786 - startup     -  [INFO    ] - ====     http://192.168.68.102:8099/config.html             ====
2025-03-16 14:09:35,786 - startup     -  [INFO    ] - ==============================================================
2025-03-16 14:09:35,796 - startup     -  [INFO    ] - Running Invertor 2 (CE2223G938) read loop every 30/120s
2025-03-16 14:09:35,806 - startup     -  [INFO    ] - Serving Web Dashboard from port 3000
[2025-03-16 14:09:35 +0000] [70] [INFO] Starting gunicorn 23.0.0
[2025-03-16 14:09:35 +0000] [70] [INFO] Listening at: http://0.0.0.0:6350 (70)
[2025-03-16 14:09:35 +0000] [70] [INFO] Using worker: sync
[2025-03-16 14:09:35 +0000] [90] [INFO] Booting worker with pid: 90
[2025-03-16 14:09:36 +0000] [81] [INFO] Starting gunicorn 23.0.0
[2025-03-16 14:09:36 +0000] [81] [INFO] Listening at: http://0.0.0.0:6345 (81)
[2025-03-16 14:09:36 +0000] [81] [INFO] Using worker: sync
[2025-03-16 14:09:36 +0000] [91] [INFO] Booting worker with pid: 91
[2025-03-16 14:09:36 +0000] [92] [INFO] Booting worker with pid: 92
[2025-03-16 14:09:36 +0000] [93] [INFO] Booting worker with pid: 93
[2025-03-16 14:09:37 +0000] [84] [INFO] Starting gunicorn 23.0.0
[2025-03-16 14:09:37 +0000] [84] [INFO] Listening at: http://0.0.0.0:6346 (84)
[2025-03-16 14:09:37 +0000] [84] [INFO] Using worker: sync
[2025-03-16 14:09:37 +0000] [94] [INFO] Booting worker with pid: 94
[2025-03-16 14:09:37 +0000] [95] [INFO] Booting worker with pid: 95
[2025-03-16 14:09:37 +0000] [96] [INFO] Booting worker with pid: 96
2025-03-16 14:09:41,051 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-03-16 14:09:41,080 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.68.110
2025-03-16 14:09:41,110 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-03-16 14:09:41,917 - GivTCP2 - read        -  [INFO    ] - Starting watch_plant loop...
2025-03-16 14:09:41,918 - GivTCP2 - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.68.100
2025-03-16 14:09:41,961 - GivTCP2 - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-03-16 14:10:28,795 - GivTCP - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
2025-03-16 14:10:29,573 - GivTCP2 - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'IndexError'>, IndexError('list index out of range'), <traceback object at 0x7f90b56f80>), 918)
2025-03-16 14:10:29,575 - GivTCP2 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f90b57840>), 1657)
2025-03-16 14:10:29,576 - GivTCP2 - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1825)
2025-03-16 14:11:00,635 - GivTCP2 - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'IndexError'>, IndexError('list index out of range'), <traceback object at 0x7f90b6e580>), 918)
2025-03-16 14:11:00,637 - GivTCP2 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f90b6d9c0>), 1657)
2025-03-16 14:11:00,638 - GivTCP2 - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1825)
2025-03-16 14:11:31,696 - GivTCP2 - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'IndexError'>, IndexError('list index out of range'), <traceback object at 0x7f90b6e5c0>), 918)
2025-03-16 14:11:31,697 - GivTCP2 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f90b6ebc0>), 1657)
2025-03-16 14:11:31,698 - GivTCP2 - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1825)
2025-03-16 14:12:02,762 - GivTCP2 - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'IndexError'>, IndexError('list index out of range'), <traceback object at 0x7f90b6e800>), 918)
2025-03-16 14:12:02,764 - GivTCP2 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f90b6f440>), 1657)

Mattntate, 2025-03-16:
@gcoan Thanks very much for replying.  I've tried that , and restarted Givtcp but nothing has changed.

gcoan, 2025-03-16:
> @gcoan Thanks very much for replying.  I've tried that , and restarted Givtcp but nothing has changed.
> 
> 

try resetting the inverter to default settings from the givenergy portal and then restarting the inverter, not givtcp

the inverter looks like its rejecting the connection which is why I suggested looking at the inverter

Mattntate, 2025-03-16:
@gcoan Yep, I reset and restarted the inverter then givtcp to see if it would connect, but there was no change.

gcoan, 2025-03-16:
sorry I can't suggest any other fixes

Mattntate, 2025-03-17:
No problem,  thanks anyway.

gcoan, 2025-03-17:
As an idea, maybe try checking the /config/GivTCP/allsettings.json file to see if it is setup correctly?  Maybe just try running it for one inverter (that you know works), and then change it for the other (that's had problems) ?

Here if it helps is mine:

```json

{
    "Web_Dash": false,
    "Web_Dash_Port": 3000,
    "MQTT_Output": true,
    "MQTT_Address": "core-mosquitto",
    "MQTT_Username": "addons",
    "MQTT_Password": "oixohwoo9CiDaeX6riegei3phohfaiyeishohshie6mai0Feewi5oyahjaeQu5di",
    "MQTT_Retain": true,
    "MQTT_Topic": "",
    "MQTT_Port": 1883,
    "inverter_enable_1": true,
    "invertorIP_1": "192.168.1.160",
    "serial_number_1": "SDXXXXGXXX",
    "inverterName_1": "G",
    "inverter_battery_only_1": false,
    "inverter_enable_2": true,
    "invertorIP_2": "192.168.1.159",
    "serial_number_2": "SDXXXXGXXX",
    "inverterName_2": "H",
    "inverter_battery_only_2": false,
    "inverter_enable_3": false,
    "invertorIP_3": "",
    "serial_number_3": "",
    "inverterName_3": "GivTCP3",
    "inverter_battery_only_3": false,
    "inverter_enable_4": false,
    "invertorIP_4": "",
    "serial_number_4": "",
    "inverterName_4": "GivTCP4",
    "inverter_battery_only_4": false,
    "inverter_enable_5": false,
    "invertorIP_5": "",
    "serial_number_5": "",
    "inverterName_5": "GivTCP5",
    "inverter_battery_only_5": false,
    "Influx_Output": false,
    "influxURL": "",
    "influxToken": "",
    "influxBucket": "",
    "influxOrg": "",
    "self_run": true,
    "self_run_timer": 30,
    "self_run_timer_full": 120,
    "HA_Auto_D": true,
    "dynamic_tariff": false,
    "export_rate": 0.04,
    "day_rate": 0.395,
    "day_rate_start": "05:30",
    "night_rate": 0.155,
    "night_rate_start": "23:30",
    "TZ": "Europe/London",
    "Print_Raw_Registers": true,
    "Log_Level": "info",
    "queue_retries": 2,
    "data_smoother": "medium",
    "Smart_Target": false,
    "GE_API": "",
    "SOLCASTAPI": "",
    "SOLCASTSITEID": "",
    "SOLCASTSITEID2": "",
    "PALM_WINTER": "01,02,03,10,11,12",
    "PALM_SHOULDER": "04,05,09",
    "PALM_MIN_SOC_TARGET": 25,
    "PALM_MAX_SOC_TARGET": 45,
    "PALM_BATT_RESERVE": 4,
    "PALM_BATT_UTILISATION": 0.85,
    "PALM_WEIGHT": 35,
    "LOAD_HIST_WEIGHT": "1",
    "evc_enable": false,
    "evc_ip_address": "",
    "evc_self_run_timer": 10,
    "evc_import_max_current": 60,
    "v3upgrade": true,
    "auto_scan": true,
    "Model_1": "Hybrid",
    "Model_2": "Hybrid",
    "Model_3": "Hybrid",
    "Model_4": "Hybrid",
    "Model_5": "Hybrid",
    "Host_IP": "192.168.1.69",
    "cache_location": "/config/GivTCP",
    "PYTHONPATH": "/app",
    "serial_number_evc": "",
    "timezone": "Europe/London",
    "Model_9": "Hybrid",
    "Model_10": "Hybrid",
    "Model_6": "Hybrid",
    "Model_8": "Hybrid",
    "Model_7": "Hybrid"
}
```

Mattntate, 2025-03-18:
@gcoan Thanks again for the help.  I've tried that - it didn't work.  To start with it kept adding the first inverter back to the allsetings file, I deleted and reinstalled Givtcpa couple of times, no joy. I then blocked the first inverter from the internet and tried again but it still didn't work.  I've installed v2.4.9 and that is working so far... so not sure why v3 won't.
Anyway, thanks again,
Matt

gcoan, 2025-03-18:
Sorry I couldn't fix the problem.  

You could maybe try the beta version of givtcp v3 and see if that works?

Mattntate, 2025-03-19:
I tried the beta version yesterday before I gave up and went with v2, but it was the same. Thanks very much for your help and time though,  it was much appreciated.