britkat1980/giv_tcp #358: Invalid value for number - discharge slots

jasonwfrancis1, 2025-03-17:
My home assistant logs report hundreds of these errors, for all the discharge slots:

Invalid value for number.givtcp_gwxxxx_gwxxxx_discharge_target_soc_5: 0 (range 4.0 - 100.0)

I think its getting 0 from GE because the slot hasn't been used, but 0 is not a valid number.

github-dpmracing, 2025-09-17:
Did you get to the bottom of this?  I’ve got the same errors and trying to resolve

britkat1980, 2026-10-11:
Closing this as it's had no activity for over 8 weeks.

GivTCP 3.6 has since been released ([latest: v3.6.2](https://github.com/britkat1980/giv_tcp/releases/latest)). It moves to a new Modbus library and includes a large number of fixes, so this may well behave differently now. If you still see this on 3.6.2, please reopen it, or open a new issue, with a debug log.

closed 2026-10-11 by britkat1980