britkat1980/giv_tcp #359: Error processing Inverter data

thewookiewon, 2025-03-17:
Updated to HAOS 15.0 and GivTCP started to produce these errors, downgraded back to 14.2 and in still getting these errors. 

I've stopped GivTCP, deleted all MQTT entries for GivEnergy and started GivTCP but that hasn't solved it, I've restarted the inverter. 

I'm lost on what else to try

```
2025-03-17 19:22:45,400 - GivTCP - read        -  [DEBUG   ] - Getting Battery Details
2025-03-17 19:22:45,400 - GivTCP - read        -  [DEBUG   ] - Building battery output: 
2025-03-17 19:22:45,400 - GivTCP - read        -  [DEBUG   ] - Battery DZ2240G341 added
2025-03-17 19:22:45,400 - GivTCP - read        -  [DEBUG   ] - Building battery output: 
2025-03-17 19:22:45,400 - GivTCP - read        -  [ERROR   ] - Error getting Battery Data: ((<class 'KeyError'>, KeyError('Battery_Details'), <traceback object at 0x7fa944ec80>), 381)
2025-03-17 19:22:45,401 - GivTCP - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'TypeError'>, TypeError("'NoneType' object is not iterable"), <traceback object at 0x7fa944f340>), 1074)
2025-03-17 19:22:45,401 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7fa944ed80>), 1657)
2025-03-17 19:22:45,401 - GivTCP - read        -  [DEBUG   ] - processData result: processData Error processing registers: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7fa944ed80>), 1657)
2025-03-17 19:22:45,401 - GivTCP - read        -  [DEBUG   ] - Running pubFromPickle
2025-03-17 19:22:45,401 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1825)
2025-03-17 19:23:15,439 - GivTCP - read        -  [DEBUG   ] - Running partial refresh
2025-03-17 19:23:16,848 - GivTCP - read        -  [DEBUG   ] - Data get was successful, now running handler if needed: 
2025-03-17 19:23:16,848 - GivTCP - read        -  [DEBUG   ] - Running processData
2025-03-17 19:23:16,848 - GivTCP - read        -  [DEBUG   ] - Beginning parsing of Inverter data
2025-03-17 19:23:16,849 - GivTCP - read        -  [DEBUG   ] - Getting Total Energy Data
2025-03-17 19:23:16,849 - GivTCP - read        -  [DEBUG   ] - Getting Today Energy Data
2025-03-17 19:23:16,849 - GivTCP - read        -  [DEBUG   ] - Getting PV Power
2025-03-17 19:23:16,850 - GivTCP - read        -  [DEBUG   ] - Getting Grid Power
2025-03-17 19:23:16,850 - GivTCP - read        -  [DEBUG   ] - Getting EPS Power
2025-03-17 19:23:16,850 - GivTCP - read        -  [DEBUG   ] - Getting PInv Power
2025-03-17 19:23:16,850 - GivTCP - read        -  [DEBUG   ] - Getting Load Power
2025-03-17 19:23:16,850 - GivTCP - read        -  [DEBUG   ] - Getting Self Consumption Power
2025-03-17 19:23:16,850 - GivTCP - read        -  [DEBUG   ] - Getting Solar to H/B/G Power Flows
2025-03-17 19:23:16,850 - GivTCP - read        -  [DEBUG   ] - Getting Grid to Battery/House Power Flow
2025-03-17 19:23:16,850 - GivTCP - read        -  [DEBUG   ] - Getting mode control figures
2025-03-17 19:23:16,850 - GivTCP - read        -  [DEBUG   ] - Calculating Mode...
2025-03-17 19:23:16,851 - GivTCP - read        -  [DEBUG   ] - Mode is: Eco
2025-03-17 19:23:16,851 - GivTCP - read        -  [DEBUG   ] - Force Export is not Running
2025-03-17 19:23:16,851 - GivTCP - read        -  [DEBUG   ] - Getting SOC
2025-03-17 19:23:16,851 - GivTCP - read        -  [DEBUG   ] - Getting Battery Energy Data
2025-03-17 19:23:16,851 - GivTCP - read        -  [DEBUG   ] - Getting Battery Power data
2025-03-17 19:23:16,851 - GivTCP - read        -  [DEBUG   ] - Getting Solar to H/B/G Power Flows
2025-03-17 19:23:16,851 - GivTCP - read        -  [DEBUG   ] - Getting Battery to House Power Flow
2025-03-17 19:23:16,851 - GivTCP - read        -  [DEBUG   ] - Getting Grid to Battery/House Power Flow
2025-03-17 19:23:16,852 - GivTCP - read        -  [DEBUG   ] - Getting Battery to Grid Power Flow
2025-03-17 19:23:16,852 - GivTCP - read        -  [DEBUG   ] - Getting TimeSlot data
2025-03-17 19:23:16,852 - GivTCP - read        -  [DEBUG   ] - Getting inverter Details
2025-03-17 19:23:16,853 - GivTCP - read        -  [DEBUG   ] - Getting Battery Details
2025-03-17 19:23:16,853 - GivTCP - read        -  [DEBUG   ] - Building battery output: 
2025-03-17 19:23:16,853 - GivTCP - read        -  [DEBUG   ] - Battery DZ2240G341 added
2025-03-17 19:23:16,853 - GivTCP - read        -  [DEBUG   ] - Building battery output: 
2025-03-17 19:23:16,853 - GivTCP - read        -  [ERROR   ] - Error getting Battery Data: ((<class 'KeyError'>, KeyError('Battery_Details'), <traceback object at 0x7fa944cfc0>), 381)
2025-03-17 19:23:16,853 - GivTCP - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'TypeError'>, TypeError("'NoneType' object is not iterable"), <traceback object at 0x7fa944d440>), 1074)
2025-03-17 19:23:16,854 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7fa944d080>), 1657)
2025-03-17 19:23:16,854 - GivTCP - read        -  [DEBUG   ] - processData result: processData Error processing registers: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7fa944d080>), 1657)
2025-03-17 19:23:16,854 - GivTCP - read        -  [DEBUG   ] - Running pubFromPickle
2025-03-17 19:23:16,854 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1825)
 Live
```

thewookiewon, 2025-03-17:
I have tried so many times to get this working, its not even writing to the MQTT broker yet so I'm at a lose.....

I've factory reset the inverter, restarted the inverter, uninstalled GivTCP and deleted the folder. Is there anywhere else like caches to delte for GivTCP?

gcoan, 2025-03-17:
the givtcp caches are /config/GivTCP/*.pkl but if you have deleted the givtcp folder then that'd cover it

there is another issue logged on this, see above, the o.p. reverted back to HAOS 14.2 to restore GivTCP working

thewookiewon, 2025-03-17:
@gcoan That was me raising it on the addon github :(, reverting back to 14.2 didn't solve it for me

gcoan, 2025-03-17:
ah yes, sorry, I see you tried reverting

have you tried:

uninstalling givtcp
reboot HA and all addons
with MQTT explorer, delete all givtcp MQTT objects
uninstall mosquitto broker
reboot HA and all addons

this should give a blank canvas to re-run the install from scratch

all on HAOS 14.2 because they works OK

thewookiewon, 2025-03-17:
I have done yes, still the same

`2025-03-17 23:17:21,538 - GivTCP - read        -  [ERROR   ] - Error getting Battery Data: ((<class 'KeyError'>, KeyError('Battery_Details'), <traceback object at 0x7faf8ff580>), 381)
2025-03-17 23:17:21,539 - GivTCP - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'TypeError'>, TypeError("'NoneType' object is not iterable"), <traceback object at 0x7faf8ff080>), 1074)
2025-03-17 23:17:21,539 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7faf8ff4c0>), 1657)
2025-03-17 23:17:21,539 - GivTCP - read        -  [DEBUG   ] - processData result: processData Error processing registers: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7faf8ff4c0>), 1657)
2025-03-17 23:17:21,539 - GivTCP - read        -  [DEBUG   ] - Running pubFromPickle
2025-03-17 23:17:21,539 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1825)`

thewookiewon, 2025-03-18:
Now that is weird, working today and didn't restart the batteries.

The batteries were depleted yesterday and now fully charged and GivTCP is working with MQTT broker full of entries, so should empty batteries cause this with GivTCP?

thewookiewon, 2025-03-18:
Only error in now getting is this one, any ideas? 

2025-03-18 08:34:48,325 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('FileNotFoundError', 'read.py', 1821)

gcoan, 2025-03-18:
> Now that is weird, working today and didn't restart the batteries.
> 
> The batteries were depleted yesterday and now fully charged and GivTCP is working with MQTT broker full of entries, so should empty batteries cause this with GivTCP? 

very strange.  The errors above looked like modbus read errors, so restarting/resetting the inverter to defaults is the fix I usually apply to this, but then it works out itself.

I wouldn't think empty batteries would cause givtcp to error because even when the batteries are saying they are empty, they are not totally exhausted, there's still a bit in them.  Potentially an empty battery could report a SoC that GivTCP doesn't expect, but that doesn't look like the error returned, and as I say, the batteries are never really totally empty 

not sure about the error you are seeing today, what file does that line of read.py indicate is being read?

mjs7uk, 2025-03-22:
I am getting something similar only it has not just disappeared, this has been working for months ok, any ideas?

2025-03-22 11:32:39 +0000] [70] [INFO] Starting gunicorn 23.0.0
[2025-03-22 11:32:39 +0000] [70] [INFO] Listening at: http://0.0.0.0:6350 (70)
[2025-03-22 11:32:39 +0000] [70] [INFO] Using worker: sync
[2025-03-22 11:32:39 +0000] [81] [INFO] Booting worker with pid: 81
[2025-03-22 11:32:39 +0000] [80] [INFO] Starting gunicorn 23.0.0
[2025-03-22 11:32:39 +0000] [80] [INFO] Listening at: http://0.0.0.0:6345 (80)
[2025-03-22 11:32:39 +0000] [80] [INFO] Using worker: sync
[2025-03-22 11:32:39 +0000] [82] [INFO] Booting worker with pid: 82
[2025-03-22 11:32:39 +0000] [85] [INFO] Booting worker with pid: 85
[2025-03-22 11:32:39 +0000] [86] [INFO] Booting worker with pid: 86
2025-03-22 11:32:39,312 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-03-22 11:32:39,312 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to xxx.xxx.xx.xx
2025-03-22 11:32:39,321 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-03-22 11:32:46,344 - GivTCP - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'IndexError'>, IndexError('list index out of range'), <traceback object at 0x7faf86efae00>), 918)
2025-03-22 11:32:46,344 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7faf86efb040>), 1657)
2025-03-22 11:32:46,344 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1825)
2025-03-22 11:33:17,378 - GivTCP - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'IndexError'>, IndexError('list index out of range'), <traceback object at 0x7faf86f12c00>), 918)
2025-03-22 11:33:17,379 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7faf86f12e40>), 1657)
2025-03-22 11:33:17,379 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 1825)
2025-03-22 11:33:48,416 - GivTCP - read        -  [ERROR   ] - Error processing Inverter data: ((<class 'IndexError'>, IndexError('list index out of range'), <traceback object at 0x7faf86f12b00>), 918)
2025-03-22 11:33:48,417 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7faf86f13300>), 1657)
2025-03-22 11:33:48,417 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registe

britkat1980, 2025-04-11:
This can sometimes be attributed to the battery being in calibration during startup. New fix in 3.1.4 coming to address this. Could that have been the issue?

mjs7uk, 2025-04-11:
Yes that was the best exactly the issue Sent from Android device

closed 2025-04-11 by britkat1980