britkat1980/giv_tcp #362: GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2150)

73ttv627re77, 2025-03-28:
Hybird gen2 with single batery running 920 with realtime enabled.

Clearing everything from mqtt explorer after removal of the add-on and with RTC disabled in the portal still shows the same error message, any solutions? (gen2 hybrid) this only broken on the beta version, prod is all good.

Clearing the files in the givtco folder and restarting mqtt also does not help.

Even though the error is there, everything like predbat seem to work ok.

here is a sample with debug on:

2025-03-28 19:58:08,783 - read - [ERROR] - Error getting data from cache: ('AttributeError', 'read.py', 2150)
2025-03-28 19:58:09,770 - HA_Discovery - [DEBUG] - Foundmessages= 0
2025-03-28 19:58:09,770 - HA_Discovery - [DEBUG] - Moremessages= 0
2025-03-28 19:58:10,771 - HA_Discovery - [DEBUG] - Foundmessages= 0
2025-03-28 19:58:10,771 - HA_Discovery - [DEBUG] - Moremessages= 0
2025-03-28 19:58:11,771 - HA_Discovery - [DEBUG] - Foundmessages= 0
2025-03-28 19:58:11,771 - HA_Discovery - [DEBUG] - Moremessages= 0
2025-03-28 19:58:12,771 - HA_Discovery - [DEBUG] - Foundmessages= 0
2025-03-28 19:58:12,771 - HA_Discovery - [DEBUG] - Moremessages= 0
2025-03-28 19:58:13,771 - HA_Discovery - [DEBUG] - Foundmessages= 0
2025-03-28 19:58:13,771 - HA_Discovery - [DEBUG] - Moremessages= 0
2025-03-28 19:58:13,771 - HA_Discovery - [DEBUG] - 0 discovery messages changed and removed
2025-03-28 19:58:16,792 - read - [ERROR] - Error getting data from cache: ('AttributeError', 'read.py', 2150)
2025-03-28 19:58:18,777 - HA_Discovery - [DEBUG] - Connecting to broker: core-mosquitto
2025-03-28 19:58:18,778 - HA_Discovery - [DEBUG] - In wait loop (sendDiscoMsg)
2025-03-28 19:58:18,978 - HA_Discovery - [DEBUG] - In wait loop (sendDiscoMsg)
2025-03-28 19:58:19,178 - HA_Discovery - [DEBUG] - In wait loop (sendDiscoMsg)
2025-03-28 19:58:19,378 - HA_Discovery - [DEBUG] - In wait loop (sendDiscoMsg)
2025-03-28 19:58:19,579 - HA_Discovery - [DEBUG] - In wait loop (sendDiscoMsg)
2025-03-28 19:58:19,778 - HA_Discovery - [DEBUG] - connected OK Returned code=Success
2025-03-28 19:58:19,779 - HA_Discovery - [DEBUG] - Publishing MQTT: core-mosquitto
2025-03-28 19:58:24,794 - read - [ERROR] - Error getting data from cache: ('AttributeError', 'read.py', 2150)
2025-03-28 19:58:25,784 - HA_Discovery - [DEBUG] - Sent 160 discovery messages
2025-03-28 19:58:25,784 - HA_Discovery - [DEBUG] - Found 160 discovery messages
2025-03-28 19:58:25,784 - read - [DEBUG] - Publish all to MQTT
2025-03-28 19:58:25,784 - mqtt - [DEBUG] - MQTT Connection appears closed, re-opening
2025-03-28 19:58:25,784 - mqtt - [DEBUG] - Opening MQTT Connection to core-mosquitto
2025-03-28 19:58:25,785 - mqtt - [DEBUG] - connected OK Returned code=Success

britkat1980, 2025-04-11:
Not a lot to go on here, in terms of debug data. Can you load the latest prod version (3.1.3) and let me know if the problem persists

73ttv627re77, 2025-04-11:
Running the latest 3.1.3 and all good, this can be closed, thank you

closed 2025-04-11 by 73ttv627re77