britkat1980/giv_tcp #367: GivTCP-BETA v3.1.3 Error

AberDino, 2025-04-08:
Not sure if you are looking for feedback regarding the BETA versions, but I gave GivTCP-BETA v3.1.3 a spin and still get the following error:

```
2025-04-08 18:36:40,199 - GW - read        -  [ERROR   ] - Error processing Gateway data: ((<class 'UnboundLocalError'>, UnboundLocalError("cannot access local variable 'power_flow_output' where it is not associated with a value"), <traceback object at 0x7fa1afb300>), 1705)
2025-04-08 18:36:40,202 - GW - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7fa1afb640>), 1945)
2025-04-08 18:36:40,202 - GW - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2126)
```
Log file attached.

[GivTCP-BETAv3.1.3Log.txt](https://github.com/user-attachments/files/19654293/GivTCP-BETAv3.1.3Log.txt)

AberDino, 2025-04-10:
I've updated the title, as I still get the error after upgrading to the "official" release. I stepped through the configuration again, saved and restarted, but no change:
```
2025-04-10 08:27:26,910 - GW - read        -  [ERROR   ] - Error processing Gateway data: ((<class 'UnboundLocalError'>, UnboundLocalError("cannot access local variable 'power_flow_output' where it is not associated with a value"), <traceback object at 0x7f8fc80480>), 1705)
2025-04-10 08:27:26,911 - GW - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f8fc81a00>), 1943)
2025-04-10 08:27:26,912 - GW - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2124)
```

Log file attached. Any thoughts on what the issue is?

[GivTCP-3.1.3Log.txt](https://github.com/user-attachments/files/19680303/GivTCP-3.1.3Log.txt)

gcoan, 2025-04-10:
I don't get this error on my gen 1 hybrid's.

Just as a wild idea, can you do a 'reset to defaults' and/or restart the gateway? I have in the past seen issues with my inverter not wanting to talk to givtcp and resetting it cures the problem.

Of course it could be a code issue with AIO gateways but would expect others would be reporting it as well

AberDino, 2025-04-10:
Thanks @gcoan, I will give that a try, but note I'm not the only one with this issue, see [here](https://github.com/britkat1980/giv_tcp/issues/364) reported for 3.12.

peter1rhodes, 2025-04-10:
I also have this issue. No data coming from the gateway, errors:

```
2025-04-10 10:34:27,226 - GivTCPGateway - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'Exception'>, Exception('Process Data Failure'), <traceback object at 0x7f433656b5c0>), 1943)
2025-04-10 10:34:27,226 - GivTCPGateway - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2124)
2025-04-10 10:34:49,169 - GivTCPGateway - read        -  [ERROR   ] - Error processing Gateway data: ((<class 'UnboundLocalError'>, UnboundLocalError("cannot access local variable 'power_flow_output' where it is not associated with a value"), <traceback object at 0x7f4336568980>), 1705)
```

gcoan, 2025-04-10:
Yes I saw that other issue, seemed to be a problem with the network scan not working properly.  Have you tried turning the autoscan off to see if it makes any difference to you?

It's just strange that it's not appearing to be a common reported problem.  If it was a general AIO/gateway issue there would be a lot more reported issues

dpiekacz, 2025-04-10:
Just updated to GivTCP v3.1.3 (not BETA).
I see the same error messages in my Gateway+AIO setup with Gateway metrics not being updated.

tKe, 2025-04-10:
Also seeing this with a GW+AIO setup since moving to 3.1.3

peter1rhodes, 2025-04-10:
> I also have this issue. No data coming from the gateway, errors:
>
> ```
> 2025-04-10 10:34:27,226 - GivTCPGateway - read        -  [ERROR   ] - inverter Update failed so using last known good data from [...]

To add more detail to this: I have restarted, and then done a hardware reboot, on the Gateway, and restarted the Add-on too. Same issue

AberDino, 2025-04-11:
Just wondering if somebody is looking into this? Anything we can do to help resolve the issue?

112233mark, 2025-04-11:
I have the same error after updating to Givtcp 3.1.3
I changed all settings.json and set “auto_scan”: false. And still get the same.

britkat1980, 2025-04-11:
Can you confirm what your setup is? How many AIO's connected to the gateway, there are some issues with single AIO setups which I need to look at, but if you have multiple then I need to dig into that as a seperate issue

peter1rhodes, 2025-04-11:
Speaking personally, I have a single AIO, a single gateway, a single gen3 hybrid inverter, and a single GE EV charger.

britkat1980, 2025-04-11:
Thanks, have found a bit of code which needs tweaking to fix this, but in general there is nothing of value from the Gateway if it only have 1 AIO attached, no controls. You are best off to disable the Gateway in the GivTCP config (inverter page) and just leave the single AIO running.

matthewturner, 2025-04-11:
I have exactly the same setup as @AberDino (Gateway, single AIO and EV charger, all updated to the latest GivEnergy firmwares). I get the same errors about `power_flow_output`.

AberDino, 2025-04-13:
GivTCP-BETA v3.1.4 out now, which resolves the issue for me.

closed 2025-04-13 by AberDino

peter1rhodes, 2025-04-14:
> Thanks, have found a bit of code which needs tweaking to fix this, but in general there is nothing of value from the Gateway if it only have 1 AIO attached, no controls. You are best off to disable the Gateway in the GivTCP config (inverter page) and just leave the single AIO running. Then leave the battery only switch off so you get all data

@britkat1980 I thought I would leave a comment on this (now-closed) issue as I see the fix has included advising users in the documentation to not use the gateway in such a setup. One example for me is that the Gateway's "load power" statistics correctly exclude the EV charger, while the AIO's do not. Thus, without the gateway, I have no measurement at all for how much energy the house has used, without the EV charger.

This may be a bug (?), because interestingly, the AIO's load _power_ figures clearly does not include the EV charger, while its load _energy_ does. Thus, the power can be reporting a couple of hundred watts, while the energy increases at 7kWh per hour!

AberDino, 2025-04-14:
I'm with you @peter1rhodes, there is useful data to be had from the gateway. I'm hoping @britkat1980 will continue to support the single AIO option **with** the gateway, despite the log entry advising not to include the gateway with a single AIO.