britkat1980/giv_tcp #368: 3.1.3 Error Invalid option for givtcp...mode in Home Assistant Log

IWR69, 2025-04-10:
This morning I upgraded GivTCP from 3.0.4 to 3.1.3 on my Home Assistant 2025.3.4 and everything appears to be running fine.

However, when I look in the Home Assistant Log I can see the following error...

[image]

I manually changed the Mode to Eco, it was set to Unknown, and restart HA but the error is still being recorded.

The GivTCP Log looks fine except line #2 which says "No HA MQTT...", which is incorrect, and may be caused by me using my own MQTT user name and password (thanks Oliver). The upgrade from 2.4.9 to 3.0.4 did not pickup my user and password but seemed to use some default. I had to correct this to get it working.
```
2025-04-10 07:13:42,481 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-04-10 07:13:42,492 - startup     -  [INFO    ] - No HA MQTT service has been found. Install and run the Mosquitto addon, or manually configure your own MQTT broker.
2025-04-10 07:13:42,722 - startup     -  [INFO    ] - Network Found: 192.168.1.103/24
2025-04-10 07:13:44,723 - startup     -  [INFO    ] - Searching for Inverters
2025-04-10 07:13:44,723 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-04-10 07:13:49,506 - client      -  [INFO    ] - Connection established to 192.168.1.105:8899
2025-04-10 07:13:49,507 - client      -  [INFO    ] - Detecting plant
2025-04-10 07:13:54,165 - client      -  [INFO    ] - Plant Detected
2025-04-10 07:13:56,925 - startup     -  [INFO    ] - Inverter ********** which is a Gen3 - Hybrid with 1 batteries has been found at: 192.168.1.105
2025-04-10 07:13:56,935 - startup     -  [INFO    ] - Setting up invertor: 1
2025-04-10 07:13:56,964 - startup     -  [INFO    ] - ==============================================================
2025-04-10 07:13:56,965 - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2025-04-10 07:13:56,965 - startup     -  [INFO    ] - ====     http://192.168.1.103:8099/config.html             ====
2025-04-10 07:13:56,965 - startup     -  [INFO    ] - ==============================================================
2025-04-10 07:13:56,966 - startup     -  [INFO    ] - Running Invertor 1 (**********) read loop every 30/120s
[2025-04-10 07:13:57 +0100] [70] [INFO] Starting gunicorn 23.0.0
[2025-04-10 07:13:57 +0100] [70] [INFO] Listening at: http://0.0.0.0:6350 (70)
[2025-04-10 07:13:57 +0100] [70] [INFO] Using worker: sync
[2025-04-10 07:13:57 +0100] [82] [INFO] Booting worker with pid: 82
[2025-04-10 07:13:57 +0100] [81] [INFO] Starting gunicorn 23.0.0
[2025-04-10 07:13:57 +0100] [81] [INFO] Listening at: http://0.0.0.0:6345 (81)
[2025-04-10 07:13:57 +0100] [81] [INFO] Using worker: sync
[2025-04-10 07:13:57 +0100] [83] [INFO] Booting worker with pid: 83
[2025-04-10 07:13:57 +0100] [84] [INFO] Booting worker with pid: 84
[2025-04-10 07:13:57 +0100] [85] [INFO] Booting worker with pid: 85
2025-04-10 07:13:57,735 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-04-10 07:13:57,735 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.1.105
2025-04-10 07:13:57,739 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-04-10 07:14:05,188 - GivTCP - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
```

It doesn't seem to be causing any issues but I thought I would report it just in case.

Thanks.

(I have a GivEnergy 5kW Hybrid Inverter Gen3 with 9.5kWh Battery Gen2)

britkat1980, 2025-04-11:
Thanks for this, Export (Paused) is a new state, and I'd missed adding that to the right LUT. I've added in upcoming release (its in Dev 3.0.54 if you want to test).
On the MQTT error, the autodetected mqtt username and password could have been left. If you've changed it that may be why it's logging the fact as it hasn't autodetected it running

britkat1980, 2025-04-14:
fixed in 3.1.6

closed 2025-04-14 by britkat1980

IWR69, 2025-04-14:
Funny enough I’ve just upgraded and about to test it now for you.
I'll let you know how it goes...

IWR69, 2025-04-14:
I can confirm there are no new system log entries for this.
Thanks for your help.