britkat1980/giv_tcp #369: How to set dynamic tariffs

sunilnc, 2025-04-10:
Hello,

Hoping someone can help me.

I'm an Octopus EV customer with peak (£0.29) and off peak (£0.07) rates.

I've set up the Dynamic tariff to reflect these different costs and the times:

[image]

On the Energy tab, I'm using **GivTCP [my inverter serial] Energy GivTCP Current Rate** entity to track costs, but this only tracks the peak rate of £0.29.

Which entity do I need to change it to so that it tracks both the peak AND off-peak rates for accurate costs?

[image]

gcoan, 2025-04-10:
I've personally never used the givtcp tariff stuff at all so can't advise how it works.

Originally I used to use a utility meter for 'today' consumption with day and night tariffs setup,then an automation that would swap the utility meter from one tariff to another.

This has the advantage of giving you two sensors, a day consumption and a night consumption sensor for the energy dashboard, so you can see the day/night kWh breakdown.

I have however now simplified to just a single consumption entity and a 'current rate now' entity.  I use the Octopus integration to provide that, but you could just as easily create a helper entity for current rate and create an automation to set it to the right £ value at the appropriate times of the day

AberDino, 2025-04-11:
@sunilnc, a while ago I implemented [this 'Speak to the Geek' setup](https://www.speaktothegeek.co.uk/2023/02/off-peak-tariffs-in-home-assistants-energy-dashboard/), which is what @gcoan is describing above, and it is working well for me (I'm not on Octopus).

closed 2025-04-11 by sunilnc

gcoan, 2025-04-11:
Yes it was that speak to the geek approach I followed. Which was fine when I was on Flux, but then on Agile it didn't make sense and everything was loaded as day rate.  Now I moved back to Cosy I could have reverted back to using it, but to be honest just a single daily sensor GBP value was good enough so I stripped it all out and reloaded my history into the single sensor.