britkat1980/giv_tcp #370: Grid Outage Sensor

lukshan13, 2025-04-11:
I'm not sure if this has been discussed before, but I could not find if there is a sensor for identifying if there is a grid outage? Would be an excellent sensor to have in HA.

gcoan, 2025-04-12:
here's the way most of us do this, create an automation based on grid frequency.  This works on AIO's as well as hybrid inverters with EPS.

Uses a custom script for notifications but you get the idea.  Need to enable uptime (inbuilt integration) to stop false alerts when HA reboots.

Could put the logic in a template binary sensor if you wanted to

```
alias: Grid Status Notification
description: Notifies loss / resumption of grid status to home battery system
triggers:
  - trigger: numeric_state
    id: grid_lost
    entity_id:
      - sensor.givtcp_xxx_grid_frequency
    below: 48
    for:
      minutes: 1
  - trigger: numeric_state
    id: grid_resumed
    entity_id:
      - sensor.givtcp_xxx_grid_frequency
    above: 48
    for:
      minutes: 1
conditions:
  - alias: Uptime is more than 5 minutes
    condition: template
    value_template: >-
      {{ ( as_timestamp(now()) - as_timestamp(states('sensor.uptime')) ) >= 300
      }}
actions:
  - if:
      - condition: trigger
        id:
          - grid_lost
    then:
      - data:
          title: Power Cut
          message: >
            Connection to the grid lost - house is on battery. Current load is
            {{ states('sensor.givtcp_xxx_load_power')|int }}kW. Current SoC is
            {{ states('sensor.givtcp_xxx_soc')|int }}%. Battery will last
            approximately {{              
            ((states('sensor.givtcp_xxx_discharge_time_remaining')|int)/60)|int              
            }} hours.
          critical: "Y"
        action: script.notify_all_devices
  - if:
      - condition: trigger
        id:
          - grid_resumed
    then:
      - data:
          title: Grid Power Restored
          message: >
            Connection to the grid resumed. Current load is {{
            states('sensor.givtcp_xxx_load_power')|int }}kW. Current SoC is {{
            states('sensor.givtcp_xxx_soc')|int }}%.
        action: script.notify_all_devices
mode: single
```

lukshan13, 2025-04-12:
Completely fair enough! I've actually got a automation to send me a push when the frenquency drops out of normal range, I was just hoping there was some way to actually get the state of the gateway, whether it's activily disconnecting the grid or returned to normal. Seemed more robust to me, thats all

closed 2025-10-23 by britkat1980