britkat1980/giv_tcp #375: Force Charge switches off export from battery but does not charge battery from grid

matthewturner, 2025-04-15:
I think this might be a problem with a recent GivEnergy firmware rather than giv_tcp but I'm putting it here in case it helps anyone.

I have a Gateway with one AIO and have recently updated the firmware to the latest (A0.010 and D0.616-A0.616 respectively).

Enabling Force Charge used to charge the battery at full power for the specified duration. It now simply stops feeding the house from the battery and switches to grid.

I've tested under both giv_tcp v2 and v3 and I'm experiencing the same behaviour with both which leads me to think it's an underlying firmware issue.

gcoan, 2025-04-15:
worth checking that AC charge enable is turned on and check what your AC charge limit is set to, if that is lower than current SoC then the AIO won't charge.

gcoan, 2025-04-15:
Yeah charging at 0% isn't going to get you anywhere 

Glad to hear you are sorted