britkat1980/giv_tcp #376: Failed to get Cache Error

AberDino, 2025-04-15:
Currently running GivTCP-DEV 3.1.8, and overnight I had the following three log entries:
```
2025-04-15 00:48:21,505 - AIO-1 - GivLUT      -  [ERROR   ] - Failed to get Cache: (<class 'EOFError'>, EOFError('Ran out of input'), <traceback object at 0x7f82a13cc0>)
2025-04-15 00:48:21,519 - AIO-1 - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'FileNotFoundError'>, FileNotFoundError(2, 'No such file or directory'), <traceback object at 0x7f8601e840>), 2017)
2025-04-15 00:48:21,520 - AIO-1 - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2130)
```

Does anybody know what that means? Is it something to be concerned about? Thanks

AberDino, 2025-05-05:
Now running GivTCP-BETA v3.1.9 and the following "Failed to get Cache" error has occurred twice in the last 24 hours (no other errors):

`2025-05-05 01:27:16,841 - AIO-1 - GivLUT      -  [ERROR   ] - Failed to get Cache: (<class 'FileNotFoundError'>, FileNotFoundError(2, 'No such file or directory'), <traceback object at 0x7f9d1fa7c0>)`

britkat1980, 2026-10-11:
Closing this as it's had no activity for over 8 weeks.

GivTCP 3.6 has since been released ([latest: v3.6.2](https://github.com/britkat1980/giv_tcp/releases/latest)). It moves to a new Modbus library and includes a large number of fixes, so this may well behave differently now. If you still see this on 3.6.2, please reopen it, or open a new issue, with a debug log.

closed 2026-10-11 by britkat1980