britkat1980/giv_tcp #377: Time read in wrong format???

tonyvfield, 2025-04-15:
Hi,
I am getting errors since I upgraded to version 3.1.6 from my previous version 2.x. When the date is read from the inverters it seems to have an extra '.' ie full stop in the time. 
 [ERROR   ] - inverter Update failed so using last known good data from cache: ((<class 'ValueError'>, ValueError("time data '11.:30' does not match format '%H:%M'"), <traceback object at 0x7f911d39c0>), 1986)
2025-04-14 17:27:54,388 - GivTCP - read        -  [ERROR   ] - runAll2 Error processing registers: ('KeyError', 'read.py', 2124)

I have 2 inverters (gen 1 3kw) each with a 9.5kwh battery.

I have stopped givtcp, then gone to each inverter and given them the current time, then deleted all the pkl files in the givtcp folder. Restarted Givtcp but still I get the problem.

I am new to HA so please make any suggestions in a simple way. However the batteries and HA have been installed for 2 years and work well. I am running the balancing script from willy rickett and also the force discharge to empty the batteries at the end of the day. All has been fine till now.

[gictcp after restart.txt](https://github.com/user-attachments/files/19754556/gictcp.after.restart.txt)

gcoan, 2025-04-15:
it looks like a date/time corruption on your inverters, similar to what happened on 31/12/24, but strange you are now seeing it

Can you check what the actual time on your inverters is set to?

use the BBC app 
[image]

search on the app store for givenergy if you don't have it

you can also synchronise the time on your inverters from the givenergy portal

tonyvfield, 2025-04-15:
> it looks like a date/time corruption on your inverters, similar to what happened on 31/12/24, but strange you are now seeing it
>
> Can you check what the actual time on your inverters is set to?
>
> use [...]

I do have the BBC ap, but I messed up setting it up and have never found how to change the config. Any ideas?
Having said that, I just fired it up and the is stuck saying connecting

gcoan, 2025-04-15:
> I do have the BBC ap, but I messed up setting it up and have never found how to change the config. Any ideas?
> Having said that, I just fired it up and the is stuck saying connecting

its probably not connecting if the config is messed up.

On my ipad, there is a folder called inverter for the app, and a file called inverter.dat

it appears to contain the IP addresses of my inverters
[image]

so you may be able to edit that and fix the config issue,  or just delete the file and it should prompt you to re-create the config

tonyvfield, 2025-04-15:
I dont even have that file but the app runs on one inverter which is the issue I had before

closed 2025-10-23 by britkat1980