britkat1980/giv_tcp #378: Force Charge fails as wrong time set in slot 1 start / end

DJ-allthetime, 2025-04-15:
I have been trying to automate a force charge in HomeAssistant, I use the Force Charge option in HA (via MQTT)

When i do this the setting is applied, but the battery fails to charge as the start / end time is set incorrectly (to one hour earlier)- for example if force charge is set to 10 minutes and it is 13:10, the start time in Slot 1  is set as 12:10, and the end time set to 12:20

If i manually reset the time for the end of slot 1 to be correct (e.g. 13.20 in the example above) it charges fine

I can only assume this is connected somehow with timezones and BST (im in UK), the GivTCP config has TZ = Europe/London and time is showing correctly in both HA and Givenergy website.

gcoan, 2025-04-15:
Check the time zone of your account in the givenergy portal and do a synchronise time to ensure that your inverter is on BST

DJ-allthetime, 2025-04-17:
Thanks - this was all set correctly, but i have found the problem and solved it - i run GivTCP in docker and hadnt set the TZ for the container - forcing this to also be Europe/London has resolved the problem

closed 2025-04-17 by DJ-allthetime