britkat1980/giv_tcp #380: 10kw inverter communication error loop

Texelo, 2025-04-16:
I've been experiencing this issue for a while across multiple versions. I'm using the gen 3, 1 phase, HV hybrid inverter.

The issue seems to fix itself after a while, sometimes minutes, sometimes hours. I have an automation that restarts the addon, and attempts to restart the inverter via a REST request, although I don't think that's working. 

Restarting the inverter manually through the settings page fixes the issue.

It's worth noting that the inverter is incorrectly recognised as a gen 1: `Gen1 - Hybrid_hv with 3 batteries`

> 2025-04-16 23:47:19,150 - startup     -  [INFO    ] - Running Invertor 1 (...) read loop every 30/180s
> 2025-04-16 23:47:19,156 - startup     -  [INFO    ] - Serving Web Dashboard from port 3000
> [2025-04-16 23:47:19 +0100] [70] [INFO] Starting gunicorn 23.0.0
> [2025-04-16 23:47:19 +0100] [70] [INFO] Listening at: http://0.0.0.0:6350 (70)
> [2025-04-16 23:47:19 +0100] [70] [INFO] Using worker: sync
> [2025-04-16 23:47:19 +0100] [87] [INFO] Booting worker with pid: 87
> [2025-04-16 23:47:19 +0100] [81] [INFO] Starting gunicorn 23.0.0
> [2025-04-16 23:47:19 +0100] [81] [INFO] Listening at: http://0.0.0.0:6345 (81)
> [2025-04-16 23:47:19 +0100] [81] [INFO] Using worker: sync
> [2025-04-16 23:47:19 +0100] [88] [INFO] Booting worker with pid: 88
> [2025-04-16 23:47:19 +0100] [89] [INFO] Booting worker with pid: 89
> [2025-04-16 23:47:19 +0100] [90] [INFO] Booting worker with pid: 90
> 2025-04-16 23:47:21,928 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
> 2025-04-16 23:47:21,929 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.0.189
> 2025-04-16 23:47:23,932 - GivTCP - GivLUT      -  [ERROR   ] - Error in getting Modbus Connection: ('CommunicationError', 'GivLUT.py', 29)
> 2025-04-16 23:47:23,933 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
> 2025-04-16 23:47:23,934 - GivTCP - read        -  [ERROR   ] - Error in inital detect/refresh: ('AttributeError', 'read.py', 77)
> 2025-04-16 23:47:23,934 - GivTCP - read        -  [ERROR   ] - Error in self_run. Re-running watch_plant: ('AttributeError', 'read.py', 2191)
> 2025-04-16 23:47:25,937 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
> 2025-04-16 23:47:25,937 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.0.189
> 2025-04-16 23:47:27,941 - GivTCP - GivLUT      -  [ERROR   ] - Error in getting Modbus Connection: ('CommunicationError', 'GivLUT.py', 29)
> 2025-04-16 23:47:27,941 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
> 2025-04-16 23:47:27,942 - GivTCP - read        -  [ERROR   ] - Error in inital detect/refresh: ('AttributeError', 'read.py', 77)
> 2025-04-16 23:47:27,943 - GivTCP - read        -  [ERROR   ] - Error in self_run. Re-running watch_plant: ('AttributeError', 'read.py', 2191)
> 2025-04-16 23:47:29,948 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
> 2025-04-16 23:47:29,948 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.0.189
> 2025-04-16 23:47:31,950 - GivTCP - GivLUT      -  [ERROR   ] - Error in getting Modbus Connection: ('CommunicationError', 'GivLUT.py', 29)
> 2025-04-16 23:47:31,950 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
> 2025-04-16 23:47:31,951 - GivTCP - read        -  [ERROR   ] - Error in inital detect/refresh: ('AttributeError', 'read.py', 77)
> 2025-04-16 23:47:31,951 - GivTCP - read        -  [ERROR   ] - Error in self_run. Re-running watch_plant: ('AttributeError', 'read.py', 2191)
> 2025-04-16 23:47:33,953 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...

leocrawford, 2025-04-30:
In your HA Core logs, do you also have something like?

`Invalid option for select.givtcp_fd2320g056_battery_calibration: 'enable' (valid options: ['Off', 'Start', '', 'Charge Only'])`

Texelo, 2025-04-30:
@leocrawford I do!

```
Logger: homeassistant.components.mqtt.select
Source: components/mqtt/select.py:125
integration: MQTT (documentation, issues)
First occurred: April 17, 2025 at 12:40:57 AM (31856 occurrences)
Last logged: 10:35:11 AM

Invalid option for select.givtcp_fh2442g134_battery_calibration: 'disable' (valid options: ['Off', 'Start', '', 'Charge Only'])
```

leocrawford, 2025-04-30:
I suspect that is the likely cause of the problem then, somewhere in the stack the battery_calibration values are misaligned causing an error.

Texelo, 2025-05-01:
Yeah definitely something to do with that, I'm not at home this week so can't use my pc to dig into the code, but the communication errors begin at the same time as that error. Next time it occurs, I'm going to try and change the battery_calibration value and see if it fixes itself (I'm guessing probably not because it's already lost communication)

I wonder if that value is set by the inverter, and then givtcp doesn't know what to do and gets in a pickle,  or the other way around. I'm guessing restarting the inverter resets the battery calibration so that they can start speaking again.

TheSargey, 2025-05-20:
Good morning folks - I've just got the HV inverter installed a week ago and have exactly the same issue, although I'm not seeing the battery_calibration error in the HA Core logs (perhaps I've not got the right level of error tracking on though?).

Have you tried GivTCP 2.4.9?  With that, it will connect but then mine begins to timeout --

 UPDATE  The latest version of `serve` is 14.2.4
 INFO  Accepting connections at http://localhost:3042
2025-05-20 09:29:31,133 - Inv1 - sync        -  [ERROR   ] - Connection to (192.168.1.236, 8899) failed: timed out

Aside to this I'm using the rubikcube inverter monitor (https://play.google.com/store/apps/details?id=uk.rubikcube.invertermonitor&hl=en_GB) and this seems to be able to maintain a relatively solid connection when GivTCP can't.  

Did you guys find any work around other than the inverter restart?

gcoan, 2025-05-20:
> Have you tried GivTCP 2.4.9? With that, it will connect but then mine begins to timeout --

I doubt you will get anywhere with GivTCP v2.  All focus has been on v3 for some time, and GivEnergy gave Mark access to some of the newer kit (like the 3 phase) and those subsequent developments came in v3.

Stick to v3, maybe try the beta branch for latest (stable) changes

TheSargey, 2025-05-21:
> I doubt you will get anywhere with GivTCP v2. All focus has been on v3 for some time, and GivEnergy gave Mark access to some of the newer kit (like the 3 phase) and those subsequent developments [...]

Agreed, I just wanted to try V2 in case it spat out a more verbose error!  Thanks as always.

leocrawford, 2025-05-21:
I'm running  3.1.9 (beta) and the very latest firmware on the Gen3 inverter, and currently having none of the errors above

TheSargey, 2025-05-21:
> I'm running 3.1.9 (beta) and the very latest firmware on the Gen3 inverter, and currently having none of the errors above

Sounds promising mate, excuse my ignorance, what firmware version are you on and is there a guide how to install the beta? I can't find a way to do it.

Many thanks for yiur help mate.

Sarge

leocrawford, 2025-05-21:
The beta is a different addon, you can install both at same time - but only
use one at a time. My firmware is D0.316-A0.316, Modbus 1.40 and Battery
firmware 4009

TheSargey, 2025-05-22:
> The beta is a different addon, you can install both at same time - but only
> use one at a time. My firmware is D0.316-A0.316, Modbus 1.40 and Battery
> firmware 4009
> […](#)

Thanks Leo - interesting, mine looks very different:

Inverter Model
GIV-HY-10.0-G3-HV
Software Version
D0.004-A0.004-A1030.004
Battery Firmware Version
7

Have you got the same inverter model?

MrSpongy, 2025-06-22:
@Texelo Did you contact Giv support about this, and did they have any idea/solution?

Texelo, 2025-06-23:
Unfortunately I've not, too many things to do in a day! From some forums I've seen about similar issues with network dropping out on the inverter, it could be a dongle firmware issue (even when using ethernat), but I'm not sure which firmware to use, so haven't attempted updating - calling support would be a good idea. 

My automation for restarting things seems to be working for now, so it's not a massive priority of mine. The automation restarts the inverter via REST request and cloud integration every 5 mins until it's fixed, a bit of a sledgehammer approach.

MrSpongy, 2025-06-23:
I did recently update the dongle firmware as it said there was an update available through the cloud dashboard. Still getting the same issue though, so the latest dongle firmware doesn't fix it unfortunately.

Texelo, 2025-06-25:
> I did recently update the dongle firmware as it said there was an update available through the cloud dashboard. Still getting the same issue though, so the latest dongle firmware doesn't fix it unfortunately.

I had no idea the dingle could be updated via the cloud dashboard, so thanks for that, updating battery to firmware 7 too.

It's very possible this is a givtcp issue as we're the only ones in aware of with this inverter, and both have identical issues, but I have no idea where to start debugging

ben209car, 2025-08-11:
I'm having the same issue.  There no rhyme or reason as to when it stops working.  Hopefully, there will be an update to sort this issue in the future.

gcoan, 2025-08-28:
> Will try the GivTCP Beta and see if any improvement.

The Beta is actually an older version of givtcp at the moment, 3.1.9., it hasn't been upgraded to 3.2 which is  the new main release released earlier this week.
There's a couple of bug fixes on the dev release but its pretty much the same

MarkWestGit, 2025-08-31:
I'm getting this error too, and notice that when the inverter goes offline Givtcp loops repeatedly trying to read. I have a 5 kW Gen3 inverter (model GIV-HY-5.0-AU) and it's connected directly to my router (Deco X55 mesh) by ethernet cable. I have HA running on a Green. It never drops connection to the router. Both the Green and the inverter are on static IP addresses. With my limited IT I don't know what the best way to systematically check things is, but I'd suspect some issue with the inverter rather than with my network or Givtcp, so hopefully like you say an update will fix this. 

Sometimes I get this:
2025-08-31 12:00:14,085 - GivTCP - read        -  [ERROR   ] - 5 consecutive timeout errors in watch loop. Restarting modbus connection:
2025-08-31 12:00:38,369 - GivTCP - read        -  [ERROR   ] - 5 consecutive timeout errors in watch loop. Restarting modbus connection:
2025-08-31 12:01:02,654 - GivTCP - read        -  [ERROR   ] - 5 consecutive timeout errors in watch loop. Restarting modbus connection:
2025-08-31 12:01:28,195 - GivTCP - read        -  [ERROR   ] - 5 consecutive timeout errors in watch loop. Restarting modbus connection:
2025-08-31 12:01:52,480 - GivTCP - read        -  [ERROR   ] - 5 consecutive timeout errors in watch loop. Restarting modbus connection:
2025-08-31 12:02:16,764 - GivTCP - read        -  [ERROR   ] - 5 consecutive timeout errors in watch loop. Restarting modbus connection:
2025-08-31 12:02:41,049 - GivTCP - read        -  [ERROR   ] - 5 consecutive timeout errors in watch loop. Restarting modbus connection:
2025-08-31 12:03:05,336 - GivTCP - read        -  [ERROR   ] - 5 consecutive timeout errors in watch loop. Restarting modbus connection:
2025-08-31 12:03:30,877 - GivTCP - read        -  [ERROR   ] - 5 consecutive timeout errors in watch loop. Restarting modbus connection:

Sometimes I get this:
2025-08-30 19:43:41,829 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-08-30 19:43:41,830 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.68.64
2025-08-30 19:43:41,834 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-08-30 19:44:20,378 - GivTCP - read        -  [ERROR   ] - Error in inital detect/refresh: ('TimeoutError', 'read.py', 77)
2025-08-30 19:44:20,380 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-08-30 19:44:20,381 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.68.64
2025-08-30 19:44:20,385 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-08-30 19:44:39,900 - startup     -  [ERROR   ] - Self Run loop process stuck. Killing and restarting...
2025-08-30 19:44:39,901 - startup     -  [INFO    ] - Restarting Invertor read loop every 60s
2025-08-30 19:44:41,886 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-08-30 19:44:41,887 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.68.64
2025-08-30 19:44:41,891 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-08-30 19:45:20,437 - GivTCP - read        -  [ERROR   ] - Error in inital detect/refresh: ('TimeoutError', 'read.py', 77)
2025-08-30 19:45:20,438 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-08-30 19:45:20,438 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.68.64
2025-08-30 19:45:20,442 - GivTCP - base_events -  [ERROR   ] - Task exception was never retrieved
future: <Task finished name='network_consumer' coro=<Client._task_network_consumer() done, defined at /app/GivTCP_1/givenergy_modbus_async/client/client.py:330> exception=ConnectionResetError(104, 'Connection reset by peer')>
Traceback (most recent call last):
  File "/app/GivTCP_1/givenergy_modbus_async/client/client.py", line 378, in _task_network_consumer
    await self.close()
  File "/app/GivTCP_1/givenergy_modbus_async/client/client.py", line 98, in close
    await self.writer.wait_closed()
  File "/usr/local/lib/python3.13/asyncio/streams.py", line 358, in wait_closed
    await self._protocol._get_close_waiter(self)
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1005, in _read_ready__data_received
    data = self._sock.recv(self.max_size)
ConnectionResetError: [Errno 104] Connection reset by peer
2025-08-30 19:45:20,451 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-08-30 19:45:39,905 - startup     -  [ERROR   ] - Self Run loop process stuck. Killing and restarting...
2025-08-30 19:45:39,905 - startup     -  [INFO    ] - Restarting Invertor read loop every 60s
2025-08-30 19:45:41,880 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-08-30 19:45:41,881 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.68.64
2025-08-30 19:45:41,885 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-08-30 19:46:20,433 - GivTCP - read        -  [ERROR   ] - Error in inital detect/refresh: ('TimeoutError', 'read.py', 77)

A successful startup looks like this:

2025-08-29 16:15:40,989 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-08-29 16:15:41,051 - startup     -  [INFO    ] - Network Found: 192.168.68.78/22
2025-08-29 16:15:43,055 - startup     -  [INFO    ] - Searching for Inverters
[... cut ...]
[2025-08-29 16:17:29 +1200] [84] [INFO] Booting worker with pid: 84
[2025-08-29 16:17:29 +1200] [85] [INFO] Booting worker with pid: 85
2025-08-29 16:17:32,269 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-08-29 16:17:32,271 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.68.64
2025-08-29 16:17:32,276 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-08-29 16:18:23,301 - GivTCP - read        -  [INFO    ] - Publishing Home Assistant Discovery messages

MarkWestGit, 2025-10-15:
> I have another issue alongside this. When MQTT and GIVTCP are ‘on’ the local control for my givenergy app messes up. The home setting doesn’t find my inverter. The away setting works fine. I have to stop MQTT and GIVTCP and reboot the inverter then it works fine again. Not sure if this issue is linked.
> 
> My inverter is connected via LAN not WiFi. It seems a lot of people have this issue when connected via LAN rather than WiFi.

Paul L from GivEnergy who is a legend on the GivTCP Facebook forum, said that it wasn't possible for the inverter to be read locally by two means (or some more IT-savvy way of putting it) so when GivTCP is running then you're not able to see the "Home" reads on the GivEnergy App, only the "Away" ones, so what you see is quite expected

gcoan, 2025-10-15:
> Paul L from GivEnergy who is a legend on the GivTCP Facebook forum, said that it wasn't possible for the inverter to be read locally by two means (or some more IT-savvy way of putting it) so when [...]

that seems a rather step backwards because the older inverters (AIO's, Hybrids and AC coupled) all support multiple connections in parallel with absolutely no issue.  And these are much older and slower CPUs in the inverter so whatever has been done in the HV?

Only issue I have ever had with multiple connections is if you set givtcp to poll too aggressively it can lock the inverter into responding to givtcp all the time and it fails to respond to portal comms.
Hence set givtcp to a 30 second autorun frequency and everything is sweet

AcidClould, 2025-10-19:
Yes vs 3.3

MarkWestGit, 2025-10-22:
> I've noticed this morning an update for the WiFi dongle through the GivEnergy cloud for my 10kw HV. (Now on version 1203)
> 
> I've updated that and GIVTCP Dev.
> 
> Giv TCP now correctly identified my inverter, so will run this for a bit and see if any errors come up.

I'm still on 1203 for a 5 kW Gen 3 hybrid. What's the new version number? I don't see any updates on my portal

Texelo, 2025-10-24:
I've just updated my inverter software (D0.004-A0.004-A1030.004, dongle firmware 1203) and givtcp to 3.4.1, and it's now being detected as a Hybrid_hv_gen3, which seems positive! I'll report back over the next day or so on if I still get lots of restarts...

TheSargey, 2025-10-25:
> I'm an Installer so I've managed to update my inverter and battery manually to the absolute latest version:
> 
> D0.005-A0.005-A1030.005
> 
> Dongle: 1203
> 
> GivTCP DEV: Latest available version
> 
> Still seeing the errors.
> 
> I think this may be a GivEnergy issue, as I've not been using Home Assistant for a while because of this error.
> 
> I get issues within the GivEnergy App.
> 
> 'Away' mode works all of the time. 'Home' will work after an inverter reset, then after what seems like a random amount of time (same as HA), will stop working and only work on 'Away', until the inverter is reset by the Cloud Dashboard. Then 'Home' will work absolutely fine again.
> 
> Which makes me believe it is an issue with the actual inverter local modbus connection.

You can try using the ge_cloud setup method within Home Assistant/Predbat which gets them working (if that's what you're using it for). Works flawlessly for me.

Sarge

MarkT439, 2025-10-28:
Hi 

I have the same issue as Texelo. I have a GIV-3PHY-11.0-HV inverter which is detected by GivTCP-BETA as: 

Gen 3 - Hybrid_3ph

I am using GivTCP-BETA 3.3.15 on a RPI 4 with HAOS 16.2. I am not using Predbat

When I start GivTCP-BETA it detects my inverter and runs for a few hours before it stops updating a dashboard displaying solar generation, battery level, export power etc.

Below is the GivTCP-BETA log, which shows a typical failure. I have an automation set up which restarts the add-on if no update is detected for 15 mins.

Summary:
14:52:07 Started GIVTCP-BETA addon
16:00:31 Successful write to inverter
22:50:51 First modbus eeror detected
23:03:05 No updates for 15 mins, add-on restarted.

Full log:
2025-10-27 14:52:07,718 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-10-27 14:52:09,777 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-10-27 14:52:09,778 - startup     -  [INFO    ] - Scanning network: 192.168.1.199/24
2025-10-27 14:52:15,550 - client      -  [INFO    ] - Connection established to 192.168.1.116:8899
2025-10-27 14:52:15,551 - client      -  [INFO    ] - Detecting plant
2025-10-27 14:52:16,719 - client      -  [INFO    ] - Plant Detected
2025-10-27 14:52:22,749 - startup     -  [INFO    ] - Inverter TDxxxxxxxx which is a Hybrid_3ph with 4 batteries and 1 meters has been found at: 192.168.1.116
2025-10-27 14:52:22,772 - startup     -  [INFO    ] - Setting up invertor: 1
2025-10-27 14:52:23,023 - startup     -  [INFO    ] - ==============================================================
2025-10-27 14:52:23,024 - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2025-10-27 14:52:23,025 - startup     -  [INFO    ] - ====     http://192.168.1.199:8099/config.html             ====
2025-10-27 14:52:23,033 - startup     -  [INFO    ] - ==============================================================
2025-10-27 14:52:23,038 - startup     -  [INFO    ] - Running Invertor 1 (TDxxxxxxxx) read loop every 60/120s
[2025-10-27 14:52:23 +0000] [70] [INFO] Starting gunicorn 23.0.0
[2025-10-27 14:52:23 +0000] [70] [INFO] Listening at: http://0.0.0.0:6350 (70)
[2025-10-27 14:52:23 +0000] [70] [INFO] Using worker: sync
[2025-10-27 14:52:24 +0000] [82] [INFO] Booting worker with pid: 82
[2025-10-27 14:52:24 +0000] [81] [INFO] Starting gunicorn 23.0.0
[2025-10-27 14:52:24 +0000] [81] [INFO] Listening at: http://0.0.0.0:6345 (81)
[2025-10-27 14:52:24 +0000] [81] [INFO] Using worker: sync
[2025-10-27 14:52:24 +0000] [83] [INFO] Booting worker with pid: 83
[2025-10-27 14:52:24 +0000] [84] [INFO] Booting worker with pid: 84
[2025-10-27 14:52:24 +0000] [85] [INFO] Booting worker with pid: 85
2025-10-27 14:52:26,656 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-10-27 14:52:26,658 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.1.116
2025-10-27 14:52:26,662 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-10-27 14:52:48,530 - GivTCP - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
2025-10-27 16:00:31,566 - GivTCP - write       -  [INFO    ] - Setting Charge Slot End 1 to: 23:34 was a success
2025-10-27 16:31:34,935 - GivTCP - write       -  [INFO    ] - Setting Charge Slot End 1 to: 05:28 was a success
2025-10-27 22:50:51,539 - GivTCP - read        -  [ERROR   ] - 5 consecutive timeout errors in watch loop. Restarting modbus connection:
2025-10-27 22:50:51,540 - GivTCP - read        -  [ERROR   ] - Error in self_run. Re-running watch_plant: ('CancelledError', 'read.py', 2251)
2025-10-27 22:50:53,543 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-10-27 22:50:53,544 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.1.116
2025-10-27 22:50:53,548 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-10-27 22:51:32,099 - GivTCP - read        -  [ERROR   ] - TimeoutError: 
2025-10-27 22:51:32,100 - GivTCP - read        -  [ERROR   ] - Error in inital detect/refresh: ('TimeoutError', 'read.py', 82)
2025-10-27 22:51:32,101 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-10-27 22:51:32,103 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.1.116
2025-10-27 22:51:32,106 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-10-27 22:52:10,667 - GivTCP - read        -  [ERROR   ] - TimeoutError: 
2025-10-27 22:52:10,668 - GivTCP - read        -  [ERROR   ] - Error in inital detect/refresh: ('TimeoutError', 'read.py', 82)
2025-10-27 22:52:10,670 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-10-27 22:52:10,670 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.1.116
2025-10-27 22:52:10,673 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-10-27 22:52:49,244 - GivTCP - read        -  [ERROR   ] - TimeoutError: 
2025-10-27 22:52:49,245 - GivTCP - read        -  [ERROR   ] - Error in inital detect/refresh: ('TimeoutError', 'read.py', 82)
2025-10-27 22:52:49,246 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-10-27 22:52:49,247 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.1.116
2025-10-27 22:52:49,250 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
[... cut ...]
2025-10-28 05:05:42,237 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-10-28 05:05:42,238 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.1.116
2025-10-28 05:05:42,241 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-10-28 05:30:18,625 - GivTCP - read        -  [INFO    ] - Saving current energy stats at start of day rate tariff (Dynamic)
2025-10-28 05:31:34,948 - GivTCP - write       -  [INFO    ] - Setting Charge Slot End 1 to: 05:28 was a success

Thanks

richardsg307, 2025-12-10:
> I'm an Installer so I've managed to update my inverter and battery manually to the absolute latest version:
> Still seeing the errors.
>... 
> I think this may be a GivEnergy issue, as I've not been using Home Assistant for a while because of this error.
> 
> I get issues within the GivEnergy App.
> 
> 'Away' mode works all of the time. 'Home' will work after an inverter reset, then after what seems like a random amount of time (same as HA), will stop working and only work on 'Away', until the inverter is reset by the Cloud Dashboard. Then 'Home' will work absolutely fine again.
> 
> Which makes me believe it is an issue with the actual inverter local modbus connection.

I tend to agree. I got my system installed 2 days ago and got GivTCP up and running that evening. It and the App have lost communication 3 times since then requiring a reset of the Inverter through its Web portal (haven't found the cloud method yet). I reported it on thread #447 before reading this one.

Is there any way of forcing a restart automatically from Home Assistant?

Earnestly, 2025-12-11:
I have taken to monitoring givtcp log output and using selected messages as sentinels to trigger a restart. The restart typically only needs to apply to the networking module. A full isolator power cycle has been needed once so far.

I do disable the givtcp addon during the restart process as that can sometimes step on toes.

```sh
curl -gsH 'Referer: http://inverter.lan/restart_en.html' -d HF_PROCESS_CMD=RESTART -u username:password http://inverter.lan
```

richardsg307, 2025-12-11:
> I have taken to monitoring givtcp log output and using selected messages as sentinels to trigger a restart. The restart typically only needs to apply to the networking module. A full isolator power cycle has been needed once so far.
> 
> I do disable the givtcp addon during the restart process as that can sometimes step on toes.
> 
> curl -gsH 'Referer: http://inverter.lan/restart_en.html' -d HF_PROCESS_CMD=RESTART -u username:password http://inverter.lan

If you are doing this from within Home Assistant it would be good to see the full yaml for the automation.

So does your command only reset the Network module?

I'd like to be able to get only the PV power more frequently (every 5s maybe) and am wondering if there is a Rest/Curl command that would achieve that for me?

Blimey, I've only had the system 2 days!

MarkWestGit, 2025-12-11:
> If you are doing this from within Home Assistant it would be good to see the full yaml for the automation.
>
> So does your command only reset the Network module?
>
> I'd like to be able to get only the [...]

I'd like to see the. yaml for that too please. I've had no success resolving the frequent timeouts with the modbus connection. I can ping the inverter when GivTCP is failing to open the modbus connection, and the inverter is online as well, apart from some longer timeouts where pinging eventually stops, and the inverter goes offline

Earnestly, 2025-12-12:
In my experience the restart is sometimes needed twice in succession but it then remains stable for roughly 6 hours (ranges from 1h to 12h).

On the HA side of things I initially added a button to restart the Inverter manually. This was done by adding to `/homeassistant/configuration.yaml` the following:

```yaml
rest_command:
  restart_pv_network_module:
    url: http://inverter.lan
    method: post
    payload: HF_PROCESS_CMD=RESTART
    authentication: basic
    username: user
    password: pass
    verify_ssl: false
```

With a dashboard card using essentially:

```yaml
- type: button
  show_name: true
  show_icon: true
  tap_action:
  action: perform-action
  perform_action: rest_command.restart_pv_network_module
  icon: mdi:button-pointer
  name: Restart PV
  show_state: false
```

The main difficulty with doing everything in HA is that it's not entirely obvious how you would use `ha addon` tooling to monitor addon logs from within. It appears that this stuff is accessible through a python API but I'm not at all familiar with it.

britkat1980, 2026-10-11:
Closing this as it's had no activity for over 8 weeks.

GivTCP 3.6 has since been released ([latest: v3.6.2](https://github.com/britkat1980/giv_tcp/releases/latest)). It moves to a new Modbus library and includes a large number of fixes, so this may well behave differently now. If you still see this on 3.6.2, please reopen it, or open a new issue, with a debug log.

closed 2026-10-11 by britkat1980