britkat1980/giv_tcp #383: REST cache read error.

johnwb87, 2025-04-20:
Seeing predbat lose the REST connection to GivTCP.

GivTCP log from the same time is below. 

`2025-04-20 19:14:46,170 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-04-20 19:14:46,199 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)`

pacemaker82, 2025-05-02:
any fix for this ? being reported over and over and no resolution I can find

gcoan, 2025-05-02:
I see the same errors myself

GivTCP reports

2025-05-02 11:55:35,726 - H - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-02 11:55:35,737 - H - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)

Predbat gets a REST error, deletes the PKL files and restarts GivTCP which the works for a while before repeating again

pacemaker82, 2025-05-03:
@gcoan is there something special I need to do for Predbat to delete the PKL file? I'm on 8.19.1 and predbat was frozen since the time it happened at 1.45am until I rolled back to GivTCP 2.4.9, seemingly not deleting the PKL file if that is the cause

gcoan, 2025-05-03:
I don't know @pacemaker82 

I'm getting more of these errors all of a sudden.  I found that stopping predbat, stopping GivTCP then giving GivTCP a clean start (as opposed to a restart), waiting until I saw the inverter date time updates coming through in the HA logfile, then I started predbat 

GivTCP 3 definitely takes longer to start than v2 and sometimes predbat restarts it prematurely. So hence the staggered startup

closed 2025-10-23 by britkat1980