britkat1980/giv_tcp #387: dev branch update from 3.1.8 to 3.1.17 failing to connect to MQTT service

Ryuisnod, 2025-04-28:
[533ea71a_givtcp-dev_2025-04-28T11-28-21.622Z.log](https://github.com/user-attachments/files/19939101/533ea71a_givtcp-dev_2025-04-28T11-28-21.622Z.log)

i've been using the dev branch as I have a 3 phase inverter.

at 02:56 my givtcp add-on updated automatically and then stopped working.

Having a look through the logs it appears it doesn't detect the MQTT service. I rolled back to 3.1.8 at 12:10, and everything seems to be working fine.

I've attached 12 hours of logs to cover the update and roll back, although looking at them there appears to be errors before the update but they didn't seem to be impacting usage. been working great with predbatt, with no noticeable issues

Tmp2k, 2025-05-08:
We have exactly the same issue. How do you rollback? I've not found how I can specify the version of an addon to install.

-----LOG----
`2025-05-08 10:18:03,534 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-05-08 10:18:03,548 - startup     -  [INFO    ] - No HA MQTT service has been found. Install and run the Mosquitto addon, or manually configure your own MQTT broker.`

Nothing has changed other than upgrading from 3.1.8 => 3.1.17 we're on the default MQTT config, running default Mosquito server.

Pinging core_mosquitto from the terminal works as expected, all other MQTT functioning correctly.

Tmp2k, 2025-05-08:
I've restored the add-on from a backup so we're now on v3.0.50  however I get the exact same error.  Any suggestions?

Ryuisnod, 2025-05-08:
Did you do mqtt server add-on? I rolled back both from nightly backup's I run. Then restarted system

Tmp2k, 2025-05-08:
So I've moved from dev to main on GivTCP, then I updated mosquito. Now it works. :/

Ryuisnod, 2025-05-08:
I've had so much hassle with updates that I've turned auto upside off on everything. Everything is working so I'm updating nothing going forward

closed 2025-10-23 by britkat1980