britkat1980/giv_tcp #389: GivTCP 3.1.6 throwing ('AttributeError', 'read.py', 2167) since 1.50am

pacemaker82, 2025-05-02:
At 1.45am the charge schedule was applied by Predbat

At 1.50am GivTCP has been throwing this error consistently:

```
2025-05-02 01:44:51,920 - GivTCP - write       -  [INFO    ] - Setting Charge Slot 1 to: 01:30 - 02:00 was a success
2025-05-02 01:44:52,415 - GivTCP - write       -  [INFO    ] - Setting Charge Schedule to enable was a success
2025-05-02 01:50:14,410 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-02 01:50:14,436 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-02 01:50:14,495 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-02 01:50:14,520 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-02 01:55:14,990 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
```

I restarted the add on, but it continues:

```
2025-05-02 08:15:14,880 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-02 08:15:14,905 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-02 08:15:14,959 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-02 08:15:14,981 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
```

I dont know how to proceed at all now, can you help? Everything else looks to be OK on the inverter, GivTCP-wise

apone2000, 2025-05-02:
Mine usually recovers when predbat detects it not working, yesterday though it didn’t as predbat though it was already restarting so it all hung. 

I restarted Givtcp addon and spring back into life, might be worth a try if it’s still giving the error. 

2025-05-01 06:30:07,850 - GivTCP - write       -  [INFO    ] - Setting Battery Pause Mode to PauseCharge was a success
2025-05-01 09:05:03,812 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-01 09:05:03,822 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-01 11:00:03,739 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-01 11:00:03,754 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-01 12:11:27,673 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-01 12:11:27,683 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-01 13:16:27,333 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-01 13:16:27,344 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-01 14:05:04,112 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-01 14:05:04,123 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-01 14:10:04,281 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-01 14:10:04,294 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-01 14:10:04,348 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-01 14:10:04,366 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-01 14:11:27,600 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-01 14:11:27,609 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2167)
2025-05-01 22:09:46,921 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-05-01 22:09:46,948 - startup     -

pacemaker82, 2025-05-02:
I tried restarting the add on and that hasn't resolved the issue for me. I am not sure what else to try? I will restart HA to see if that helps.

pacemaker82, 2025-05-02:
Just to add, rebooting my Home Assistant Device (HA GREEN) also did not resolve this issue.

pacemaker82, 2025-05-02:
disabled 3.1.6 add on and re-enabled 2.4.9 add on has at least resolved the issue and allowing me to use Predbat again

britkat1980, 2025-05-02:
This is a problem with the cache, deleting the regchache.pkl files should restore thing to a workign state. The question is what attribute crapped out and how do I stop it happening again.
I can detect the attribute failure then auto delete pkl to continue

apone2000, 2025-05-02:
I’ve changed my restart code in predbat to

auto_restart:
- shell:
cd /homeassistant/GivTCP || exit 1
timestamp=$(date +%d%H%M)
zip "$timestamp.zip" *-pkl
rm -f *.pkl
- service: hassio/addon_restart addon: aбa2857d_givtcp

I was hoping if I can backup the pkl files then delete that might help look what was wrong in the cache, but not sure yet in the above works or if it will help ;-) if it will I can put some regcache files that get stored before  predbat deletes. 

Also happy if there is a debug version to log more of the cache code?

apone2000, 2025-05-04:
I'm away at the moment so looking at this on iPad so a bit harder to past different logs etc, but from the crash last night the pkl file I copied before predbat at delete them seemed to be 

[040616.pkl.zip](https://github.com/user-attachments/files/20026856/040616.pkl.zip)

 missing the Power_Power_Charge_Completion_Time I believe (may be wrong as hard to decode on a iPad!) the attachment is not zipped, just my iPad wouldn't upload a pkl file

apone2000, 2025-05-04:
If it's the to_date time this may help to check for missing value?

def to_datetime(
    self,
    y: Register,
    m: Register,
    d: Register,
    h: Register,
    min: Register,
    s: Register,
):
    """Combine 6 registers into a datetime, with safe defaults for zeroes and missing values."""
    try:
        year = self.get(y, 0)
        month = self.get(m, 1) or 1
        day = self.get(d, 1) or 1
        hour = self.get(h, 0) or 0
        minute = self.get(min, 0) or 0
        second = self.get(s, 0) or 0

        if None in (year, month, day, hour, minute, second):
            raise ValueError("Missing datetime component")

        return datetime.datetime(
            2000 + year,  # year offset handling
            month,
            day,
            hour,
            minute,
            second,
        )
    except Exception as e:
        _logger.debug(f"Cache: Error processing to_datetime ({e}). Sending Zero Date")
        return datetime.datetime(2000, 1, 1, 0, 0, 0)`

paulgee31, 2025-05-20:
I'm also suffering from this issue, in v3.1.6, I don't know if the RTC returned unknown status is relevant...

2025-05-19 13:28:25,249 - GivTCP - read        -  [ERROR   ] - RTC returned Unknown status (3), keeping last state
2025-05-19 13:28:57,069 - GivTCP - read        -  [ERROR   ] - RTC returned Unknown status (3), keeping last state
2025-05-19 13:29:28,880 - GivTCP - read        -  [ERROR   ] - RTC returned Unknown status (3), keeping last state
2025-05-19 13:30:00,708 - GivTCP - read        -  [ERROR   ] - RTC returned Unknown status (3), keeping last state
2025-05-19 13:30:01,657 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2173)
2025-05-19 13:30:01,664 - GivTCP - read        -  [ERROR   ] - Error getting data from cache: ('AttributeError', 'read.py', 2173)

If there is anything more I can do to help, let me know.

8muirfield, 2025-05-31:
i've turned off predbat, seems to resolve the issue.

apone2000, 2025-06-02:
Interesting, I have tried it by just not deleting the pkl files as there may be cases where Givtcp is unresponsive and it does need restarting still?

gcoan, 2025-07-25:
> I see exactly the same with AC3.0 on D0.212-A0212 and GivTCP Beta 3.1.9. Happy to provide more diagnostics if required.

try commenting givtcp_rest out of apps.yaml to use the HA entities in Predbat and see if it stops the REST errors in predbat and cache read errors in GivTCP.

I have BTW as suggested earlier commented the auto_restart out in predbat and rely on the givtcp error monitor automation to detect and restart givtcp if its stopped polling the inverters.  This works well for me and means predbat isn't restarting givtcp on every rest failure

pacemaker82, 2025-10-20:
I've updated to 3.3 and will see how it does. I have both 2x and 3x add ons installed so can switch back easily if I need to. Will confirm on here.

gcoan, 2025-10-20:
Been running GivTCP v3.3 since about 8pm last night.

Am getting more sets of REST errors, always on the second inverter, none so far on the first.

Errors all look like this with different traceback object id:

```
2025-10-20 07:15:28,482 - H - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7fd52140e900>)
2025-10-20 07:15:28,504 - H - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7fd5213cef00>)
2025-10-20 07:15:28,514 - H - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7fd52144fec0>)
2025-10-20 09:50:01,910 - H - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7fd52144b0c0>)
2025-10-20 09:50:01,918 - H - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7fd52128a180>)
2025-10-20 09:50:01,940 - H - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7fd521411a40>)
2025-10-20 09:50:01,950 - H - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7fd5213bdd40>)
2025-10-20 11:10:01,749 - H - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7fd52157fe00>)
2025-10-20 11:10:01,760 - H - read        -  [ERROR   ] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7fd5213ac380>)
```

gcoan, 2025-10-20:
@britkat1980 I'm still getting REST errors on my second inverter, about every hour or two.  Not a single error on the first inverter.  Strange.

Turned on debug mode, here is the log at the time an error occurred 22:40:

```
2025-10-20 22:39:59,637 - read - [DEBUG] - firstrun exists, so this should already have been run
2025-10-20 22:39:59,637 - read - [DEBUG] - Publish all to MQTT
2025-10-20 22:39:59,637 - mqtt - [DEBUG] - Publishing: GivEnergy/SD2237G395/Battery_Details
2025-10-20 22:39:59,637 - mqtt - [DEBUG] - Prepping BJ2238G271 for publishing
2025-10-20 22:39:59,638 - mqtt - [DEBUG] - Prepping Battery_Stack_1 for publishing
2025-10-20 22:39:59,639 - mqtt - [DEBUG] - Publishing: GivEnergy/SD2237G395/Control
2025-10-20 22:39:59,641 - mqtt - [DEBUG] - Publishing: GivEnergy/SD2237G395/Energy
2025-10-20 22:39:59,641 - mqtt - [DEBUG] - Prepping Rates for publishing
2025-10-20 22:39:59,641 - mqtt - [DEBUG] - Prepping Today for publishing
2025-10-20 22:39:59,642 - mqtt - [DEBUG] - Prepping Total for publishing
2025-10-20 22:39:59,645 - mqtt - [DEBUG] - Publishing: GivEnergy/SD2237G395/Meter_Details
2025-10-20 22:39:59,646 - mqtt - [DEBUG] - Publishing: GivEnergy/SD2237G395/Power
2025-10-20 22:39:59,646 - mqtt - [DEBUG] - Prepping Flows for publishing
2025-10-20 22:39:59,646 - mqtt - [DEBUG] - Prepping Power for publishing
2025-10-20 22:39:59,647 - mqtt - [DEBUG] - Publishing: GivEnergy/SD2237G395/SD2237G395
2025-10-20 22:39:59,648 - mqtt - [DEBUG] - Publishing: GivEnergy/SD2237G395/Stats
2025-10-20 22:39:59,650 - mqtt - [DEBUG] - Publishing: GivEnergy/SD2237G395/Timeslots
2025-10-20 22:39:59,650 - mqtt - [DEBUG] - Publishing: GivEnergy/SD2237G395/raw
2025-10-20 22:39:59,651 - mqtt - [DEBUG] - Prepping BJ2238G271 for publishing
2025-10-20 22:39:59,651 - mqtt - [DEBUG] - Prepping batteries for publishing
2025-10-20 22:39:59,653 - mqtt - [DEBUG] - Prepping invertor for publishing
2025-10-20 22:40:02,798 - read - [ERROR] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7f093421f280>)
2025-10-20 22:40:02,808 - read - [ERROR] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7f093418e3c0>)
2025-10-20 22:40:02,839 - read - [ERROR] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7f0934064a00>)
2025-10-20 22:40:02,855 - read - [ERROR] - Attribute Error: (<class 'AttributeError'>, AttributeError("'NoneType' object has no attribute 'isoformat'"), <traceback object at 0x7f0934233600>)
2025-10-20 22:40:29,984 - read - [DEBUG] - Running partial refresh
2025-10-20 22:40:31,287 - read - [DEBUG] - Data get was successful, now running handler if needed: 
2025-10-20 22:40:31,290 - read - [DEBUG] - Running processData
2025-10-20 22:40:31,295 - read - [DEBUG] - Beginning parsing of Inverter data
2025-10-20 22:40:31,301 - read - [DEBUG] - Getting Total Energy Data
2025-10-20 22:40:31,302 - read - [DEBUG] - Getting Today Energy Data
2025-10-20 22:40:31,302 - read - [DEBUG] - Getting PV Power
2025-10-20 22:40:31,303 - read - [DEBUG] - Getting Grid Power
2025-10-20 22:40:31,304 - read - [DEBUG] - Getting EPS Power
2025-10-20 22:40:31,304 - read - [DEBUG] - Getting PInv Power
2025-10-20 22:40:31,304 - read - [DEBUG] - Getting Load Power
2025-10-20 22:40:31,305 - read - [DEBUG] - Getting Self Consumption Power
2025-10-20 22:40:31,306 - read - [DEBUG] - Getting Solar to H/B/G Power Flows
2025-10-20 22:40:31,306 - read - [DEBUG] - Getting Grid to Battery/House Power Flow
2025-10-20 22:40:31,306 - read - [DEBUG] - Getting mode control figures
2025-10-20 22:40:31,308 - read - [DEBUG] - Saving the battery reserve percentage for later: no need, it's currently at 100 and we don't want to save that.
2025-10-20 22:40:31,308 - read - [DEBUG] - Calculating Mode...
2025-10-20 22:40:31,308 - read - [DEBUG] - Mode is: Eco
2025-10-20 22:40:31,308 - read - [DEBUG] - RTC returned Unknown status (3), keeping last state
2025-10-20 22:40:31,315 - read - [DEBUG] - Force Export is not Running
2025-10-20 22:40:31,316 - read - [DEBUG] - Getting SOC
2025-10-20 22:40:31,316 - read - [DEBUG] - Getting Battery Energy Data
2025-10-20 22:40:31,317 - read - [DEBUG] - Getting Battery Power data
2025-10-20 22:40:31,317 - read - [DEBUG] - Getting Solar to H/B/G Power Flows
2025-10-20 22:40:31,318 - read - [DEBUG] - Getting Battery to House Power Flow
2025-10-20 22:40:31,321 - read - [DEBUG] - Getting Grid to Battery/House Power Flow
2025-10-20 22:40:31,321 - read - [DEBUG] - Getting Battery to Grid Power Flow
2025-10-20 22:40:31,322 - read - [DEBUG] - Getting TimeSlot data
2025-10-20 22:40:31,324 - read - [DEBUG] - Suprious Timeslot data: using last good data
2025-10-20 22:40:31,324 - read - [DEBUG] - Suprious Timeslot data: using last good data
2025-10-20 22:40:31,324 - read - [DEBUG] - Getting inverter Details
2025-10-20 22:40:31,331 - read - [DEBUG] - Getting Battery Details
2025-10-20 22:40:31,332 - read - [DEBUG] - Building battery output: 
2025-10-20 22:40:31,333 - read - [DEBUG] - Battery BJ2238G271 added
2025-10-20 22:40:31,341 - read - [DEBUG] - cache len= 11
2025-10-20 22:40:31,346 - read - [DEBUG] - Running the data cleansing process
2025-10-20 22:40:31,346 - read - [DEBUG] - Data cleansed for: Power
2025-10-20 22:40:31,347 - read - [DEBUG] - Data cleansed for: Flows
[... cut ...]
One thing I notice is that both target SoC and battery reserve are reported several times as having unallowable values.

On inverter 1 I don't get any such errors about target SoC or battery reserve, but I do get the unallowable error on the battery firmware:

```
2025-10-20 22:26:21,802 - read - [DEBUG] - Battery_Firmware_Version is outside of allowable bounds so using old value. Out of bounds value is: 199. Min limit: 500. Max limit: 5000
```

BMS is 3015 on this inverter's battery which is what I correctly see in the HA entity

gcoan, 2025-10-20:
A bit more digging on what predbat is doing when givtcp reports the error

predbat makes a get call to http://192.168.70.0:6346/readData and received the response "'Result': 'Error, no data available'"

it then retries, making a call to http://192.168.70.0:6346/runAll and when that fails it then uses the givtcp HA entities

5 minutes later when predbat next runs, it works fine, so I'm going to try adding a delay between the first and retry calls

johnwb87, 2025-10-23:
Thanks @britkat1980. On 3.4 overnight and first time in a long time the off-peak battery charge is showing correctly in the Energy Dashboard.

closed 2025-10-23 by britkat1980