britkat1980/giv_tcp #390: Temp Pause Charge Reverted

leocrawford, 2025-05-17:
I am trying to pause charging for a period of time, e.g. 10 minutes.

This works, and will count down from 10 in increments of 1 until it suddenly reverts to zero (before it shoud).

Checking the logs this appears to be

leocrawford, 2025-05-20:
I have now switched on debug, and see.

`Discharge_Time_Remaining is outside of allowable bounds so using old value. Out of bounds value is: 39168. Min limit: 0. Max limit: 20000`

leocrawford, 2025-10-08:
I've retested this in my kit, and it seems that  the timed pause fails after about 2m40 (for a 5 min pause). The only error line I can see is.

**2025-10-08 13:52:18,064 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.3.158**

I can't see any reason for that. I also note repeated " firstrun exists, so this should already have been run" and reopening of mqtt connection which seems bad.

Inverter Model
GIV-HY5.0
Software Version
D0.316-A0.316

The write log contains

 **13:49:24,841 - write - [DEBUG] - Pausing Charge for 5.0 minutes**
2025-10-08 13:49:24,847 - write - [DEBUG] - Setting battery charge rate to: 0 (0)
2025-10-08 13:49:24,970 - write - [DEBUG] - Pushing control update to mqtt: givtcp/FD2320G056/Control/Battery_Charge_Rate - 0
2025-10-08 13:49:24,977 - write - [DEBUG] - Pushing control update to pkl cache: Battery_Charge_Rate - 0
2025-10-08 13:49:24,977 - write - [INFO] - Setting battery charge rate 0 was a success
2025-10-08 13:49:24,980 - write - [DEBUG] - Temp Pause Charge revert jobid is: de4a622d-d9f6-41a0-83ad-03a558be6f11
2025-10-08 13:49:24,980 - write - [DEBUG] - Pushing control update to mqtt: givtcp/FD2320G056/Control/Temp_Pause_Charge - Running
2025-10-08 13:49:24,986 - write - [DEBUG] - Pushing control update to pkl cache: Temp_Pause_Charge - Running
2025-10-08 13:49:24,987 - write - [DEBUG] - Pushing control update to mqtt: givtcp/FD2320G056/Control/Battery_Charge_Rate - 0
2025-10-08 13:49:24,993 - write - [DEBUG] - Pushing control update to pkl cache: Battery_Charge_Rate - 0
2025-10-08 13:49:24,994 - write - [INFO] - Charge paused for 300.0 seconds
**2025-10-08 13:52:18,060 - write - [DEBUG] - Reverting Temp Pause Charge...**
2025-10-08 13:52:18,064 - write - [DEBUG] - Setting battery charge rate to: 0 (0)
2025-10-08 13:52:18,202 - write - [INFO] - Closing non readloop modbus connection
2025-10-08 13:52:18,212 - write - [DEBUG] - Pushing control update to mqtt: givtcp/FD2320G056/Control/Battery_Charge_Rate - 0
2025-10-08 13:52:18,421 - write - [DEBUG] - Pushing control update to pkl cache: Battery_Charge_Rate - 0
2025-10-08 13:52:18,422 - write - [INFO] - Setting battery charge rate 0 was a success
2025-10-08 13:52:18,424 - write - [DEBUG] - Pushing control update to mqtt: givtcp/FD2320G056/Control/Temp_Pause_Charge - Normal
2025-10-08 13:52:18,431 - write - [DEBUG] - Pushing control update to pkl cache: Temp_Pause_Charge - Normal
2025-10-08 13:52:18,433 - write - [DEBUG] - Pushing control update to mqtt: givtcp/FD2320G056/Control/Battery_Charge_Rate - 0
2025-10-08 13:52:18,438 - write - [DEBUG] - Pushing control update to pkl cache: Battery_Charge_Rate - 0
2025-10-08 13:52:18,439 - write - [INFO] - Temp Pause Charge Reverted

and the main logs

2025-10-08 13:49:23,780 - GivTCP - read        -  [DEBUG   ] - Data get was successful, now running handler if needed: 
2025-10-08 13:49:23,780 - GivTCP - read        -  [DEBUG   ] - Running processData
2025-10-08 13:49:23,780 - GivTCP - read        -  [DEBUG   ] - Beginning parsing of Inverter data
2025-10-08 13:49:23,783 - GivTCP - read        -  [DEBUG   ] - Getting Total Energy Data
2025-10-08 13:49:23,783 - GivTCP - read        -  [DEBUG   ] - Getting Today Energy Data
2025-10-08 13:49:23,784 - GivTCP - read        -  [DEBUG   ] - Getting PV Power
2025-10-08 13:49:23,784 - GivTCP - read        -  [DEBUG   ] - Getting Grid Power
2025-10-08 13:49:23,784 - GivTCP - read        -  [DEBUG   ] - Getting EPS Power
2025-10-08 13:49:23,784 - GivTCP - read        -  [DEBUG   ] - Getting PInv Power
2025-10-08 13:49:23,784 - GivTCP - read        -  [DEBUG   ] - Getting Load Power
2025-10-08 13:49:23,784 - GivTCP - read        -  [DEBUG   ] - Getting Self Consumption Power
2025-10-08 13:49:23,785 - GivTCP - read        -  [DEBUG   ] - Getting Solar to H/B/G Power Flows
2025-10-08 13:49:23,785 - GivTCP - read        -  [DEBUG   ] - Getting Grid to Battery/House Power Flow
2025-10-08 13:49:23,785 - GivTCP - read        -  [DEBUG   ] - Getting mode control figures
2025-10-08 13:49:23,785 - GivTCP - read        -  [DEBUG   ] - Calculating Mode...
2025-10-08 13:49:23,785 - GivTCP - read        -  [DEBUG   ] - Mode is: Eco
2025-10-08 13:49:23,785 - GivTCP - read        -  [DEBUG   ] - Force Export is not Running
2025-10-08 13:49:23,785 - GivTCP - read        -  [DEBUG   ] - Getting SOC
2025-10-08 13:49:23,786 - GivTCP - read        -  [DEBUG   ] - Getting Battery Energy Data
2025-10-08 13:49:23,786 - GivTCP - read        -  [DEBUG   ] - Getting Battery Power data
2025-10-08 13:49:23,786 - GivTCP - read        -  [DEBUG   ] - Getting Solar to H/B/G Power Flows
2025-10-08 13:49:23,786 - GivTCP - read        -  [DEBUG   ] - Getting Battery to House Power Flow
2025-10-08 13:49:23,786 - GivTCP - read        -  [DEBUG   ] - Getting Grid to Battery/House Power Flow
2025-10-08 13:49:23,787 - GivTCP - read        -  [DEBUG   ] - Getting Battery to Grid Power Flow
2025-10-08 13:49:23,787 - GivTCP - read        -  [DEBUG   ] - Getting TimeSlot data
2025-10-08 13:49:23,787 - GivTCP - read        -  [DEBUG   ] - Getting inverter Details
2025-10-08 13:49:23,788 - GivTCP - read        -  [DEBUG   ] - Getting Battery Details
2025-10-08 13:49:23,788 - GivTCP - read        -  [DEBUG   ] - Building battery output: 
2025-10-08 13:49:23,788 - GivTCP - read        -  [DEBUG   ] - Battery AH2425G440 added
2025-10-08 13:49:23,791 - GivTCP - read        -  [DEBUG   ] - cache len= 11
[... cut ...]
2025-10-08 13:52:18,213 - GivTCP - mqtt        -  [DEBUG   ] - MQTT Connection appears closed, re-opening
2025-10-08 13:52:18,213 - GivTCP - mqtt        -  [DEBUG   ] - Opening MQTT Connection to core-mosquitto
2025-10-08 13:52:18,214 - GivTCP - mqtt        -  [DEBUG   ] - In wait loop (single_MQTT_publish)
2025-10-08 13:52:18,218 - GivTCP - mqtt        -  [DEBUG   ] - MQTT connection disconnected
2025-10-08 13:52:18,219 - GivTCP - mqtt        -  [DEBUG   ] - connected OK Returned code=Success
2025-10-08 13:52:18,219 - GivTCP - mqtt        -  [DEBUG   ] - Subscribing to givtcp/control/FD2320G056/#

britkat1980, 2026-10-11:
Closing this as it's had no activity for over 8 weeks.

GivTCP 3.6 has since been released ([latest: v3.6.2](https://github.com/britkat1980/giv_tcp/releases/latest)). It moves to a new Modbus library and includes a large number of fixes, so this may well behave differently now. If you still see this on 3.6.2, please reopen it, or open a new issue, with a debug log.

closed 2026-10-11 by britkat1980