britkat1980/giv_tcp #391: Battery Charge Value resets after next update from Add in.

APoller1, 2025-05-20:
The Battery Charge Rate in the GIVTCP is not persistent. I can change it in the add on, but it reverts after about 30 seconds to 2600.
The log shows
"2025-05-20 09:16:34,566 - GivTCP - write       -  [INFO    ] - Setting battery charge rate AC to 10% was a success
2025-05-20 09:18:57,996 - GivTCP - write       -  [INFO    ] - Setting battery charge rate 2100 was a success
2025-05-20 09:20:19,645 - GivTCP - write       -  [INFO    ] - Setting battery charge rate 150 was a success
2025-05-20 09:26:13,623 - GivTCP - write       -  [INFO    ] - Setting battery charge rate 1250 was a success
2025-05-20 09:33:18,606 - GivTCP - write       -  [INFO    ] - Setting battery charge rate 2100 was a success
2025-05-20 09:39:49,412 - GivTCP - write       -  [INFO    ] - Setting battery charge rate 1529 was a success
2025-05-20 09:39:50,914 - GivTCP - write       -  [INFO    ] - Setting battery charge rate 827 was a success
2025-05-20 09:39:55,429 - GivTCP - write       -  [INFO    ] - Setting battery charge rate 308 was a success
2025-05-20 09:47:33,813 - GivTCP - write       -  [INFO    ] - Setting battery charge rate 1566 was a success
2025-05-20 09:48:03,265 - GivTCP - write       -  [INFO    ] - Setting battery charge rate 694 was a success
2025-05-20 09:49:34,533 - GivTCP - write       -  [INFO    ] - Setting battery charge rate 849 was a success
2025-05-20 09:52:52,736 - GivTCP - write       -  [INFO    ] - Setting battery charge rate 1082 was a success"

The entity log shows the value persists for about 30 seconds (or until the next poll maybe)

This is preventing Clipping control and avoiding overheating of the inverter/battery

gcoan, 2025-05-20:
Have you got any other automations going on, are you on Octopus Intelligent Flux or any other similar smart tariff?

What kit have you got?

APoller1, 2025-05-20:
I have only Home Assistant.
I am on Octopus Go (EV economy 6 tariff) , but it does not control my
inverter.
I just changed the setting and checked whether any other automations were
triggered. There were none.

I have a Gen1 inverter hybrid
With a 9.5kWh battery.

The battery charge rate update used to work, but something has changed
since last Autumn when I last ran the clipping automations.

I have upgraded to V3 since last Autumn...

APoller1, 2025-05-20:
No change to firmware in either the battery or inverter.
Battery is offering an update to 3020, but I've held off.

gcoan, 2025-05-20:
Might be worth rebooting your inverter in case it's having problems.  I have a gen 1 hybrid and am not seeing such issues with givtcp 3

APoller1, 2025-05-21:
Thought I'd done that, but it seems ok now. Thank you.

closed 2025-05-21 by APoller1

gcoan, 2025-05-21:
> Thought I'd done that, but it seems ok now. Thank you.

sometimes the inverter just stops accepting commands.  Its happened to me, and a reboot sorts it out.

Cheers