britkat1980/giv_tcp #394: Only 1 battery detected on 2 battery system

holdestmade, 2025-06-09:
I have 2 batteries connected to an AC3 inverter but only 1 is detected. I think this is because EON have wired them to battery 1 and battery 3 slots, according to GivEnergy.

Using GivTCP v2, they are both detected but only if I select that I have 3 batteries, and then there is a warning in the log:

`2025-06-09 09:43:56,955 - Inv1 - read        -  [ERROR   ] - Battery Object empty so skipping`

both battery devices are created in HA though

Is there any way for GivTCP v3 to work around this ?

britkat1980, 2025-08-26:
OK, GivTCP will search through batteries until it finds a "blank" battery then stops. Rather than iterating through all 8 possible slots to check. Let me see if there's an elegant way to check all, which doesn't require a major refactor in the code.

gcoan, 2025-08-26:
> I have 2 batteries connected to an AC3 inverter but only 1 is detected. I think this is because EON have wired them to battery 1 and battery 3 slots, according to GivEnergy.
> 

@holdestmade The battery 'number' is controlled by DIP switches on the side panel so it should be straight forward enough to change them to set the second battery to id 2.  Power the inverter and batteries off before making the change.

See page 7 of the battery installation manual https://givenergy.co.uk/wp-content/uploads/UK-Installation-Manual-Gen-2-Battery.pdf

holdestmade, 2025-08-27:
So I looked and indeed one of the batteries had the dipswitch set to 0 and one set to 2, so changed it to 1 and now it works great, many, many thanks @gcoan.

@britkat1980 Up to you if you want to close this now or add the refactor to include all battery addresses. Only issue is the Inverter name is "na - AC", I have an AC3.

gcoan, 2025-08-27:
@britkat1980 probably worth adding something to the readme explaining that GivTCP stops scanning when it finds a gap in the battery numbers, so the batteries need to be set without any numbering gaps or they won't be detected