britkat1980/giv_tcp #408: GivTCP Export entity statistics negative on 3.x compared to 2x

pacemaker82, 2025-08-03:
Entity: sensor.givtcp_ceXXXXXXXXX_export_energy_today_kwh

I was looking at historical statistics of this entity and noticed that during the time I was on GivTCP 3.x, this entity was reporting negative numbers.

Here is a chart (using statistics card) showing the daily statistics records for the entity above:

[image]

You can see during the period January to early May, the export numbers show negative export, which is when I was on 3.x. On 2.x before and after these periods the entity is reporting statics correctly, e.g. always positive.

If I look at the hard state data, they appear all positive, it just seems to be something to do with how the statistics are saved for this entity that is different between GivTCP 2.x and 3.x

[image]

[image]

gcoan, 2025-08-03:
@pacemaker82 I think you're going to find it very difficult to get anything identified on this without being able to see it happening in real time.

On my today sensors I don't see any negative values for import or export today over the last year.  Not sure when I moved to v3, it was a while ago, but if the behaviour changed then it would have shown up 
[image]

Statistics are automatically captured by HA based on the change in entity value every 5 minutes then every hour.  There were/still sometimes are problems with v3 not resetting the today entities precisely at midnight which might be the cause of what you see?   For this reason I moved to using the total values in the energy dashboard.

Probably only way forward is to fix the errant statistics values

gcoan, 2025-08-03:
I see these negative export values as well with the statistics card on export:

[image]

Zooming in to the last 5 days:

[image]

And looking at the breakdown by hour shows that the negatives occur at midnight:

[image]

I can explain why I don't think there is a problem here:

The today state increases through the day, e.g.

7pm 23kWh
8pm 25kWh
9pm 30kWh
10pm 30kWh
11pm 30kWh
Midnight 30kWh
1am 0kWh
2am 0kWh

Long Term Statistics are captured every hour and record the change in state value over that hour.  So the statistics will be:

8pm 2kWh
9pm 5kWh
10pm 0kWh
11pm 0kWh
Midnight 0kWh
1am -30kWh
2am 0kWh

Which explains how you could see the result you do from the givtcp sensor.

But curiously I don't see the same behaviour from other non-givtcp 'today' sensors, e.g. from Shelly Energy monitors:
[image]

Maybe its due to the givtcp sensors not quite resetting at midnight which confuses HA?

gcoan, 2025-08-03:
Update, looking at it more, I am pretty confident that the issue is caused by the givtcp sensors not resetting to zero at midnight and this causes the HA stats to be confused.

If I zoom in and look at the hourly stats (top chart) and state (bottom chart), showing a givtcp and two different Shelly sensors (one from a Shelly PM, one from a Shelly EM):

[image]

Can see that the two Shelly devices the today values reset to zero at midnight precisely

Whereas the (red line) givtcp export entity resets at 00:00:24.  And so this means that the entity changes state by minus the previous days export value in the hour from midnight to 1am, and you see this in the stats top graph.

In the previous post you could see some days there were negative export values, others there were not, I guess the variability is due to when the givtcp sensors reset to zero.

joshblease, 2025-08-06:
I've noticed an issue with my setup where `sensor.giv_abc_load_energy_total_kwh` is producing negative values.
This is causing issues with my utility meters which I use instead of the `sensor.giv_abc_load_energy_today_kwh` entities due to the values not resetting at midnight.
In the screenshot below you can clearly see a lot of values being less than the previous value which is causing me headaches

[image]

pacemaker82, 2025-10-20:
@paulgee31 is it at midnight though? I think that was the issue

paulgee31, 2025-10-20:
@pacemaker82 Certainly resetting at midnight for me and no negative values.