britkat1980/giv_tcp #420: Cannot connect to inverter

lamztib, 2025-09-18:
I have tried to reinstall - but still get these messages in GivTCP, and connection to inverter is not working...

2025-09-18 22:27:44,460 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-09-18 22:27:44,470 - startup     -  [INFO    ] - Network Found: 192.168.1.70/24
2025-09-18 22:27:46,475 - startup     -  [INFO    ] - Searching for Inverters
2025-09-18 22:27:46,476 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-09-18 22:28:11,798 - client      -  [INFO    ] - Connection established to 192.168.1.5:8899
2025-09-18 22:28:11,798 - client      -  [INFO    ] - Detecting plant
2025-09-18 22:28:16,215 - client      -  [INFO    ] - Plant Detected
2025-09-18 22:28:21,999 - client      -  [INFO    ] - Connection established to 192.168.1.5:8899
2025-09-18 22:28:21,999 - client      -  [INFO    ] - Detecting plant
2025-09-18 22:28:36,926 - client      -  [INFO    ] - Plant Detected
2025-09-18 22:28:47,991 - startup     -  [INFO    ] - Setting up invertor: 1
2025-09-18 22:28:48,002 - client      -  [INFO    ] - Connection established to 192.168.1.5:8899
2025-09-18 22:28:48,002 - client      -  [INFO    ] - Detecting plant
[2025-09-18 22:28:48 +0200] [90] [INFO] Starting gunicorn 23.0.0
[2025-09-18 22:28:48 +0200] [90] [INFO] Listening at: http://0.0.0.0:6350 (90)
[2025-09-18 22:28:48 +0200] [90] [INFO] Using worker: sync
[2025-09-18 22:28:48 +0200] [100] [INFO] Booting worker with pid: 100
2025-09-18 22:28:52,672 - client      -  [INFO    ] - Plant Detected
Traceback (most recent call last):
  File "/app/startup.py", line 631, in <module>
    createsettingsjson(inv)
    ~~~~~~~~~~~~~~~~~~^^^^^
  File "/app/startup.py", line 93, in createsettingsjson
    setts["Model_"+str(inv)]= inverter_type['Model'].name.capitalize()
                              ~~~~~~~~~~~~~^^^^^^^^^
TypeError: 'NoneType' object is not subscriptable

gcoan, 2025-09-18:
The error message indicates GivTCP is having problems determining your inverter model type, but yet you haven't given any information about what equipment, inverters, batteries, etc you have.

Please help us by supplying as much information as you can

gcoan, 2025-09-19:
> I had the idea that the integration worked before with this 3.2 version... but when i check now the changelog it tells "Refactored inverter Model to include new/upcoming models" - so maybe it has something to do with that?

sounds likely.

The 3 phase hybrid inverters were supported in earlier versions of givtcp v3, not perfect, there were some issues, but I think they got further than you are getting.

Can you turn on debug mode in the addon and save the log here to give some more info for @britkat1980.

If you try the beta version of givtcp, that is I think still on the 3.1.6 codebase.  You just need to install it and stop givtcp then start the beta addon and it will pickup your config etc.

lamztib, 2025-09-19:
I just tried the DEV and BETA versions, but i get the same... I posted the log with debugging on below:

2025-09-19 10:30:23,949 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-09-19 10:30:23,957 - startup     -  [INFO    ] - Network Found: 192.168.1.70/24
2025-09-19 10:30:25,968 - startup     -  [INFO    ] - Searching for Inverters
2025-09-19 10:30:25,969 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-09-19 10:30:51,486 - client      -  [INFO    ] - Connection established to 192.168.1.5:8899
2025-09-19 10:30:51,486 - client      -  [INFO    ] - Detecting plant
2025-09-19 10:30:55,906 - client      -  [INFO    ] - Plant Detected
2025-09-19 10:31:06,430 - client      -  [INFO    ] - Connection established to 192.168.1.5:8899
2025-09-19 10:31:06,430 - client      -  [INFO    ] - Detecting plant
2025-09-19 10:31:17,859 - client      -  [INFO    ] - Plant Detected
2025-09-19 10:31:27,392 - startup     -  [INFO    ] - Setting up invertor: 1
2025-09-19 10:31:27,410 - client      -  [INFO    ] - Connection established to 192.168.1.5:8899
2025-09-19 10:31:27,410 - client      -  [INFO    ] - Detecting plant
[2025-09-19 10:31:27 +0200] [90] [INFO] Starting gunicorn 23.0.0
[2025-09-19 10:31:27 +0200] [90] [INFO] Listening at: http://0.0.0.0:6350 (90)
[2025-09-19 10:31:27 +0200] [90] [INFO] Using worker: sync
[2025-09-19 10:31:27 +0200] [100] [INFO] Booting worker with pid: 100
2025-09-19 10:31:31,377 - framer      -  [WARNING ] - Unexpected header values found (len=0042, u_id=01, f_id=03), discarding candidate frame and resuming search
2025-09-19 10:31:31,378 - framer      -  [INFO    ] - No frame header found, await more data
2025-09-19 10:31:33,837 - framer      -  [WARNING ] - Candidate frame found 68 bytes into buffer, discarding leading garbage: 0x004201035754323333374731333203495020616464726573733a003139322e3136382e312e352c3235352e3235352e3235352e302c3139322e3136382e312e310d0a0d0a
2025-09-19 10:31:35,338 - client      -  [INFO    ] - Plant Detected
Traceback (most recent call last):
  File "/app/startup.py", line 631, in <module>
    createsettingsjson(inv)
    ~~~~~~~~~~~~~~~~~~^^^^^
  File "/app/startup.py", line 93, in createsettingsjson
    setts["Model_"+str(inv)]= inverter_type['Model'].name.capitalize()
                              ~~~~~~~~~~~~~^^^^^^^^^
TypeError: 'NoneType' object is not subscriptable

britkat1980, 2025-09-26:
So this is a three-phase 11kW inverter? I haven't tested on one of those, so I might need to do some testing to see what the data returns, so I can make it compatable.

lamztib, 2025-09-26:
> So this is a three-phase 11kW inverter? I haven't tested on one of those, so I might need to do some testing to see what the data returns, so I can make it compatable.

Before it was working / gave some data back, but now with this 3.2 version or with the update of HA it gives these errors.
The inverter is indeed a 11 kW 3-Phase inverter (model GIV-3HY-11) and it has 6 stacked battery's (model GIV-BAT-3.4).
Is there anything what i can do/provide to help?

LeeFarm11, 2025-09-26:
@lamztib I am also running same Inverter and Batteries, and I don't have the same issue. I am still running DEV 3.1.8

It does not work very well but I don't have time to try updating to later versions at the moment.

There are other GivTCP GitHub issues raised relating to 3 phase inverters which appear to work completely differently to the single phase inverters. 

With my current install, GivTCP (not predbat, not Givenergy) now keeps the maximum charge and discharge at 46%. Go figure.... I have created automations to check the values and if they are not 100%, then set them to 100%. The Discharge works fine, but the Charge seems to have a timing issue so ends up only about 50% of the time at 100%.

Predbat does not talk well with GivTCP yet for 3 phase inverter. I have relatively excess solar panels so I end up setting the controls manually each day, depending if I want to prevent clipping on sunny days, and if I need to add charging from the grid on cold cloudy days. 

I do have predbat and GivTCP DEV running almost all the time, except when there is free electricity. Then I turn off predbat and GivTCP, so that I can ensure that charging the battery is at 100%. I use the solcast integration to help decide what to do.

```
2025-09-26 22:31:56,538 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-09-26 22:31:56,586 - startup     -  [INFO    ] - Network Found: 192.168.1.204/24
2025-09-26 22:31:58,589 - startup     -  [INFO    ] - Searching for Inverters
2025-09-26 22:31:58,590 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-09-26 22:31:59,853 - startup     -  [INFO    ] - GivEVC 11288853538258 found at: 192.168.1.102
2025-09-26 22:32:05,221 - client      -  [INFO    ] - Connection established to 192.168.1.101:8899
2025-09-26 22:32:05,221 - client      -  [INFO    ] - Detecting plant
2025-09-26 22:32:09,643 - client      -  [INFO    ] - Plant Detected
2025-09-26 22:32:17,925 - startup     -  [INFO    ] - Inverter TD2343G049 which is a Gen1 - Hybrid_3ph with 6 batteries has been found at: 192.168.1.101
2025-09-26 22:32:17,944 - startup     -  [INFO    ] - Setting up invertor: 1
2025-09-26 22:32:18,102 - startup     -  [INFO    ] - ==============================================================
2025-09-26 22:32:18,103 - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2025-09-26 22:32:18,104 - startup     -  [INFO    ] - ====     http://192.168.1.204:8099/config.html             ====
2025-09-26 22:32:18,105 - startup     -  [INFO    ] - ==============================================================
2025-09-26 22:32:18,108 - startup     -  [INFO    ] - Running Invertor 1 (TD2343G049) read loop every 60/300s
2025-09-26 22:32:18,121 - startup     -  [INFO    ] - Serving Web Dashboard from port 3000
[2025-09-26 22:32:19 +0100] [50] [INFO] Starting gunicorn 23.0.0
[2025-09-26 22:32:19 +0100] [50] [INFO] Listening at: http://0.0.0.0:6350 (50)
[2025-09-26 22:32:19 +0100] [50] [INFO] Using worker: sync
[2025-09-26 22:32:19 +0100] [62] [INFO] Booting worker with pid: 62
[2025-09-26 22:32:19 +0100] [58] [INFO] Starting gunicorn 23.0.0
[2025-09-26 22:32:19 +0100] [58] [INFO] Listening at: http://0.0.0.0:6345 (58)
[2025-09-26 22:32:19 +0100] [58] [INFO] Using worker: sync
[2025-09-26 22:32:19 +0100] [63] [INFO] Booting worker with pid: 63
[2025-09-26 22:32:19 +0100] [64] [INFO] Booting worker with pid: 64
[2025-09-26 22:32:19 +0100] [65] [INFO] Booting worker with pid: 65
2025-09-26 22:32:24,280 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-09-26 22:32:24,281 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.1.101
2025-09-26 22:32:24,289 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-09-26 22:32:49,566 - GivTCP - read        -  [DEBUG   ] - Running full refresh
```

AcidClould, 2025-10-20:
Edited the Json file with Network address of the inverter established connection but failed to gather data 

2025-10-20 20:52:14,217 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-10-20 20:52:16,288 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-10-20 20:52:16,288 - startup     -  [INFO    ] - Scanning network: 192.168.0.87/24
2025-10-20 20:52:28,123 - client      -  [INFO    ] - Connection established to 192.168.0.56:8899
2025-10-20 20:52:28,123 - client      -  [INFO    ] - Detecting plant
2025-10-20 20:53:06,667 - startup     -  [ERROR   ] - Gathering inverter details for 192.168.0.56 failed.
2025-10-20 20:53:06,671 - client      -  [INFO    ] - Connection established to 192.168.0.56:8899
2025-10-20 20:53:06,671 - client      -  [INFO    ] - Detecting plant
2025-10-20 20:53:45,211 - startup     -  [ERROR   ] - Gathering inverter details for 192.168.0.56 failed.
2025-10-20 20:53:45,296 - startup     -  [INFO    ] - EVC IP is missing from config. Please update and restart GivTCP
2025-10-20 20:53:45,298 - startup     -  [INFO    ] - Setting up invertor: 1
[2025-10-20 20:53:45 +0100] [70] [INFO] Starting gunicorn 23.0.0
[2025-10-20 20:53:45 +0100] [70] [INFO] Listening at: http://0.0.0.0:6350 (70)
[2025-10-20 20:53:45 +0100] [70] [INFO] Using worker: sync
[2025-10-20 20:53:45 +0100] [76] [INFO] Booting worker with pid: 76
2025-10-20 20:53:48,414 - startup     -  [ERROR   ] - Gathering inverter details for 192.168.0.56 failed.
Traceback (most recent call last):
  File "/app/startup.py", line 634, in <module>
    createsettingsjson(inv)
    ~~~~~~~~~~~~~~~~~~^^^^^
  File "/app/startup.py", line 98, in createsettingsjson
    setts["Model_"+str(inv)]= inverter_type['Model'].name.capitalize()
                              ~~~~~~~~~~~~~^^^^^^^^^
TypeError: 'NoneType' object is not subscriptable

lamztib, 2025-10-21:
I've edited the json file with these:
"invertorIP_1": "192.168.1.5",
    "serial_number_1": "TC2xxxxxxx",
    "inverterName_1": "GivTCP",
 "auto_scan": false,
    "Model_1": "Hybrid",

And now it seems to detect my plant again...

2025-10-21 07:34:21,640 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-10-21 07:34:23,659 - startup     -  [INFO    ] - Setting up invertor: 1
2025-10-21 07:34:23,696 - startup     -  [INFO    ] - ==============================================================
2025-10-21 07:34:23,696 - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2025-10-21 07:34:23,696 - startup     -  [INFO    ] - ====     http://192.168.1.10:8099/config.html             ====
2025-10-21 07:34:23,696 - startup     -  [INFO    ] - ==============================================================
2025-10-21 07:34:23,698 - startup     -  [INFO    ] - Running Invertor 1 (TC2xxxxxxx) read loop every 30/120s
[2025-10-21 07:34:23 +0200] [10] [INFO] Starting gunicorn 23.0.0
[2025-10-21 07:34:23 +0200] [10] [INFO] Listening at: http://0.0.0.0:6350 (10)
[2025-10-21 07:34:23 +0200] [10] [INFO] Using worker: sync
[2025-10-21 07:34:23 +0200] [19] [INFO] Booting worker with pid: 19
[2025-10-21 07:34:24 +0200] [18] [INFO] Starting gunicorn 23.0.0
[2025-10-21 07:34:24 +0200] [18] [INFO] Listening at: http://0.0.0.0:6345 (18)
[2025-10-21 07:34:24 +0200] [18] [INFO] Using worker: sync
[2025-10-21 07:34:24 +0200] [20] [INFO] Booting worker with pid: 20
[2025-10-21 07:34:24 +0200] [21] [INFO] Booting worker with pid: 21
[2025-10-21 07:34:24 +0200] [24] [INFO] Booting worker with pid: 24
2025-10-21 07:34:24,467 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-10-21 07:34:24,467 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.1.5
2025-10-21 07:34:24,469 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-10-21 07:34:45,498 - GivTCP - read        -  [INFO    ] - Dynamic Tariff enabled so defaulting to Day Rate and waiting external control
2025-10-21 07:34:45,506 - GivTCP - read        -  [INFO    ] - Publishing Home Assistant Discovery messages

tonka-truck, 2025-10-23:
I’ve updated to 3.4.1 and 3.4.8 and unfortunately this means GivTCP no longer starts

2025-10-23 05:06:38,340 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-10-23 05:06:38,340 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.73.41
2025-10-23 05:06:38,345 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-10-23 05:07:03,582 - GivTCP - read        -  [ERROR   ] - AttributeError: 'NoneType' object has no attribute 'start'
2025-10-23 05:07:03,582 - GivTCP - read        -  [ERROR   ] - Error processing Three Phase data: ('AttributeError', 'read.py', 1945)
2025-10-23 05:07:03,582 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: (Exception: Process Data Failure - 1994)
2025-10-23 05:07:03,583 - GivTCP - read        -  [ERROR   ] - Key Error getting Battery Data: Missing key 'multi_output'
2025-10-23 05:07:03,583 - GivTCP - read        -  [ERROR   ] - Error in calling handler: cannot access local variable 'multi_output' where it is not associated with a value
2025-10-23 05:07:36,556 - GivTCP - read        -  [ERROR   ] - AttributeError: 'NoneType' object has no attribute 'start'
2025-10-23 05:07:36,556 - GivTCP - read        -  [ERROR   ] - Error processing Three Phase data: ('AttributeError', 'read.py', 1945)
2025-10-23 05:07:36,557 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: (Exception: Process Data Failure - 1994)
2025-10-23 05:07:36,557 - GivTCP - read        -  [ERROR   ] - Key Error getting Battery Data: Missing key 'multi_output'
2025-10-23 05:07:36,557 - GivTCP - read        -  [ERROR   ] - UnboundLocalError: cannot access local variable 'multi_output' where it is not associated with a value
2025-10-23 05:07:36,557 - GivTCP - read        -  [ERROR   ] - Error in calling handler: ('UnboundLocalError', 'read.py', 257)
2025-10-23 05:08:09,526 - GivTCP - read        -  [ERROR   ] - AttributeError: 'NoneType' object has no attribute 'start'
2025-10-23 05:08:09,527 - GivTCP - read        -  [ERROR   ] - Error processing Three Phase data: ('AttributeError', 'read.py', 1945)
2025-10-23 05:08:09,527 - GivTCP - read        -  [ERROR   ] - inverter Update failed so using last known good data from cache: (Exception: Process Data Failure - 1994)
2025-10-23 05:08:09,527 - GivTCP - read        -  [ERROR   ] - Key Error getting Battery Data: Missing key 'multi_output'
2025-10-23 05:08:09,527 - GivTCP - read        -  [ERROR   ] - UnboundLocalError: cannot access local variable 'multi_output' where it is not associated with a value
2025-10-23 05:08:09,527 - GivTCP - read        -  [ERROR   ] - Error in calling handler: ('UnboundLocalError', 'read.py', 257)

tonka-truck, 2025-10-23:
I've updated to 3.4.14 and rerun the config page and restarted.  The log file records the correct serial number, number of batteries and meter, however the Model_1  is not being saved to allsettings.json. Manually setting the model version and  GivTCP runs without error and data published in the dashboard aligns with GivEnergy.cloud.

britkat1980, 2025-10-23:
I've just tested on my setup and when I delete my allsettings.json file and re-run it saves properly. Does it identify the model in the start-up message?

britkat1980, 2025-10-23:
What does redoing the config entail? Making a change in the web tool or manually changing the json file?

tonka-truck, 2025-10-23:
Re running the web tool

britkat1980, 2026-10-11:
Closing this as it's had no activity for over 8 weeks.

GivTCP 3.6 has since been released ([latest: v3.6.2](https://github.com/britkat1980/giv_tcp/releases/latest)). It moves to a new Modbus library and includes a large number of fixes, so this may well behave differently now. If you still see this on 3.6.2, please reopen it, or open a new issue, with a debug log.

closed 2026-10-11 by britkat1980